MarlinFirmware/Marlin #2695: M666 - Missing online adjustment options

DIYlover, 2015-10-10:
I found there is missing some options for Delta parameter adjustment on-the-fly as following.

@Marlin_main.cpp

#ifdef DELTA
      case 666: // M666 set delta endstop and geometry adjustment
         for(int8_t i=0; i < 3; i++)
         {
           if (code_seen(axis_codes[i])) endstop_adj[i] = code_value();
         }
	   if (code_seen('A')) {
		tower_adj[0] = code_value();
		set_delta_constants();
	   }
	   if (code_seen('B')) {
		tower_adj[1] = code_value();
		set_delta_constants();
	   }
	   if (code_seen('C')) {
		tower_adj[2] = code_value();
		set_delta_constants();
	   }
           if (code_seen('I')) {
		tower_adj[3] = code_value();
		set_delta_constants();
	   }
	   if (code_seen('J')) {
		tower_adj[4] = code_value();
		set_delta_constants();
	   }
	   if (code_seen('K')) {
		tower_adj[5] = code_value();
		set_delta_constants();
	   }
           if (code_seen('R')) {
           delta_radius = code_value();
           set_delta_constants();
         }
           if (code_seen('D')) {
             delta_diagonal_rod = code_value();
             set_delta_constants();
         }
           if (code_seen('H')) {
             max_pos[Z_AXIS]= code_value();
	     set_delta_constants();
         }
	   if (code_seen('P')) {
             z_probe_offset[Z_AXIS]= code_value();
	   }
	   if (code_seen('L')) {
	     SERIAL_ECHOLN("Current Delta geometry values:");
	     SERIAL_ECHOPAIR("X (Endstop Adj): ",endstop_adj[0]);
             SERIAL_ECHOLN("");
	     SERIAL_ECHOPAIR("Y (Endstop Adj): ",endstop_adj[1]);
             SERIAL_ECHOLN("");
	     SERIAL_ECHOPAIR("Z (Endstop Adj): ",endstop_adj[2]);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("P (Z-Probe Offset): X", z_probe_offset[0]);
             SERIAL_ECHOPAIR(" Y", z_probe_offset[1]);
             SERIAL_ECHOPAIR(" Z", z_probe_offset[2]);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("A (Tower A Position Correction): ",tower_adj[0]);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("B (Tower B Position Correction): ",tower_adj[1]);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("C (Tower C Position Correction): ",tower_adj[2]);
	     SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("I (Tower A Radius Correction): ",tower_adj[3]);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("J (Tower B Radius Correction): ",tower_adj[4]);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("K (Tower C Radius Correction): ",tower_adj[5]);
	     SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("R (Delta Radius): ",delta_radius);
             SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("D (Diagonal Rod Length): ",delta_diagonal_rod);
	     SERIAL_ECHOLN("");
             SERIAL_ECHOPAIR("H (Z-Height): ",max_pos[Z_AXIS]);
             SERIAL_ECHOLN("");
             }
         break;
    #endif

thinkyhead, 2016-02-07:
Has a PR been made for this? We would just need to extend the code in the `gcode_M666` function with these added options.

Roxy-3DPrintBoard, 2016-02-07:
@thinkyhead 

Just as an FYI:    I extended M665 when I was bringing up my printer.   The changes are in the Jan10 MarlinDev branch:

~~~~
#if ENABLED(DELTA)
  /**
   * M665: Set delta configurations
   *
   *    L = diagonal rod
   *    R = delta radius
   *    S = segments per second
   *    A = Alpha (Tower 1) diagonal rod trim
   *    B = Beta (Tower 2) diagonal rod trim
   *    C = Gamma (Tower 3) diagonal rod trim
   *    I = Alpha (Tower 1) radius trim
   *    J = Beta (Tower 2) radius trim
   *    K = Gamma (Tower 3) radius trim
   */
  inline void gcode_M665() {
    if (code_seen('L')) delta_diagonal_rod = code_value();
    if (code_seen('R')) delta_radius = code_value();
    if (code_seen('S')) delta_segments_per_second = code_value();
    if (code_seen('A')) delta_diagonal_rod_trim_tower_1 = code_value();
    if (code_seen('B')) delta_diagonal_rod_trim_tower_2 = code_value();
    if (code_seen('C')) delta_diagonal_rod_trim_tower_3 = code_value();
    if (code_seen('I')) delta_radius_trim_tower_1 = code_value();
    if (code_seen('J')) delta_radius_trim_tower_2 = code_value();
    if (code_seen('K')) delta_radius_trim_tower_3 = code_value();
    recalc_delta_settings(delta_radius, delta_diagonal_rod);
  }
~~~~

thinkyhead, 2016-02-19:
@Roxy-3DPrintBoard Aha. I will have to remember to check all the branches at `MarlinDev` to see what's new. I've been bringing in bug-fixes from the `dev` branch, but forgetting to check others also.

Roxy-3DPrintBoard, 2016-02-19:
> I assume that the changes suggested above are still pending…?

Well...  There is no Pull Request or anything formal associated with it.   It is part of the work I'm doing trying to make a High Resolution (and as much as possible automatic) Mes Leveling system for Delta's.     But you might want to also look at the G27.   It is a Physical Bed Leveling command focused on Delta Printers.   It makes it a lot easier to get your bed and Delta Radius parameters dialed in.

thinkyhead, 2016-02-28:
> no Pull Request or anything formal associated 

Maybe I can throw together a PR with the `M666` extensions above.

> also look at the G27. It is a Physical Bed Leveling

It sounds intriguing! A lot of interesting projects going on around here. I will check it out soon.

closed 2016-05-04 by jbrazio