MarlinFirmware/Marlin #2697: 1.1.RC1 LCD flow rate adjustment causes zero extrusion

o-lampe, 2015-10-11:
I see we have a RC2 version already, but I checked the issue list and this one wasn't mentioned yet.

When I adjust extruder flow rate ( reduce to 92% ) the extrusion stops completely. Only retraction moves are done.
After a few seconds ( when the printbuffer is refreshed ?) the extrusion starts working as expected..

My config: Delta/Ramps/RRD-LCD2004

AnHardt, 2015-10-26:
@o-lampe 
How many extruders do you have defined? Only one or more?

o-lampe, 2015-10-26:
@AnHardt 
Only one extruder.

Wurstnase, 2015-10-30:
A friend has the same issue. Setup is Rumba/Viki2/dual Extruder.

thinkyhead, 2016-02-10:
Anyone still seeing this with the latest RC? If not, we can close it and assume it's not an issue any longer.

jbrazio, 2016-04-23:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-04-23 by jbrazio

toby4548, 2016-08-01:
Hello,
I'm using v1.1 RC7 now and have the same problem.
Any flowrate under 100% will cause the extruder stop.

 My config is: Kossel mini/Ramps 1.4/RRD-LCD2004

jbrazio, 2016-08-01:
@toby4548 I'd suggest to open a new ticket.