MarlinFirmware/Marlin #2702: Can get my Lcd to work

exsaevio, 2015-10-13:
Sorry if this is not that place to make this requast. But i have an Wanhao I3 printer and i cant get the lcd to work. There is a Repetier on the machine stock. But i would like to have marlin. I have the original Repetier on file. Is there any way of figure out what setting there is? i have triede to wrap my head around it but no luck. There maust be some pin setting?
Best regards

KiteLab, 2015-10-13:
Ask Wanhao for the original source code and the used configuration file. The configuration file is not that difficult to read.

At the picture on their website it looks like one of the graphic displays. If you have already the right MOTHERBOARD for your printer you could try to test REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER or one of the other displays/panels using u8glib.

exsaevio, 2015-10-13:
Hello Again and thanks for answer. I have management to get The original source file . Been sitting a week trying to figurer out how to read The original file so i can takeit to marlin. And i have tried REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER but no  luck. And i tried customer service but no answer There. I wish it was East :(

exsaevio, 2015-10-13:
Meant easy

exsaevio, 2015-10-13:
And i have tried all of The display!

NarimaanV, 2015-10-13:
I searched "Wanhao I3 LCD" and the second result I got was a link to a site selling replacement parts for Wanhao printers, include the screen for the Duplicator I3:
http://www.3dprima.com/en/accessories/spare-parts/wanhao-i3/wanhao-duplicator-i3-lcd-display.html
The PCB of the screen in the picture on that site says "LCD for Melzi V4", which makes sense because it seems the Wanhao I3 runs off of a Melzi board. From a little more research, there doesn't seem to be an "official Melzi display", but both the Panelolu 2 and the RA Control Panel are theoretically compatible with the Melzi. Just to confirm, you've tried both those two for your display and it hasn't worked?

exsaevio, 2015-10-13:
Hello i can´t go heads or tails with this!  I tried Again´with Panelou 2 When u wanted confirmation and noo luck 
And i cant see what setting in repitier that do this! And as u did! I research and got the same results as u So what so special with this panel? They made somthing strange. Dont want to give up but maby i have to?

NarimaanV, 2015-10-13:
Can you post the config file(s) and post a link here? It might help to have a few other pairs of eyes look through to see if anyone can identify what display is being used.

exsaevio, 2015-10-13:
Ofcourse thought you never Ask :-)
https://dl.dropboxusercontent.com/u/12600947/NoBeep.zip

KiteLab, 2015-10-13:
Your configs
#define FEATURE_CONTROLLER 11
translates to
11 or CONTROLLER_REPRAPDISCOUNT_GLCD = RepRapDiscount Full Graphic Smart Controller
in Marlin
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

KiteLab, 2015-10-13:
Your configurations
#define MOTHERBOARD 63
translates to MELZI
translates to Marlin
#define BOARD_MELZI             63   // Melzi

KiteLab, 2015-10-13:
Start reading at #1386

exsaevio, 2015-10-13:
Hmm yes all that i figured  with the Melzy
and i tried REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER to but no luck and i tried Makr3d with the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. 
Seem like a lost cause! And i mad that the company wont answer my mails
im going to look at the #1386 see if that do something

AnHardt, 2015-10-26:
Any progress here?

closed 2015-11-03 by AnHardt