MarlinFirmware/Marlin #2703: [Request] ZeroPI support

Grogyan, 2015-10-13:
I'm sure you guys are aware that there is a new board coming out soon called the ZeroPI
https://www.kickstarter.com/projects/1204283/zeropi-arduino-and-raspberry-pi-compatible-develop/description

Will Marlin be able to support this?

KiteLab, 2015-10-13:
Unlikely. There is no AVR in.

Grogyan, 2015-10-13:
That's right.
It has an Atmel ARM
This has been discussed about the future potential of Marlin beyond 8 and 16 bit MCU's

Grogyan, 2015-10-13:
I'm getting one, and will see how it goes.

paulusjacobus, 2015-10-13:
They say on the kick starter page that they have successfully ported Marlin.

Grogyan, 2015-10-13:
It's a really old port. and no mention of using it with OctoPrint

KiteLab, 2015-10-13:
If a bit older it's likely a fork of 
https://github.com/bobc/Marlin
if not that old likely a fork of 
https://github.com/Wurstnase/Marlin4Due

closed 2015-11-24 by AnHardt