MarlinFirmware/Marlin #2705: z_min_probe_endstop unexpected behavior and suggestions

sarf2k4, 2015-10-14:
Hi, earlier this morning I tried using the 1.1.0 rc2 with the separate probe and endstop enabled, I also disabled the z safe homing feature and the first results were great, as expected that the head would go out of the way and hit the min endstop. But at some point, the g28 would go wrong and would crash the z axis. I had times where the z axis crashed to the printed couplers  even though z min endstop were hit, the motor would just kept going

I modified the pin assignment, the z min endstop would go to 19, probe at 18. the probe were originally 32 but it was used for my lcd panel. The m119 shows it is good to go even triggering it with hand and check it again with m119. I attached the auto bed leveling snippet from mine, actually disabled the z_min_probe_endstop, don't think there's a help there. 

For the strange behavior, I got this since the dev 1.1.0, before moved as rc2 and currently I use the z safe home

As for my suggestion/opinion;
1. servo shouldn't be activated when g28 initiated if z min probe endstop defined
2. do check for the strange behavior
3. great idea because I'm able to fully utilize my max z travel up to 180mm
4. add an option for user to choose the sequence of g28, to home either Y or X axis first, because there are some build that put the clip on the x axis, to avoid head crash on the clips

I have video about the strange g28 behavior as proof but I'm going to upload it tomorrow

I'm unsure about the z probe offset affecting my head position, after g28 with z min probe endstop, initiating g29, it doesn't go up, rather just go to probe point, activate servo, marlin restarted due to power overload; the servo arm were obstructed because the z axis isn't high enough for clearance

//===========================================================================
//============================ Bed Auto Leveling ============================
//===========================================================================

// @section bedlevel

#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
//#define DEBUG_LEVELING_FEATURE
#define Z_MIN_PROBE_REPEATABILITY_TEST  // If not commented out, Z-Probe Repeatability test will be included if Auto Bed Leveling is Enabled.

#if ENABLED(AUTO_BED_LEVELING_FEATURE)

  // There are 2 different ways to specify probing locations:
  //
  // - "grid" mode
  //   Probe several points in a rectangular grid.
  //   You specify the rectangle and the density of sample points.
  //   This mode is preferred because there are more measurements.
  //
  // - "3-point" mode
  //   Probe 3 arbitrary points on the bed (that aren't colinear)
  //   You specify the XY coordinates of all 3 points.

  // Enable this to sample the bed in a grid (least squares solution).
  // Note: this feature generates 10KB extra code size.
  #define AUTO_BED_LEVELING_GRID

  #if ENABLED(AUTO_BED_LEVELING_GRID)

    #define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 145
    #define FRONT_PROBE_BED_POSITION 20
    #define BACK_PROBE_BED_POSITION 150

    #define MIN_PROBE_EDGE 10 // The Z probe minimum square sides can be no smaller than this.

    // Set the number of grid points per dimension.
    // You probably don't need more than 3 (squared=9).
    #define AUTO_BED_LEVELING_GRID_POINTS 3

  #else  // !AUTO_BED_LEVELING_GRID

      // Arbitrary points to probe.
      // A simple cross-product is used to estimate the plane of the bed.
      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 180
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 170
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID

  // Offsets to the Z probe relative to the nozzle tip.
  // X and Y offsets must be integers.
  #define X_PROBE_OFFSET_FROM_EXTRUDER -46     // Z probe to nozzle X offset: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -14     // Z probe to nozzle Y offset: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -16.95  // Z probe to nozzle Z offset: -below (always!)

  #define Z_RAISE_BEFORE_HOMING 23       // (in mm) Raise Z axis before homing (G28) for Z probe clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case.

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min.

  #define Z_RAISE_BEFORE_PROBING 23   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 5    // How much the Z axis will be raised after the last probing point.

//#define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10" // These commands will be executed in the end of G29 routine.
                                                                            // Useful to retract a deployable Z probe.

  //#define Z_PROBE_SLED // Turn on if you have a Z probe mounted on a sled like those designed by Charles Bell.
  //#define SLED_DOCKING_OFFSET 5 // The extra distance the X axis must travel to pickup the sled. 0 should be fine but you can push it further if you'd like.

  //If you have enabled the Bed Auto Leveling and are using the same Z Probe for Z Homing,
  //it is highly recommended you let this Z_SAFE_HOMING enabled!!!

  #define Z_SAFE_HOMING   // This feature is meant to avoid Z homing with Z probe outside the bed area.
                          // When defined, it will:
                          // - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
                          // - If stepper drivers timeout, it will need X and Y homing again before Z homing.
                          // - Position the Z probe in a defined XY point before Z Homing when homing all axis (G28).
                          // - Block Z homing only when the Z probe is outside bed area.

  #if ENABLED(Z_SAFE_HOMING)

    #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
    #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).

  #endif

  // Support for a dedicated Z probe endstop separate from the Z min endstop.
  // If you would like to use both a Z probe and a Z min endstop together,
  // uncomment #define Z_MIN_PROBE_ENDSTOP and read the instructions below.
  // If you still want to use the Z min endstop for homing, disable Z_SAFE_HOMING above.
  // Example: To park the head outside the bed area when homing with G28.
  //
  // WARNING:
  // The Z min endstop will need to set properly as it would without a Z probe
  // to prevent head crashes and premature stopping during a print.
  //
  // To use a separate Z probe endstop, you must have a Z_MIN_PROBE_PIN
  // defined in the pins_XXXXX.h file for your control board.
  // If you are using a servo based Z probe, you will need to enable NUM_SERVOS,
  // Z_ENDSTOP_SERVO_NR and SERVO_ENDSTOP_ANGLES in the R/C SERVO support below.
  // RAMPS 1.3/1.4 boards may be able to use the 5V, Ground and the D32 pin
  // in the Aux 4 section of the RAMPS board. Use 5V for powered sensors,
  // otherwise connect to ground and D32 for normally closed configuration
  // and 5V and D32 for normally open configurations.
  // Normally closed configuration is advised and assumed.
  // The D32 pin in Aux 4 on RAMPS maps to the Arduino D32 pin.
  // Z_MIN_PROBE_PIN is setting the pin to use on the Arduino.
  // Since the D32 pin on the RAMPS maps to D32 on Arduino, this works.
  // D32 is currently selected in the RAMPS 1.3/1.4 pin file.
  // All other boards will need changes to the respective pins_XXXXX.h file.
  //
  // WARNING:
  // Setting the wrong pin may have unexpected and potentially disastrous outcomes.
  // Use with caution and do your homework.
  //
  // #define Z_MIN_PROBE_ENDSTOP

#endif // AUTO_BED_LEVELING_FEATURE

sarf2k4, 2015-10-15:
Here is the video, I wonder if it is still being processed or not, hopefully can be viewed
https://www.youtube.com/watch?v=P1LtJsepTgs

NarimaanV, 2015-10-15:
It's still processing and it's been 13 hours now, I think something might be wrong with it...

sarf2k4, 2015-10-16:
I am sorry, it turns out to be the original file cannot be rendered on my pc and I deleted the source as well from my phone's camera.

I try to describe the problems I faced, all of these below are with Z MIN PROBE ENDSTOP enabled and disabled Z SAFE HOMING then use g28 or auto home from lcd menu.

1. it goes well by using the old endstop making the head parked outside of XY plane and homes z axis, but initiates servo movement

2. it tries to home X Y axis at the same time tries to raise the Z axis, making the movement of raising the Z axis stutters (no steps skipped), more like it is raised with several pauses along the way, then it descends, at this point, manually triggering min endstop usually ignored, even though the z axis are physically triggering the endstops, it ignores and continues ignoring triggered state

3. In some cases, the z axis ascends indefinitely without end and this shouldn't happen

So far I got these 3 scenarios only, it is rare to get the first scenario. The first scenario is working but shouldn't initiate servo movement. 

I believe there's something wrong with the code for homing though

thinkyhead, 2016-03-02:
@sarf2k4 Sorry for long neglecting this topic. Are you still seeing these issues in the latest code (`RCBugFix`)?

sarf2k4, 2016-03-11:
@thinkyhead, give me several days to test out the latest rcbugfix code, along with the z axis not stopping issues on 1mm increment

thinkyhead, 2016-03-13:
> z axis not stopping issues on 1mm increment

I see this is still lingering (#3134) but can't reproduce the behavior. Maybe it's fixed in `RCBugFix` or maybe it's only on certain types of machines – delta perhaps.

sarf2k4, 2016-03-14:
I did a preliminary test before using the rcbugfix branch and using the 1mm increment on z axis would result marlin to crash/reset automatically when large travel imposed for example after homed, z at 0 moving straight to 170, marlin would crash

As for now in rcbugfix, i dont see this particular problem yet, will report back if similar thing came up

sarf2k4, 2016-03-14:
Preliminary test when this option enabled so far it is still okay, the z min registered properly when manually triggered during homing (this to avoid potential crashes). 

I still think that it is better not to trigger the servo if a dedicated z min probe is enabled because having servo activated, there are no means to trigger the switch on the servo arm, unless the z min probe are disabled then servo would be activated (assuming there are dedicated plate to trigger the endstop outside the bed)

closed 2016-04-28 by jbrazio