MarlinFirmware/Marlin #2708: SCARA imnplementation is complete brocken

morriswinkler, 2015-10-15:
I think no one has ever tested that feature, look at that 

G28 
M114
Response...

ok
X:0.00 Y:0.00 Z:100.00 E:0.00 Count X: -30199460.00 Y:-30199460.00 Z:100.00
SCARA Theta:0.00   Psi+Theta:0.00
SCARA Cal - Theta:0.00   Psi+Theta (90):0.00
SCARA step Cal - Theta:0.00   Psi+Theta:0.00

so Count X + Y is completely off, i think the error is somewhere in the G28 routine or the conversion from steps per mm

KiteLab, 2015-10-15:
Whow - a SCARA issue! The first one i have seen since i read this issue tracker.
That does not mean SCARA is free of errors, but tells a story how much this code is used.
As far as i know non of the regular contributes has a SCARA machine - so tests are unlikely.
At least two of the regular contributors have build, or are building their, the more and more popular, delta machines. Don't expect them to build a SCARA soon.  
That's your chance to become the Marlin-SCARA hero. Try to fix it. We'll try to give you as much support as we can.

morriswinkler, 2015-10-15:
well scara is completely useless in the current release,

i am still tracking the problem , in the moment i am at prepare_move()

https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Marlin_main.cpp#L4044

will return always 0.00, any idea why that could be, anyway it looks like the SCARA code was just pulled, without ever testing it, so i guess the code was pulled from an older version of Marlin and stuff had changed meanwhile, especially G28 routine has changed dramatically, it might also be brocken for SCARA, i have to say, the code style in Marlin is everything but readable, it is difficult to debug the code while reading it, looks like that the firmware needs a major rewrite to clean up stuff.

KiteLab, 2015-10-15:
If you are beginning to analyse the code, don't start with the `release` branch. I'd suggest to begin with `RC` - the code, especially in G28, G29, changed dramaticly. Everything you could suggest to improve `release` will be difficult to port to one of the newer branches and has no chance to be integrated in the current `release`.

AnHardt, 2015-11-24:
I'll label this as potential bug. Non of the regular contributors has a SCARA printer to verify.

thinkyhead, 2016-02-22:
I would like to see this addressed before 1.1.0 release, if possible. Let's start by looking at all the code customizations that happen when `SCARA` is enabled…

closed 2016-04-23 by jbrazio