MarlinFirmware/Marlin #2711: LCD bug with Marlin Version

mibeauvais, 2015-10-17:
Hello,
I use two different version of Delta Marlin for my printer, a version without the autobed leveling and the other with autobed leveling.
The version without autobed leveling work properly.
The version with autobed leveling works correctly, but generates a bug on the display screen.
[image]

I do not understand what is this default.

Version name Marlin4 lcd works withou autobedleveling:
https://www.dropbox.com/s/kj67whqt8i4wqjl/Marlin4.7z?dl=0
Version name Marlin_3 lcd not works with autobedleveling:
https://www.dropbox.com/s/rs02z7ogyc5ni52/Marlin_Version_3.7z?dl=0

Thanks for help me.

KiteLab, 2015-10-17:
The main difference is, your 3.7 is not from here, but an ancient branch.

mibeauvais, 2015-10-17:
Yes, but in this old version it's possible to use the "Bed Auto Leveling" for printing DELTA, in the current release it is noted "Bed Auto Leveling is still not compliant with Delta Kinematics".

NarimaanV, 2015-10-17:
I've had that glitch happen before to me too. Try just clicking your encoder when that happens and see if that fixes the screen (that worked for me).

mibeauvais, 2015-10-18:
It does not work for me, I clicked in the encoder, the screen turns blue, after is display error.

NarimaanV, 2015-10-18:
Huh, well, seems to be more of a serious issue than I had :/

mibeauvais, 2015-10-23:
now,I used the Marlin RC 2

Roxy-3DPrintBoard, 2015-10-23:
And is it sick with RC-2 ?     Is that a LCD2004 display?   I'm using a LCD2004 with RC-2 and I don't have any gibberish on the screen.   We do have some people here that know a lot about the LCD Panels.   Maybe one of them will jump in and help.

mibeauvais, 2015-10-26:
Hello Roxy, No it work very well with the RC2, I'm out of default now.
But I can not run the auto bed leveling, when I send G29, I receive "G29 PROBE LEFT POSITION OUT OF RANGE"

Roxy-3DPrintBoard, 2015-10-26:
What that message means is you are trying to probe a position on the bed that is too close to the edge.  It will force the extruder nozzle to a position where it is not allowed to go.    How this number gets calculated depends on what kind of printer you have.

But for a Delta...  You might want to go into Configuration.h and increase the

#define LEFT_PROBE_BED_POSITION number 

But do that with the understanding that these #defines have to be reasonable:

    #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define MIN_PROBE_Y (max(Y_MIN_POS, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #define MAX_PROBE_Y (min(Y_MAX_POS, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER)

mibeauvais, 2015-10-26:
Thank you, auto bed leveling works, but I have a lot of vibration motor during auto bed leveling.

mibeauvais, 2015-10-26:
I lowered a little movement speed, there is more vibration to motor, they are glossy when ascending  from auto bed leveling.

Roxy-3DPrintBoard, 2015-10-26:
> I lowered a little movement speed, there is more vibration to motor, they are glossy when ascending from auto bed leveling.

When you say "They are glossy" do you mean they move very smoothly?

The Auto Bed Leveling (G29) uses do_blocking_move_to() in the run_z_probe() function.  It does not do the extra work that happens during a print to make the movement smooth.  How much more vibration do the motors have during Auto Bed Leveling?   My guess is what you are seeing is not a  big deal.  Posting a video of the vibration you are seeing would give people an opportunity to see the same thing you are saying.   Is that possible to do?

mibeauvais, 2015-10-27:
With printing I don't have see these vibrations, or they were less.
If you see the video:
https://www.youtube.com/watch?v=d04ubwlaTng

thinkyhead, 2016-02-07:
How's it going? Can we close this issue? Be sure to try 1.1.0-RC3 if you can. The updates may or may not impact your vibration situation, but it would be good to know. I believe the reason for the difference in the way movements "feel" is that bed leveling uses raw movement between probes so the timing and coordination between the steppers is different. With raw moves, the stepper speeds stay constant throughout the move, whereas with interpolated moves the stepper speeds constantly change. _Just a theory…_

closed 2016-04-23 by jbrazio