MarlinFirmware/Marlin #2714: Thermal Runaway RC 1.1

psavva, 2015-10-17:
Hi, 

I have installed Version RC 1.1, and have been getting issues with Thermal Runaway.
I was able to print before just fine with the latest stable.

What can I provide to help diagnose if there is a bug, and how to fix it.

Roxy-3DPrintBoard, 2015-10-17:
I went through the same thing...   I think the parameters are set tighter than they need to be.

In Configuration_adv.h I changed these lines:   If you make the same changes it will probably relax the checks enough you don't have any problems and can leave the Thermal Runaway stuff turned on.

~~~~
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
~~~~

ruggb, 2015-10-18:
i would be interested to see what your temp curve looks like. I had a similar issue and the temp was all over the place. Mine is now w/i -/+1°. My issue was fan wires bundles with thermistor wires injecting lots of noise.

Roxy-3DPrintBoard, 2015-10-18:
I do have a set of fan wires bundled in with the wires that have the termistors on them.   But I was watching the numbers.   If I have the extruder turned on and stable, and then turn it off for 15 seconds, I'm watching the temperature drop in real time and a constant rate.   When I turn the extruder back on, I get to watch the temperature continue to drop.   It takes time for the heat to get from the heater, through the metal and into the thermistor.

ruggb, 2015-10-18:
if u look at the temp curve (I am assuming u r using Repetier - but maybe whatever u r using has a similar display) u will see that the temp jumps around and u see spikes on the curve.
That is noise from the fans. Remove the fan wires from the bundle and look at it again.
It should come out very stable (assuming u have done the PID thing) within +/- 2° or better and ur problem will go away. What u describe is just the way it operates and has nothing to do with ur thermal runaway issue. I had some pics of mine but I have no idea where i saved them. If I find them I will post as they r interesting.

ruggb, 2015-10-18:
I found them
This is not good! That noise appears to Marlin as an out of control temp. Approx 15° spike will cause a shutdown. PID was previously set.
[image]

This is what it looks like after removal of the fan wires. Same PID - though it probably can be improved with a little more damping.
[image]

My fan wires r no longer twisted with the therms but they r taped to the outside of the tube that contains the heater and therm wires.

Roxy-3DPrintBoard, 2015-10-18:
My wires are wrapped in a tight bundle with all of the other wires tightly together.   I'll keep an eye on it.   It will be a lot of work to get the fan wires out of the bundle.   I don't think I have a noise problem with my fan, but I'll watch for it.   If I do see something, I would be tempted to add a couple of capacitors right at the fan to dampen the noise rather than tear the bundle apart.

And my fan wires are not twisted which was a bad over sight!  Oh well.

ruggb, 2015-10-18:
Ya, I thought about that issue when I did it but did it anyway. Then forgot that I thought about it and spent a week trying to figure out why it was shutting down. I ran a substitute thermistor and couldn't believe how stable the temp was. The caps might work, but I bit the bullet and removed the wires. If the only other thing is the fans it would be simple just to run another pair of wires outside the bundle.
good luck.

psavva, 2015-10-21:
I'll test the if the wires bundles with the thermistor make any difference over the weekend in RC1.1.

I can confirm that I've tested this now on 1.0.2, and there is no issue with noise!
If switching my fan on and off multiple times during heating, and no issues.

Once I test with RC 1.1, I'll most the results

ruggb, 2015-10-22:
it wasn't the switching that was the issue for me. They were always on. They just produce a lot of electrical noise.

psavva, 2015-10-23:
Hi,

I've run a test now, and still am getting a thermal runaway.
I have set 

 #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius

[image]

Roxy-3DPrintBoard, 2015-10-23:
Is it the Red line that shows your Extruder's temperature?   Are you saying you are getting Thermal Runaway towards room temperature?

psavva, 2015-10-24:
The red line is the Extruder.
The Y Axis is the Temperature.
The X Axis is the Time.

You'll notice that the Temperature was around 200 degrees Celsius. shows that it dropped 20 degrees, and the thermal runaway is triggered, bringing it down to zero.  

For some reason, it shows a drop in 20 Degrees Celsius from 30:30 to around 31:05.

3machine, 2015-10-25:
I also had thermal runaways, i fixed it putting non conductive thermal paste around the thermistor. now there is less lag when mesuring.

Roxy-3DPrintBoard, 2015-10-25:
> I also had thermal runaways, i fixed it putting non conductive thermal paste around the thermistor. now there is less lag when measuring.

I did the same thing on my first printer.  I thought I was being clever.   It did help for a while.  The problem is at the temperature the nozzle runs even super high quality thermal paste dries out.  It andes up kind of like Black Board Chalk.   And it stops conducting heat very well.

CONSULitAS, 2015-10-25:
@psavva If you look at the red curve, you see, that the temperature drops and then rises again. As you see in the lower curve (green block), there is no heating causing this rise.

So there is no software problem, but a hardware problem instead. You have either a) a bad electrical connection to your thermal probe on the hot end or b) a bad mechanical contact (for heat conduction) between probe and hot end.

Check every electrical connection twice.

Make sure that the thermal probe doesn't move by the vibration. Mine dropped out of the hot end some time ago while printing! For me it worked to glue it with glue for ovens (good up to 1000 degrees centigrade).

Good luck

psavva, 2015-10-25:
From a software development point of view, perhaps the parameters detecting the thermal runway are too sensitive, or somewhat unrealistic. I assume the reason for a thermal runaway protection is to avoid excessive uncontrollable heating, causing fires. 

If thermal runaway is being detected for such small variances, perhaps it should be adjusted in the main Marlin branch to avoid constant false positives...

psavva, 2015-10-25:
@CONSULitAS, I never considered that. I will use thermal paste, and monitor it again, and post the results.

Thank you

CONSULitAS, 2015-10-25:
@psavva You are welcome.

>From a software development point of view, perhaps the parameters detecting the thermal runway are too sensitive, or somewhat unrealistic. I assume the reason for a thermal runaway protection is to avoid excessive uncontrollable heating, causing fires.

>If thermal runaway is being detected for such small variances, perhaps it should be adjusted in the main Marlin branch to avoid constant false positives...

No, the parameters are not to sensitive. A physical system like your hot end never makes a "jump" in temperature down and up like in your graph. Sense of thermal runaway protection ist to sense this jumps, because they are a indicator for a problem with temperature measurement.

Imagine:  Your marlin sees a falling temperature and switches the heater on and doesn't stop heating. What will happen? The hot end will get really hot and your filament could start burning...

People reported melting metal blocks and nearly burnt houses. With a physical problem on your hot end (or a 230V heat bed) you and you children can die!

This is the reason for sensitivity. Safety first!

ruggb, 2015-10-25:
@psavva - there may be 2 issues here - urs and the feature program.
consider that I may not be interpreting the operation of the program correctly.

I am not sure u r looking at the actual cause of the shutdown
ur parameters r set for 20sec and 2°
if u look @ time from 31:00 to 31:15 there is a 2°+ rise there - it isn't the fall that is triggering it
but ur ? has to be why is it falling in the first place. the target is remaining the same.
The time duration of the rise and fall appear that it is real and not from noise.
but then why didn't the previous rises @ 29:00 to 29:15 and 29:50 to 30:05 trigger a shutdown.
Maybe someone can post a clear flow description of the operation or point to one.

However, if the operation of the feature resets the base point to a number below the set point and causes a trigger at a point between that point and the set point I would think that is a problem with the feature operation. The system is trying to recover from something that dropped the temp and the feature is triggering a fault. It should only be looking at temp rises above the set point.

In my case, with the noise, it was looking at a transient that couldn't possibly be a temp rise so it might be good to filter out things that r not possible - but then I wouldn't have found my problem.

Because of the slow response to the output of the extruder after 30:30 (it appears to be full on and nothing is happening to the temp) I might start looking for a high resistance connection on the hotend 12V line. Just feel for warm spots on the wires.
That non-rise may also be what is triggering shutdown.

ghuser2, 2015-10-26:
I set mine to 40 seconds, and 30C, because when grabbing the nozzle with pliers (to change nozzles) the temp drops 25C. But before that, I was getting shutdowns just while calibrating, so it was too sensitive.

AnHardt, 2015-10-26:
The correct parameters to tune, if you get a "Thermal Runaway" error are
```
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius
```

Tune the parameters
```
  #define WATCH_TEMP_PERIOD 16                // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
```
for a "Heating failed" error.

Your diagram shows a drop off temperature by ~20° at ~30:30. Until ~31:10 (40 seconds later) the temperature has not reached the 4°C margin around the given temperature again. The error is triggered, the machine shuts down. That is exactly the expected behavior.

Find out what causes, and avoid, the (unusual) 20°C drop. If you cant avoid it, tune the (right) parameters.

ruggb, 2015-10-26:
@AnHardt 
so what u r saying is that the temp dwell @ 180° for over 20 sec is what caused the shutdown.
Since the heater was on and it expected to see the temp rising toward 200°.
that makes sense.

I would be interested to learn why the temp didn't start to change for so long. It did finally start to rise, but too late I guess.

Roxy-3DPrintBoard, 2015-10-26:
Well...  It is hard to say without actually having the hardware and some test equipment attached to it.  But if there was a bad electrical connection to the heater but it was not totally disconnected, you could get that kind of behavior.   For example, think about a stranded wire twisted to another wire.  If just one strand of the stranded wire was making electrical contact, you would have a lot of resistance and even though the heater is getting power, it would not be getting enough power.

And remember...  with that much power and heat, the wires do corrode.  It is possible to have wires twisted together but not making perfect electrical contact.

(I'm just guessing...  I don't know what is going wrong!!!!)

arthur500, 2015-10-27:
Hi all.
Beware of misinterpreting the Repetier-Host temperature graphs.  I was just about to jump in to agree with @psavva as I was getting the same sort of apparent large temperature glitches (using 1.1RC2).  However, what I found was that in long running operations, Marlin appears to not return temperature values to R-Host in response to M105 (I guess).  So the R-Host graphs freeze until the operation finishes and then RH simply interpolates the gap.  On particularly long operations such as the mysterious auto bed levelling (45-60 secs) or just long Z travels you can see the freeze.  (Just how R-Host calculates power, I have no idea, as it has no direct way of knowing what Marlin is actually doing.  I guess it must be based on temperature gradients etc but I would not base anything on it).
Anyway to be certain, I mounted the thermocouple of my DMM to the heater during all these operations.  I can confirm the temperature does not change in unexpected ways regardless of the R-Host graphs. If there is a thermal runaway such as the sensor drop to zero as in the graph of @psavva I would backup the comments of others that this is a bad electrical connection on the temperature sensor given the sudden drop and return.  Previously I've encountered a poor thermistor connection in a connector which gives same indications as this.  Your heater would likely have a thermal mass too large to cool and reheat in 15 secs if it was a bad heater connection.  The zero reading is another indicator that it is a sensor electrical problem not a physical mounting issue.
@Roxy-3DPrintBoard is correct about the thermal paste drying out.  It's only meant for max temps half that of your hotend.  I use 6mm wide kapton tape under a thermal sleeve to bind the thermistor to my old hand wound hotends and never have an issue.
Regards.

AnHardt, 2015-10-27:
@arthur500 
There is no interpolation in the temp. graphs in Repetier host. It's just connecting the known points with a straight line.
The power information is transmitter with M105 behind the '@'s.
The downward line on the big sawtoth is drawn between the time the error occurred and the reset  what is needed to continue. The fist 0°C reading after the reset is meaningless (before the temp system is initialized properly). After that the readings are normal, as expected, again.

An interesting detail in the graph is the difference of slope between 30:30 - 30:40 and 31:45 - 32:00. In the first interval the slope is much higher then in the second. In the second interval, after the reset, i expect the system to cool down in in free air. So for the first interval i expect active cooling by ether a fan, or touching the nozzle with something cold.

@psavva 
What hotend do you use?

ghuser2, 2015-10-27:
Wow I did not know there were more runaway settings in config advanced. How does it differ from the runaway setting in config.h ?

AnHardt, 2015-10-27:
@ghuser2 
The difference is not that big. In the newer branches the settings are moved from Configuration.h to Configuration_adv.h and a bit renamed. In Configuration.h you now can only switch the `THERMAL_PROTECTION_HOTENDS`, `THERMAL_PROTECTION_BED` on and off.

The new settings are explained in Configuration_adv.h
```
  /**
   * Whenever an M104 or M109 increases the target temperature the firmware will wait for the
   * WATCH_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M104/M109,
   * but only if the current temperature is far enough below the target for a reliable test.
   */
  #define WATCH_TEMP_PERIOD 16                // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
```

arthur500, 2015-10-28:
@AnHardt Thanks , I understand your explanation. Also I was completely surprised that power info is returned along with temp from M105 (I know you're saying .... just look at the code!). Learnt something new.
Regards

closed 2015-11-24 by AnHardt