MarlinFirmware/Marlin #2715: IMPORTANT: RC2: GCode SPACES should NOT be required (regression?)

gruvin, 2015-10-18:
I notice that having no spaces between commands in GCODE in Marlin 1.1 RC2 is no longer supported. This is bad. The GCODE spec allows for no spaces and so should Marlin. Marlin always uses to work without spaces, far as I know.

Examples:

G1Z10 -- this gets executed as simply, "G0", while the Z0 part is seemingly ignored.
G1 Z10 -- works as expected

M851Z-1.23 -- executes as "M851" with the "Z-1.23" parameter being ignored. 
M851 Z-1.23 -- works as expected.

dgoadby, 2015-10-18:
Well GCODE (RS274) does indeed say that whitespace is ignored. However,  I for one encourage spaces for clarity. Most of the GCODE generators seem to add spaces too. 

If Marlin is supposed to be fully RS274 then you are right in requesting the reinstatement of the feature. If spaces are going be mandatory from now on then the diagnostics should report as appropriate. 

As my GCODE will still work in 1.1 RC2 I'm not fussed either way but, I do think that some clarification/statement would be a good idea as some users might be caught out.

Roxy-3DPrintBoard, 2015-10-18:
For sure...  Spaces should be 'tolerated'.    ThinkyHead just published a bunch of changes for RC2 and I added a request to change the code_has_value() function.   Right not it isn't behaving consistent.  It only works if you have no space between the GCode command letter and the paramter (if it is there).

I requested the front end of the code look like this:

~~~~
bool code_has_value() {
  int i = 1;
  char c = seen_pointer[i];
  
  while (c == ' ')		// skip past any white space between GCode and number
     c = seen_pointer[++i];
~~~~

gruvin, 2015-10-20:
Well, of course space should be both tolerated *and *supported and I agree
that it makes for more readable code -- naturally.

But my point is the exact opposite, namely that NO SPACES _must_ be
tolerated. They used to be but now are not.

Bryan.

gruvin, 2015-10-20:
Hmmm ... my last comment was still ambiguous. Let me restate for clarity ...

We must be able to enter GCODE *with no spaces present *and have it
executed correctly. That used to be the case, right up until RC2, afaik.

That is, spaces are no longer simply optional or tolerated. They are
presently mandatory. This is not good.

Bryan.

gruvin, 2015-10-20:
Here are the examples again, this time with pre-formatting to hopefully make things more clear ... 

`G1Z10`  -- this gets executed as simply, "G0", while the Z0 part is seemingly ignored.
`G1 Z10` -- works as expected.
`M851Z-1.23` -- executes as "M851" with the "Z-1.23" parameter completely ignored. 
`M851 Z-1.23` -- works as expected.

And here's a copy/paste from the OctoPrint terminal window ...

```
Send: M851Z-5.45
Recv: echo:Z Offset : -5.45
Recv: ok
Send: M851Z-5
Recv: echo:Z Offset : -5.45
Recv: ok
```

Note that the `Z-5 is IGNORED in the second send.

Roxy-3DPrintBoard, 2015-10-20:
Oh!    OK!   I see what you are saying!    And I have verified the difference of behavior between an old version of Marlin I have on one printer and the RC2 not doing any thing with  

~~~~
G0X0Y0
~~~~

This needs to get fixed and put back the way it was!

gruvin, 2015-10-22:
@Roxy-3DPrintBoard ... thanks for the verification and quite so. Thanks!

I know what it's like. Someone comes along and tells us about some bug that's so unlikely as to be unbelievable ... unless of course, it's not. hehe

Roxy-3DPrintBoard, 2015-10-27:
A quick question to everybody:   These routines used to not be sensitive to spaces being between numbers and/or parameters.   code_has_value() is a new function so this thought doesn't apply to it.
  
But for code_value() and code_seen() should we just revert what ever changes got made to those routines and get rid of this bug?   Or should we actually 'fix' the issue and do a Pull Request?

Wackerbarth, 2015-10-28:
Don't revert. Although a clean commit stream should not have entries that "undo" other entries, the way Github handles merging, a revert is just another merge. Particularly if there are any other changes in the commit that you are reverting, it creates more of a mess than if you just do a normal pull request to change from the current situation to the desired one.

thinkyhead, 2016-02-10:
Does this issue still exist? Has a patch been submitted to fix this issue? I can see nothing in the current code that would lead to spaces having any bearing on the result of `code_seen`, `code_value`, `code_value_long`, or `code_value_short`.

thinkyhead, 2016-02-10:
Actually, scratch that. I see that the code won't tolerate no space after the initial command code:
```cpp
  current_command_args = current_command;
  while (*current_command_args && *current_command_args != ' ') ++current_command_args;
  while (*current_command_args == ' ') ++current_command_args;
```
…and this should be changed to…
```cpp
  current_command_args = current_command;
  while (*current_command_args >= '0' && *current_command_args <= '9') ++current_command_args;
  while (*current_command_args == ' ') ++current_command_args;
```
Patch coming shortly!

Roxy-3DPrintBoard, 2016-02-10:
Correct....   code_has_value() needs to be like:

~~~~
bool code_has_value() {
  int i = 1;
  char c = seen_pointer[i];
  while (c == ' ')  // skip past any white space between GCode and number
     c = seen_pointer[++i];
  if (c == '-' || c == '+') c = seen_pointer[++i];
  if (c == '.') c = seen_pointer[++i];
  return (c >= '0' && c <= '9');
}
~~~~

closed 2016-04-23 by jbrazio