MarlinFirmware/Marlin #2718: You have to feed the troll....

justmyopinion, 2015-10-19:
In version RC1 it was reported that display rate was too high for the printer and RC2 2 was born with a reduced update rate from 10 Hz to 1 hz (I think) due to reports on "stuttering printers.
However the problem that was reported has not neen resolved, which was that at "higher" printing speeds a lot of people reports that printers are stuttering.
i decided to dive into this and I have some conclusions on this that should be able to remove the problem for a while, I hope :-),
The printer FW is pure (almost) sequentiel which means that any event in the printing process has  some influence on the behaveour.
If we take away that printer has some hw to be serviced (User, endstops, temperatures, fans etc...)
the dataflow is very sequentiel, here very simplified:
Hostcmd->serielTX->serielRX->serielbuffer->cmdtranslator->printerservice->cmdbuffer->steppermotors.
The input from host has a fixed speed set by the baudrate which (my printer) is 250000bit/s which is 25000 chars/s (40uS) which is the first limit. printer cannot go beyond that speed. However commands could be from a few to many (5-20) chars so the command rate could be app 10 chars/cmd whic is 400us/cmd. This is the highest rate that cmds could do average due to the host feeding.
Cmd's are stored in command buffer which the printer has to translate to stepping rates which is taking care of in the next step in printer. However if printer should not service user or display or blinking lamps this could be simple so the FW has a point between stepper commands issued and buffer translations that is called idle time.where printer is serviced by displaying temperatures, buttons etc while stepping the motors is done in parallel This service is somehow a limit to the printer data flow as the service takes time and while the printer is servicing, no data is flowing from cmd buffer to the stepper cmd buffering.
If this service (delay) is too big the stepper cmd processing is starving as no new stepper cmds are flowing into the process due to the resources used (display) and the printer is "stuttering" as it is forced to halt between command and of course this is  dependant on display delay and stepper frequency (mms/s selected).
To make it short i have verified that if you take away the delay (no display or updates of buttons) the printer is printing smooth and very fast but if you have to display etc your printer is stuttering at some speed, depending on item printed. I have selected a cylinder as test object which typically has many very short commands as circle is printed as many short lines and this object is very easy "stuttering" at some higher speed.
And how do we remove this "stuttering".
Well there is a fine solution and a fast solution...
One solution is that the "delay causing this" for the most it is display so remove the display updating and stutter disappears (try it). A compromise is to reduce the time it takes to update display. By some refinement of drivers I could reduce the update from 100-150ms to 10-15ms and the problem would only show above 100mm/s (if that is a problem). This include also having some TWI rates replaced fromm 100kHz to 4-800kHz which can be made more or less without big changes.
A more sophisticated solution would be to have a "background process" tranferring cmds from input buffer to stepper buffer while display etc is done in foreground. I personally prefer this as the optimum solution as being transparent to the process. 
This could be made by some interrupt processing doing cmd translation/transfer (millis) or similar (Timer2 is available) or it could be synchronized by stepper-cmd-buffer empty somehow but it is vital to be interrupt processed as it needs to be independant from foreground updates.
i have not made the coding needed for this as i would think some nerds may have more experience on implementing this strategy than me, but I am convinced that this is what is needed to remove any more talk of "stuttering".
My printer is at the moment running "smooth as silk" and almost sound less at high speeds only by reducing display updates and a few things down to a few ms, which unfortunately is needed to get smooth at high rates with current implementation (RC1/RC2).
I hope you forgive me this long talking smooth :-)

Roxy-3DPrintBoard, 2015-10-19:
> A more sophisticated solution would be to have a "background process" transferring cmds from input buffer to stepper buffer while display etc is done in foreground. I personally prefer this as the optimum solution as being transparent to the process. 

Moving that much logic to a 'Background Process' when we don't even have a Real Time OS to build on is going to be problematic.  

I think we should (as a team) verify what you are saying.  I suspect your assertions are not far off from what the developers already believe.

Assuming we get consensus, my vote is we look at what the LCD code is doing and how it does it.   I believe it can be stream lined so that it does not use up so much of the processor.   I did look at the LCD libraries that get used and we can speed those up.  And probably, we don't need to update the entire LCD screen every time we update it.  It might make sense to just update what has changed because I suspect we can cut out a lot of activity by doing that.   It will be a good discussion to have.

justmyopinion, 2015-10-19:
@Roxy-3DPrintBoard 
Maybe my explanation was too long, but i already did the work you'r suggesting.
1:I rewrote the TWI2 drivers to be faster (2x), i installed another Wire library I2c replacing default Wire library to remove the interrupts Wire is doing/cmd. 
2; i raised the TWi frequency from default 100kHz to 800kHz which is the max my installation will allow
3: i split the display update in four updates. 1 line/update (mine is 20x4)
My update of display was reduced from 100-150ms/update to 2.5-3ms/update. when printing.
The printer is still stuttering above 100mm/s when stressed. if you think you can do it better please go ahead.
So you see that is why i am convinced that there is only one solution: the printer data flow MUST have the highest priority so any slow updates MUST be in background and the weak link at the moment is the transfer from input buffer to stepperbuffer including processing.

Roxy-3DPrintBoard, 2015-10-20:
> 1:I rewrote the TWI2 drivers to be faster (2x), i installed another Wire library I2c replacing default Wire library to remove the interrupts Wire is doing/cmd. 

> 2; i raised the TWi frequency from default 100kHz to 800kHz which is the max my installation will allow

I did see the discussion about increasing the speed of the I^2 communications and the improvement on the processor overhead.  I did not see any actual code to look at to see how that was done.   Is it possible to provide that code so people can look at it and understand the specifics of how that was done?

>3: i split the display update in four updates. 1 line/update (mine is 20x4) My update of display was reduced from 100-150ms/update to 2.5-3ms/update. when printing. 

That is very impressive.  Can you provide a way to look at this code?  I did look at https://github.com/justmyopinion/Marlin and the most recent update I see is 18 days ago and affect sthe Watch Dog Timer.

My thoughts about the display update were mostly aimed at how the u8g library does its work.  I was thinking if the library was made more efficient it would be possible to save some amount of processing by the AVR microprocessor.   But if you have the display update reduced to 2.5 to 3 ms per update, it may be the u8g library doesn't even need to be looked at.

> The printer is still stuttering above 100mm/s when stressed. if you think you can do it better please go ahead.

I would first like to see your code and see what it accomplished!   Because if you reduced the display update to 2.5 to 3 ms, that may not be the source of the problem.  It would be good to hear from other developers on this!

justmyopinion, 2015-10-21:
if you use u8g library you may have quite different problems than decribed above and you would not gain anything with the updates done.
You have to understand that it is not all about optimizing drivers here as decribed above, that was just a tool for me to get to the understanding of the "problem".
It is about delays in the "foodchain" for the printer and LCD updating is just a visble symptom for the actual problem but not the real problem.
The best way to see that this is how it is, is to replace the LCD updating (lcd-update() with a delay (delay(100ms) or similar. The printer is still working but now you can simulate the lcd update time taken. You will now see that printer is stuttering at certain speeds with certain delays and it gets worse with higher delays, so this is what needs to be solved. i removed a symptom by reducing the update delay but the printer is still suffering from a bad design unfortunately and the challenge is to find out how and the "problem" is not solved before this is accomplished.

Roxy-3DPrintBoard, 2015-10-21:
OK!  I understand your point.

justmyopinion, 2015-10-22:
i intend to close this one as the conclusions I made on the above tests has led me to make a pullrequest for changes that solves most of the above noticed problems. it is also necessary that obvious delays besides that are kept to a minimum such as display updates etc.

atunguyd, 2015-10-23:
To be honest for a while now I have wondered why Marlin is not running ontop of something like FreeRTOS which has been successfully ported to the arduino platforms.
Seems the entire system would be much simpler if it was multitasking.

justmyopinion, 2015-10-23:
@atunguyd 
Well You have only one procesor even in RTOS, so you have to split the functions in marlin into timeslices that RTOS delegete, so the limits are about the same. You cannot have one "thread" taking the lot of processortime as RTOS would expect thread to be finished within time slice delegated.
There is a lot of "realtime" in marlin driven by Interrupt routines which you may consider a thread or proces that takes time from the rest of resources needed, so if you violate that timing in other "threads" needs processor at specific time windows, marlin would hang or crash somehow, which actually happens sometimes but often a simple hack has been added to make it work anyway.
If you can control these limited resources i would say that Marlin is not perfect but is actually very functional but there is still a need for improvement.
But adding RTOS would probably not solve anything and programming a RTOS system is not necessary simple.

Roxy-3DPrintBoard, 2015-10-23:
Certainly not everybody would agree with my perspective.  But if we had a RTOS in place I think it would simplify a lot of issues.   There are a lot of very time sensitive events happening in Marlin.  But a lot of the expensive things happening are not time sensitive.   

If just the LCD Panel stuff was broken out as a very low priority task that updated if and only if there were extra processor cycles available, that would help a lot.  I would also be tempted to run the task that fetches GCode commands at a varying priority.  If there were a couple of GCode commands already buffered up and waiting, I would probably have that task lower its priority to free up resources for the real work that was actually in process.  But maybe that isn't necessary...  After all, in a RTOS environment, if nothing is making the printer move it is because the printer doesn't have any commands to execute.  There would be lots of time for the command buffering task to go fetch some work to do.

A lot of interesting discussions would happen if we had an RTOS.

atunguyd, 2015-10-24:
@justmyopinion

Of course you don't assign most of your time to a single task, that not how an embedded RTOS system works. 

Currently the marlin firmware does a lot in the interrupt service routines which itself is not a sound design without a system with different priority interrupts. 

I have been designing various forms of payTV decoders for most of my professional career so perhaps I might be one of those "if the only tool you have is a hammer" types, but if I were to spec a system like the marlin firmware I would not dream of going single threaded.  In fact I must commend the marlin team for creating a single threaded system like this that actually works. 

Using an RTOS would allow proper decoupling of the various functions of the system ensuring that if someone for example made the UI prettier this would not impact on printing performance.

Wackerbarth, 2015-10-24:
@atunguyd -- There are two major problems attempting to use a RTOS in our environment.
Given that we have marginal computing power, the additional interrupt servicing overhead of a generalized RTOS can further reduce the computational capacity. But, perhaps the biggest problem is lack of RAM.  I do agree that we need to utilize some multilevel scheme to appropriately apportion the limited capacity that we do have.

closed 2015-10-30 by justmyopinion

bkubicek, 2015-10-30:
realtime os things are not capable of driving the stepper motors. The
granularity is just not cutting it.
Marlin is basically an operating system, that _already_ does realtime very
well, through real hardware timer interrups.
The problem is mainly, that graphical lcds and that do not mix, its a
design problem.

Bernhard