MarlinFirmware/Marlin #2721: [Request] Extruder babystepping

sarf2k4, 2015-10-20:
I think it would be great if there's also babystepping for extruder because during printing the skirts sometimes the plastic didn't get out in time upon printing the object's first layer

paulusjacobus, 2015-10-21:
Maybe I'm missing an obvious conclusion here but please can you explain to me how baby stepping is enhancing the plastic deposition? I have replaced my steppers with DRV8255 steppers in 32 micro-step mode. The difference is just the addtional current which gives my extruder just a bit more oompf. Any stepper mode makes no difference (direct drive) since the difference is almost negligible. For the X-Y axis it makes a huge difference (more fluent/smoother printing moves and less noise).

sarf2k4, 2015-10-21:
After heating up the hotend, the plastic oozes and of course we are going to remove that. But during the prints (after last one has been completed which I added filament retract a little bit on gcode), it oozes but then during deposition on the first layer, the plastic didn't came out. on small objects where the skirts is small even though I put extrude min of 5mm on skirt just to be sure, the plastic still didn't come out of the nozzle and might lead to bed adhesion problem when changed to second layer where the first layer didn't have any plastic deposited

My idea here actually is to speed up the extruder motor upon turning the knob so that the plastic will be deposited for the very first layer

I'm sorry, it is kinda hard to explain it, just that sometimes my plastic didn't come out in time before the object's first layer. I know this might be mechanical or filament issues, but this feature might come in handy where I can reduce the risk of failing where the first layer didn't have any plastic deposited

ZetaPhoenix, 2015-10-21:
I suggest lengthening the minimum skirt length in your slicer.

StepMan91, 2015-10-21:
you can also place on your gcode some mm out of platic before printing.

sarf2k4, 2015-10-21:
I did, but the plastic didn't come out immediately

Roxy-3DPrintBoard, 2015-10-21:
> I did, but the plastic didn't come out immediately

Well...  You just need it to come out eventually before the actual print starts.   I have Slic3r configured to do 10 mm of material for a skirt before it starts the actual print.   There are times when it just barely starts putting down material for the skirt before it switches to doing the print.

AnHardt, 2015-10-26:
I can't see any problem what is solved by a few extruder steps.  Use move-z instead.

sarf2k4, 2015-10-26:
So... if i move the z axis through babystepping then the plastic will be
extruded?

Roxy-3DPrintBoard, 2015-10-26:
Well, you don't need to baby step it.   I use PronterFace.  I just disable the motors and turn the extruder by hand.   If the reason the filament isn't coming out right away is because it dripped and ooozed out earlier...  There is lots of space inside the extruder that has to be filled.   And you can feel when it is full why you turn the gear by hand (assuming you have a plastic gear to turn).

brainscan, 2015-10-26:
I just put a line of gcode in my start routine to prime the extruder by the amount it retracts at the end of a print. The only time I have a problem is if I've been changing something and forget to manually prime it. Also like other people mentioned you can change settings in your slicer to extrude more before printing. It's just one of those annoying steps we all have to take when we commission a new machine to find which numbers work for you. I think this is better than having to keep watching the nozzle like a hawk and then trying to manually prime the nozzle whilst printing eveytime you print something.

AnHardt, 2015-10-26:
Sorry for the nonsens typo. I mend move-e in the move menu.

sarf2k4, 2015-10-27:
Yes, before print the nozzle did ooze when heated up, but after i remove
the ooze, then the extrude routine by 5mm, remove ooze again, plastic still
have some lags, didnt come out immediately

sarf2k4, 2015-10-27:
I mean during prints, i put a gcode to extrude 5mm of plastic first

AnHardt, 2015-10-27:
Usually one loop of skirt gives a perfect primed nozzle.

sarf2k4, 2015-10-27:
Took me more than one loop though

Grogyan, 2015-10-27:
I generally find that setting a skirt of 3 provides the right priming, I also extrude 5 mm once at temperature as well.  This is just common practice to me.

sarf2k4, 2015-10-28:
I strongly think that its based on the filament brand as well, on my side
it didnt come out at the right time, at slic3r i set to 5 loop, then 5mm
min on skirt, before printing extrude 5mm as well, so its a total of about
10mm extrusion before moving on to the objects first layer

brainscan, 2015-10-28:
I think it's more to do with your printer, I've never had that much trouble getting my nozzle primed. 5 skirts and 5 mm is a lot, how much oozing do you have? Ive changed between lots of filaments too and never had that much of a problem. I think it's probably a hardware problem on your side.

sarf2k4, 2015-10-28:
It oozes when the hotend reaches the target temp, extruder didnt move they
just oozes up to 20mm i think, at the end of print if i remember correctly
will retract around 5mm, if thats the case the nozzle shouldve been primed.
When the plastic extruded at the nozzle end theres no pop sound either nor
missing or unextruded plastic during the prints

brainscan, 2015-10-28:
If you're oozing 20mm of plastic you definitely have a problem, and if you think it's 20mm why aren't you asking it to prime by 20mm? I get maybe 1-2mm of ooze.

closed 2015-11-08 by AnHardt