MarlinFirmware/Marlin #2722: Delta G29 Autolevel does not work

mateddy, 2015-10-21:
Running 1.1-RC2 on Kossel Mini with Allen Key Probe (#define AUTO_BED_LEVELING_FEATURE and also #define Z_PROBE_ALLEN_KEY).

G28 command runs fine (all three carriages moves up).  On running G29, Z probe is deployed (marlin_main.cpp line 2876 deploy_z_probe(); ), but then hotend move like this:
1. to (x=0,y=-90) outside printable area, and then
2. moves UP, crashing the endstops, also to centre (x=0,y=0), and then
3. moves DOWN, crashing the platform.

As seen in #2713, if it is working, it should try to probe at 9 points.  Any hint about this issue?

Roxy-3DPrintBoard, 2015-10-21:
You are moving the nozzle outside of the  DELTA_PRINTABLE_RADIUS.  

If you change the front end of this function in Marlin_main.cpp you will get confirmation that is the problem.   The problem is when you move the nozzle outside of the printable radius you end up taking the square root of a negative number and pretty much the world ends for a Delta printer.  

I'm trying to figure out a nice clean way to clamp this movement to keep this from happening.  (I'm going through the same thing right now.)   What I'm thinking is instead of blowing up I want to limit the destination to something that can be accomplished.   That is a little bit tricky because it can be done in this routine but the code higher up on the stack doesn't know the nozzle is not where it thought it was told to be.    So, I'm trying to figure out a nice, clean way to do this.

The root of the problem is the Delta Spider's are so big and the Z-Probes are (almost) always mounted very far from the extruders.  This causes the probe radius to be much smaller than what is ideal.   Rather than use a 'radius', I'm looking at probing as much of the bed as the probe can reach.

~~~~
void calculate_delta(float cartesian[3]) {
float check1, check2, check3;   // RoxyHack

	check1 = delta_diagonal_rod_2_tower_1 - sq(delta_tower1_x-cartesian[X_AXIS]) - sq(delta_tower1_y-cartesian[Y_AXIS]) ;
	if (check1<0.0)  {
		SERIAL_ECHOPGM("YIKES! 1 Negative # in sqrt(");
		SERIAL_ECHO(check1);
		SERIAL_ECHOPGM(")\n");
	}

	check2 = delta_diagonal_rod_2_tower_2 - sq(delta_tower2_x-cartesian[X_AXIS]) - sq(delta_tower2_y-cartesian[Y_AXIS]) ;
	if (check2<0.0)  {
		SERIAL_ECHOPGM("YIKES! 2 Negative # in sqrt(");
		SERIAL_ECHO(check2);
		SERIAL_ECHOPGM(")\n");
	}

	check3 = delta_diagonal_rod_2_tower_3 - sq(delta_tower3_x-cartesian[X_AXIS]) - sq(delta_tower3_y-cartesian[Y_AXIS]) ;
	if (check3<0.0)  {
		SERIAL_ECHOPGM("YIKES! 3 Negative # in sqrt(");
		SERIAL_ECHO(check3);
		SERIAL_ECHOPGM(")\n");
	}

~~~~

mateddy, 2015-10-21:
Forget to specify - probably does not matter - Kossel Mini running on ATMega2560 and RAMPS 1.4.  No heat bed, PLA 1.75mm.

thetazzbot, 2015-10-24:
Does anyone have a working delta, inductive probe, auto bed level enabled Marlin?

Roxy-3DPrintBoard, 2015-10-24:
I'm making a fork for a Manually deployed Allen Key.  With a lot of goodies!  But making it handle an inductive probe shouldn't be hard.

thetazzbot, 2015-10-24:
I have been through several versions of Marlin, non of which actually have a working auto bed level.

My current problem is that no matter what grid points i give it, it physically starts dead center and goes to the right, back quadrant of the bed.  Probe functions fine.  Just will not start at the front,left corner like it should.

max-232, 2015-10-26:
Roxy & everybody, hello and thanks for your commitment.
I'm trying to figure out of the same issue and have found something interesting (at least I think).
I'm on 1.1.0-RC2 custom delta. If I set #define MANUAL_Z_HOME_POS to a lower value, so that while probing the bed, the min endstop is not touched, then I don'ìt have a U-shaped movement, but I have exactly what expected (a grid with radial limits - though it's done in the air).
At this point I'm investigating on the z min endstop. Seems to me that after the z min endstop is touched for the first time, it resets the bed center from 0,0 to -x,-y.
toughts?

Roxy-3DPrintBoard, 2015-10-26:
> At this point I'm investigating on the z min endstop. Seems to me that after the z min endstop is touched for the first time, it resets the bed center from 0,0 to -x,-y.
toughts?

I have not seen this behavior.   But if you want to get rid of the U-Shaped move...  You can add a few lines of code here:
~~~~
#define MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	25

  static void do_blocking_move_to(float x, float y, float z) {
    float oldFeedRate = feedrate;

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (marlin_debug_flags & DEBUG_LEVELING) {
        print_xyz("do_blocking_move_to", x, y, z);
      }
    #endif

    #if ENABLED(DELTA)

      feedrate = XY_TRAVEL_SPEED;

 //
 // Should we break up the move because it is too big ???
 //

      if (abs(x-current_position[X_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(y-current_position[Y_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(z-current_position[Z_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE ) {

  		do_blocking_move_to( current_position[X_AXIS] + (x-current_position[X_AXIS])/2.0,
  				     current_position[Y_AXIS] + (y-current_position[Y_AXIS])/2.0,
  				     current_position[Z_AXIS] + (z-current_position[Z_AXIS])/2.0);
//
// At this point, we are halfway to where we want to be.  Do the other half of the move.
//

		do_blocking_move_to(x, y, z);
		feedrate = oldFeedRate;
		return;
      } else {		// No...  It is small enough to do in one piece.

      	destination[X_AXIS] = x;
      	destination[Y_AXIS] = y;
      	destination[Z_AXIS] = z;
      	prepare_move_raw(); // this will also set_current_to_destination

      	st_synchronize();
      }
    #else

      feedrate = homing_feedrate[Z_AXIS];

      current_position[Z_AXIS] = z;
      line_to_current_position();
      st_synchronize();

      feedrate = xy_travel_speed;

      current_position[X_AXIS] = x
~~~~

max-232, 2015-10-26:
thanks for the answer.
I start believing we're not speaking about the same issues.
only on g29 my printer mess up y=0 with y=-190
x & y min endstops are disabled
here it is:
https://youtu.be/b07x0XUm0nU

Roxy-3DPrintBoard, 2015-10-26:
Does your printer all of a sudden start going up towards the Home position when it is trying to position very far from one of the towers?

max-232, 2015-10-26:
not yet experienced that.
only way for seeing it (on g29) correctly move Y starting from [radius -10] is to set MANUAL_Z_HOME_POS to a value that does not let z min endstop to be hit (which is useless for the use of the printer), otherwise all seems moved of half Y.
tried with stable and dev version. I'm on ramps 1.4.

Roxy-3DPrintBoard, 2015-10-26:
What are your   X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER numbers?   Are they big enough to cause the Probe radius to go somewhere it shouldn't ?

max-232, 2015-10-26:
both set to "0"
I'm probing with the nozzle pressure on the glass. as you can see in the video the glass is on 3 piezo that sense the pressure and set to high the z min pin.
after the first 0,0,0 probe, it moves away.

Roxy-3DPrintBoard, 2015-10-26:
I'm not an expert on the Z_MIN_PROBE stuff.   I'm bringing up a Delta printer for the first time and it has a Manually Deployed AllenKey Z-Probe.   I ran into trouble with the Z_MIN_PROBE stuff.   Some of the people here like @AnHardt can probably give you better advice than me.

I think part of the problem I ran into was the AllenKey probe stuff isn't well documented.  I have an AllenKey probe but I think there is some issue where you don't want to specify that if you have a Z_MIN_PROBE.   I ran into some configuration problems and just to keep making forward progress I made a few tweaks to the code.  I don't know this for sure, but I think I was supposed to not declare my printer as having an AllenKey probe even though it does have one if I was going to use the Z_MIN_PROBE.

If you want to try disabling the Z_MIN_PROBE from being checked except when you are actually doing a probe, you are welcome to try my code tweaks.   I added a few lines of code to deploy_z_probe() and stow_z_probe() so that the update_endstops() function in Stepper.cpp would not go crazy when my AllenKey probe was stowed.

I'm kind of jumping to a conclusion here, but we have had a lot of problems with the Z_Probe and I'm wondering if it is something like that happening.   Of course, in your case it is kind of the opposite of what I ran into.  In my case having the Z-Probe stowed caused it to give the update_endstops() function a 'Triggered' state.   That probably is not the case with your piezo sensors.

AnHardt, 2015-10-26:
@max-232 
Is there any possibility for noise on your z-probe/z-min sensors?
Does it make a difference if you use your sensor as a z-min-endstop or a z-min-probe?
Do you have BED_CENTER_AT_0_0 defined?
I'd love to have a look into your Configuration.h. Please give us a link.

max-232, 2015-10-26:
Good night (at least here is night) everybody and thanks a lot for your support.
This is the first delta I build too.
BED_CENTER_AT_0_0 is defined.
Here is my Marlin folder with all the files: https://goo.gl/Uo4vTN
Tomorrow I'll try out your suggestions.
Thank you!

AnHardt, 2015-10-27:
@max-232 
Could not find any obvious problem in your configuration.
But i'd try to define ENDSTOPS_ONLY_FOR_HOMING (Configuration_adv.h). When you try to print you'll probably get the z-min-endstop triggered.
Have you tested your endstops with M119

lordfwahfnah, 2015-10-28:
my Printer is doing the same weird thing after deploying the allen-key-probe. If i deploy it manualy and then send G29, he emediatly starts probing and retracts the probe with no problems. 
Is there a problem in the transition from probe deploy to actual probing? Seems like the coordination system gets messed up.

AnHardt, 2015-10-28:
@lordfwahfnah 
The same as @mateddy or the same as @max-232 ?
If the same as @mateddy likely your defined `DELTA_PRINTABLE_RADIUS` is to big.

The last move after stowing the allen-key moves to [0, 0, Z_PROBE_ALLEN_KEY_STOW_3_Z+Z_RAISE_AFTER_PROBING].
The next move is the first where the correction of the leveling process is used. So your move is (1mm down + correction_for_this_place).

lordfwahfnah, 2015-10-28:
> So your move is (1mm down + correction_for_this_place).

that makes sense i guess.

The DELTA_PRINTABLE_RADIUS should be correct. Its set slightly smaller than the actual bed (95mm. Bed diameter is somewhere 195mm).
Still weird that it works without the deploy.

Is there some kind of Guide to setup the ABL for a kossel mini with allen key probe?

AnHardt, 2015-10-28:
The beds diameter is not the only imported value.
https://github.com/MarlinFirmware/Marlin/issues/2679#issuecomment-146261100

```
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_X 30.0
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y DELTA_PRINTABLE_RADIUS
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Z 100.0
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE HOMING_FEEDRATE_XYZ
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_X 0.0
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_Y DELTA_PRINTABLE_RADIUS
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_Z 100.0
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE (HOMING_FEEDRATE_XYZ/10)
```
This points have to be reachable. Try to move there manually.

lordfwahfnah, 2015-10-28:
yeah i set them up correctly
i drove there manually and changed the code.
deploying the probe itself isn't making problems. the weird part comes after the deploying 

```
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_X 30.0
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y DELTA_PRINTABLE_RADIUS-3
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_Z 50.0
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE XY_TRAVEL_SPEED
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_X 5.0 // 0
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_Y Z_PROBE_ALLEN_KEY_DEPLOY_1_Y
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_Z Z_PROBE_ALLEN_KEY_DEPLOY_1_Z
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE (XY_TRAVEL_SPEED/10)
```

AnHardt, 2015-10-29:
After moving to Z_PROBE_ALLEN_KEY_DEPLOY_2 follows a check if the probe is deployed now and the code unifies again. The only difference i can find is a different position.  Next is a move to 
```
      // Partially Home X,Y for safety
      destination[X_AXIS] = destination[X_AXIS] * 0.75;
      destination[Y_AXIS] = destination[Y_AXIS] * 0.75;
```
That's a move towards the center of the bed.
Next move after that is to the first probable point of the grid.

What is your command sequence?
G28, G29 vs. G28, manual deploy, G29?
When does your error happen - exactly?
What happens?

thetazzbot, 2015-10-29:
@max-232 
I can tell you exactly what is happening because it was happening to me until I figured it out.

The settings in the config file for probe position tell only part of the story.

They are designed to define the perimeter of a probe area.  The next variable is the offset from the center to the allen key.  When the firmware is moving, it checks if the requested probe position x,y is beyond the "probable radius" and if it is, it skips that move!   So your challenge is to define the box so that the probe offset is deducted already.  Dont trust the example configs because they are wrong.

Example
I use an inductive probe, so its "always deployed"

 #define X_PROBE_OFFSET_FROM_EXTRUDER 0     // Probe on: -left  +right
#define Y_PROBE_OFFSET_FROM_EXTRUDER 20 // Probe on: -frontf +behind

#define DELTA_PROBABLE_RADIUS DELTA_PRINTABLE_RADIUS-10  

If you trusted the defaults:
#define LEFT_PROBE_BED_POSITION -DELTA_PROBABLE_RADIUS
#define RIGHT_PROBE_BED_POSITION DELTA_PROBABLE_RADIUS
#define FRONT_PROBE_BED_POSITION -DELTA_PROBABLE_RADIUS
#define BACK_PROBE_BED_POSITION DELTA_PROBABLE_RADIUS

But the actual, working settings for me are:

#define LEFT_PROBE_BED_POSITION -65
#define RIGHT_PROBE_BED_POSITION 55
#define FRONT_PROBE_BED_POSITION -60
#define BACK_PROBE_BED_POSITION 70

This is because when the auto bed level code sends a move x,y command, it automatically subtracts the probe offset from the X,Y

in Marlin_Main.cpp, probe_pt function

  do_blocking_move_to_xy(x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER); // this also updates current_position

problem is, my probe is on the positive side, so a move to -70-20 = -90, my printable radius is 90, probable = 90-10 which is 80, so i am telling it to go outside the probable radius and it wont, so it does exactly what yours does.

compare my settings above to where your probe is. and play with the numbers.

thetazzbot, 2015-10-29:
@max-232  I just reread your post about having offset set to zero.  but you're also not using an allen key probe so I'm not sure what problem you're having, I guess my response was for the allen key issue.  I had the exact "off bed probing" that your video showed though.

lordfwahfnah, 2015-10-29:
i got a small video now.
https://youtu.be/VrYWeMsALis
here you can see the difference between probe deployed and stowed during G29
hope that clarifies the situation a bit

AnHardt, 2015-10-29:
Thank you for answering my first question (https://github.com/MarlinFirmware/Marlin/issues/2722#issuecomment-151929670) now, after searching a couple of hours in the wrong direction.

AnHardt, 2015-10-29:
@lordfwahfnah 
Please activate DEBUG_LEVELING_FEATURE, send  `M111 S32` to actually switch on the extra output, retest both cases and post both logs. A link to your configs may be helpfull.

lordfwahfnah, 2015-10-29:
http://pastebin.com/CvQ5vfiS here is the log. I logged everything with MatterControl since Pronterface got a small cache and i dont know how to export it there.

I cut the power after about 5-7 probes.

AnHardt, 2015-10-30:
@lordfwahfnah 
Analysed the log, but could not find something strange. Line 146 of your log tells about moving into the negative quadrant, what is correct. 
Sadly the standard debug output does not include the tower positions. 
Would you please be so kind to put in some debug code manually?
I prepared https://github.com/AnHardt/Marlin/pull/6 to show you where. The actual line numbers in RC may differ.

I don't need the `M111 S32` output. Just from sending G29  to (including) the first probed point, for both cases - with AND without pre-deployed probe.

Thanks

lordfwahfnah, 2015-10-31:
hope i got it right. i'm not very experienced with programming yet. http://pastebin.com/X2TX0iDF (modded marlin_main.cpp)
sadly it's not gonna compile "'gcode_M114()' not declared in this scope"
it seems to work when i put it in parenthesis
`  else
            {
            act = ProbeStay;
            gcode_M114();

AnHardt, 2015-10-31:
@lordfwahfnah 
Yes i missed the forward declaration.

Thanks. It helps to see that the tower coordinates right before doing the first probe (behind the Count) still reflect the cartesian coordinates (before the Count).
```
<-X:-23.00 Y:-81.00 Z:5.80 E:0.00 Count X: 206.59 Y:184.41 Z:108.01
```
```
<-X:-23.00 Y:-81.00 Z:14.99 E:0.00 Count X: 215.78 Y:193.60 Z:117.20
```
difference cartesian z = 9.19 mm
difference tower x = 9.19 mm
difference tower y = 9.19 mm
difference tower z = 9.19 mm

That means we did moves not recognized by our counters. 
The coordinate systems as such are ok.
Maybe we synced somewhere before a move was ready?

Have to dig into the code again.

lordfwahfnah, 2015-10-31:
wohoo i found a bug :D 
Did you changed the ABL code recently? it worked flawless in the older Marlin versions. 
If i can help debugging (e.g. testing code) then just tell me.

max-232, 2015-11-05:
Hello everybody,
thanks for your support and sorry for the late reply.
I solved my issue, though it's not an elegant solution.
I'm on Marlin 1.1.0-RC2

Problem:
non allen key G29 Probe produced a "weird" 0,0 moved center after first probe

Solution:
in Marlin_main.cpp, after:

    // Move Z up to the z_before height, then move the Z probe to the given XY
    do_blocking_move_to_z(z_before); // this also updates current_position

add:

    do_blocking_move_to_z(25);

before:

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (marlin_debug_flags & DEBUG_LEVELING) {
        SERIAL_ECHOPAIR("> do_blocking_move_to_xy ", x - X_PROBE_OFFSET_FROM_EXTRUDER);

The problem was that after the first 0,0 probe, while moving up, it considered to have completed the second probe.
I'm using a piezo based probe, so I tried with a direct (mechanical switch) and got the same problem.
With the dirty solution above, the first 0,0 probe of G29 is just quicker and the whole probe -for me- works.

AnHardt, 2015-11-05:
@max-232 
Lets see what you got. (RCBugFix should be the same as RC here)

Inside the function 
```
static float probe_pt(float x, float y, float z_before, ProbeAction probe_action = ProbeDeployAndStow, int verbose_level = 1) {
...
```
you now, after your patch, have sequence of two calls to `do_blocking_move_to_z(   );` 
```
...
// Move Z up to the z_before height, then move the Z probe to the given XY
do_blocking_move_to_z(z_before); // this also updates current_position
do_blocking_move_to_z(25);
...
```
Basically this means: "First go to z=z_before, then go to z=25."

`z_before` is a parameter of the function `probe_pt` where we are in. So lets see how this function is called.
```
.../Marlin/marlin (MarlinFirmware/RCBugFix)
$ grep -r probe_pt *
Marlin_main.cpp:  static float probe_pt(float x, float y, float z_before, ProbeAction probe_action = ProbeDeployAndStow, int verbose_level = 1) {
Marlin_main.cpp:        SERIAL_ECHOLNPGM("probe_pt >>>");
Marlin_main.cpp:        SERIAL_ECHOLNPGM("<<< probe_pt");
Marlin_main.cpp:          measured_z = probe_pt(xProbe, yProbe, z_before, act, verbose_level);
Marlin_main.cpp:      float z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, Z_RAISE_BEFORE_PROBING, p1, verbose_level),
Marlin_main.cpp:            z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p2, verbose_level),
Marlin_main.cpp:            z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, p3, verbose_level);
```
All matches are in 'Marlin_Main.cpp'.
The first 3 matches are the definition it selves and 2 text-matches.
The last 3 are for 3-point leveling.
So only 
```
 measured_z = probe_pt(xProbe, yProbe, z_before, act, verbose_level);
```
is interesting. We are looking for the third parameter - ` z_before`.
Some lines above we find, for all architectures:
```
z_before = probePointCounter ? Z_RAISE_BETWEEN_PROBINGS + current_position[Z_AXIS] : Z_RAISE_BEFORE_PROBING;
```
For the first probing the `probePointCounter` is 0 - so z_before becomes Z_RAISE_BEFORE_PROBING.

For most configs Z_RAISE_BEFORE_PROBING is defined as 15 - only for the Kosse_Pro it's 100.
Or you have defined your own value.

Your command sequence now looks like: "Go first to z=YOUR_Z_RAISE_BEFORE_PROBING, then to z=25."

To be honest - i can not see ho this could improve the behavior of your printer.
You should have the same effect by setting your Z_RAISE_BEFORE_PROBING to 25.

An other thread of thoughts.
A delta should never do a probing at [0,0,0]! Regardless if with or without allen key probe.
Right after homing we are at [0,0,top]. If we have no allen key probe the first move is exactly that we discussed above - straight down to  Z_RAISE_BEFORE_PROBING. In the next step we drive to the first x,y. 

Could it eventually be your Z_RAISE_BEFORE_PROBING was set to 0 or undefined?
Or alternatively your MANUAL_Z_HOME_POS could have been more then Z_RAISE_BEFORE_PROBING to big?
Did you try to home to delta towards z-min? (don't try it)
A undefined ENDSTOPS_ONLY_FOR_HOMING will probably make unwanted "Z-Endstop hit at: xxx" messages during print.

max-232, 2015-11-06:
@AnHardt Thanks!
"A delta should never do a probing at [0,0,0]!" did the trick, I didn't know that.
Solution for me, in the end, was "Z_RAISE_BEFORE_PROBING 50" instead of 15.
Big apologies for the mess :(

AnHardt, 2015-11-06:
@max-232 
I'm just glade your printer is running now.

I'm currently investigating the endstop code. Could be, this first endstop hit at [0,0,0] messed up the coordinate system. Maybe it was interpreted as homing signal - or something like that.

Roxy-3DPrintBoard, 2015-11-06:
> I'm currently investigating the endstop code. Could be, this first endstop hit at [0,0,0] messed up the coordinate system.

I'm seeing something weird on my Delta.   I'm seeing the origin get shifted towards 4 O'Clock on the bed.   (I'm adding your random() calls to the M48)   I'm not sure what is causing it yet.  But there is something funny going on.

AnHardt, 2015-11-07:
I have a rough idea what could happen.
If here
```
  static void run_z_probe() {

    #if ENABLED(DELTA)

      float start_z = current_position[Z_AXIS];
      long start_steps = st_get_position(Z_AXIS);

      #if ENABLED(DEBUG_LEVELING_FEATURE)
        if (marlin_debug_flags & DEBUG_LEVELING) {
          SERIAL_ECHOLNPGM("run_z_probe (DELTA) 1");
        }
      #endif

      // move down slowly until you find the bed
      feedrate = homing_feedrate[Z_AXIS] / 4;
      destination[Z_AXIS] = -10;
      prepare_move_raw(); // this will also set_current_to_destination
      st_synchronize();
      endstops_hit_on_purpose(); // clear endstop hit flags

      // we have to let the planner know where we are right now as it is not where we said to go.
      long stop_steps = st_get_position(Z_AXIS);
      float mm = start_z - float(start_steps - stop_steps) / axis_steps_per_unit[Z_AXIS];
      current_position[Z_AXIS] = mm;

      #if ENABLED(DEBUG_LEVELING_FEATURE)
        if (marlin_debug_flags & DEBUG_LEVELING) {
          print_xyz("run_z_probe (DELTA) 2 > current_position", current_position);
        }
      #endif

      sync_plan_position_delta();

    #else // !DELTA
```
the move ` prepare_move_raw(); ` is not absolutely straight down (all steppers move with the same speed the same way) the coordinate system is messed up.

A
```
st_synchronize();
set_destination_to_current()
```
right after `#if ENABLED(DELTA)` may help to ensure that.

If we get a unexpected endstop hit the above correction is not done!

AnHardt, 2015-11-07:
The macro in `update_endstops()`
```
  #define UPDATE_ENDSTOP(AXIS,MINMAX) \
    SET_ENDSTOP_BIT(AXIS, MINMAX); \
    if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX))  && (current_block->steps[_AXIS(AXIS)] > 0)) { \
      endstops_trigsteps[_AXIS(AXIS)] = count_position[_AXIS(AXIS)]; \ // endstops_trigsteps is evaluated only in checkHitEndstops()
      _ENDSTOP_HIT(AXIS); \                                       // set the endstip bit we can evaluate outside the ISR
      step_events_completed = current_block->step_event_count; \   // stop the move by claming the endposition is reached
    }
```

This basically means the planner thinks the move has been finished normally, but the steppers stopped before.
Any noise on the endstop lines will mess up the correlation  of the planer and stepper coordinate system, for a delta..

Tej1996, 2015-11-08:
Not sure if this is the same issue, but I've got my delta printer on RC2 and its manually calibrated so that movement is already fairly flat. However, after a G29, the extruded hits the bed at the left side and is slightly lifted on the right. 

The probe is not set to be automatically deployed or retracted, I'm doing both manually. 

I can provide Z Probe readings and configs later if needed. Probe is consistent, I am almost certain it's not a hardware problem.  I tried various things to fix it, but haven't had any luck.

jeremy-brenner, 2016-01-08:
I had the same issue with the auto leveling starting at 0,0 and heading off the bed and crashing into my belts.  
This fixed it: 
"Solution for me, in the end, was "Z_RAISE_BEFORE_PROBING 50" instead of 15."
It seems that the premature triggering of the probe makes it go nuts,  raising it to 50 before starting ensured a clean start and it is working perfectly now.

Roxy-3DPrintBoard, 2016-01-08:
> It seems that the premature triggering of the probe makes it go nuts, raising it to 50 before starting ensured a clean start and it is working perfectly now.

In general the Alan Key Probes are a problem.  The issue is they are triggered when they are deployed.  If you set all the #defines right, they work.  But the MarlinDev branch has more robust handling of that issue.   And also more robust handling of the manually deployed and stowed probes.

jeremy-brenner, 2016-01-08:
Mine is an inductive sensor.  Z_RAISE_BEFORE_PROBING 15 didn't work for me because initially 0,0,0 is about 10mm below my bed ( I should change that ) so the probe was way to close when it would start probing.  My problem is solved.  It might be reasonable to simply go into a fail condition if the sensor triggers before the head can even get to the correct x/y position though,  continuing to test points makes for some really odd behavior including crashing into the towers.

bradanlane, 2016-01-19:
I'm running 1.1.0-RC3 and have the same issue as reported by @max-232. His video is exactly what I am experiencing.

It is unlikely related but my G28 Home is very abrupt and does not ease away from the end stops.

max-232, 2016-01-19:
@bradanlane
Ciao! My issue was with the first probe, here is the solution from previous comment:
"A delta should never do a probing at [0,0,0]!" did the trick, I didn't know that.
Solution for me, in the end, was "Z_RAISE_BEFORE_PROBING 50" instead of 15.

bradanlane, 2016-01-19:
Thanks. I'll make that change and test.

bradanlane, 2016-01-20:
reporting back _(so my problem doesn't cloud this issue thread further)_:

My problem is that all movements on my delta printer are "2X" - eg, move from G0 Z200 to G0 Z190 is moving 20mm.

I'm using the same defines as I had from the old jcrocholl version of Marlin that came with my delta printer.
```
#define XYZ_FULL_STEPS_PER_ROTATION 200
#define XYZ_MICROSTEPS 16
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 16
```

Roxy-3DPrintBoard, 2016-01-20:
Don't forget to check all your DELTA numbers in Configuration.h:
~~~~
#define DELTA_DIAGONAL_ROD 	198.5			
#define DELTA_SMOOTH_ROD_OFFSET 160.0 // mm
#define DELTA_EFFECTOR_OFFSET 34.0 // mm
#define DELTA_CARRIAGE_OFFSET 25.0 // mm
#define DELTA_RADIUS (DELTA_SMOOTH_ROD_OFFSET-DELTA_EFFECTOR_OFFSET-DELTA_CARRIAGE_OFFSET+0.00)  //Total value at 101.00
#define DELTA_PRINTABLE_RADIUS 9
~~~~

bradanlane, 2016-01-20:
Thanks 'roxy'. they are all correct for my printer.

I forgot to edit the above (just did). The "2X" movement is in all three axis.

AnHardt, 2016-01-20:
Did you paste in the biv2.5, gereric, kossel_mini or kossel_pro Configuration.h?
Only the last one colculates
```
#define XYZ_FULL_STEPS_PER_ROTATION 200
#define XYZ_MICROSTEPS 32
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 20
#define XYZ_STEPS (XYZ_FULL_STEPS_PER_ROTATION * XYZ_MICROSTEPS / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH))

// default settings
// delta speeds must be the same on xyz
#define DEFAULT_AXIS_STEPS_PER_UNIT   {XYZ_STEPS, XYZ_STEPS, XYZ_STEPS, 184.8}
```
the others have kind of
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80, 80, 80, 760*1.1}  // default steps per unit for Kossel (GT2, 20 tooth)
```

bradanlane, 2016-01-20:
Started with "kossel pro". Set my rod details, turned on auto bed leveling. Set necessary offsets. Then set feed info. My old configuration.h as well as the new one from 1.1.0-rc3 use the calculation for default_axis_steps_per_unit

AnHardt, 2016-01-20:
Than i have no clue where the problem is.
Lie!

bradanlane, 2016-01-20:
Found it (but still very confused).

Here are the configuration settings ...
```
#define XYZ_FULL_STEPS_PER_ROTATION 200
#define XYZ_MICROSTEPS 16
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 16
```
and here is my interpretation of the math ...
```
#define XYZ_STEPS (XYZ_FULL_STEPS_PER_ROTATION * XYZ_MICROSTEPS / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH))
#define DEFAULT_AXIS_STEPS_PER_UNIT   {XYZ_STEPS, XYZ_STEPS, XYZ_STEPS, 184.8}
```
XYZ_STEPS = 200* 16 / 2 / 16 = 100

However, after loading the firmware, M501 reports back ...
```
echo:  M92 X200.00 Y200.00 Z200.00 E184.80
```

Everything works as expected if I manually correct things with `M92 X100 Y100 Z100`

Jan9320, 2016-01-28:
Is there any running Kossel mini with allen probe key? Can't get auto bed levelling to start, it keeps skipping to retracting the probe.
Thanks

Roxy-3DPrintBoard, 2016-01-28:
> Is there any running Kossel mini with allen probe key? Can't get auto bed levelling to start, it keeps skipping to retracting the probe.

You might want to bring up the MarlinDev  Jan20 branch.   The Allen Key Probes make a lot more sense in that branch.

thinkyhead, 2016-03-11:
Where does this issue stand?

@Roxy-3DPrintBoard Should it be on my list to get those fancy new Allen Key probes into a release candidate, or are they incompatible with the "old" codebase? Sounds like the older Allen Key probe stuff is a little buggy compared to the newer.

closed 2016-05-04 by jbrazio