MarlinFirmware/Marlin #2723: Dual-X-Carriage parking speed

pbrgle1, 2015-10-22:
This is a support question (or feature request if no solution available).

I would like to increase the parking speed in my dual x carriage setup. While the Y axis can only handle max. 150mm/s, the X axis is far lighter and capable of higher speeds. This would greatly reduce print times overall and probably less leaking (since the nozzle is blocked when parked). How is the parking speed currently defined?

Wackerbarth, 2015-10-22:
How do you "park"?  Usually it is the slicer that issues directions to move the carriage to some location that it knows or it perform a "home" operation. If the slicer defines a movement rate, that rate will be used unless it exceeds the maximum movement rate for some axis axis.

You can limit each axis individually  by setting the movement parameters
`
/**
 * MOVEMENT SETTINGS
 */

#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)

// default settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,500}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {300, 300, 5, 25}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {3000,3000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)
`

pbrgle1, 2015-10-22:
No, the firmware parks the carriage (in my case). 

// There are a few selectable movement modes for dual x-carriages using M605 S<mode>
  //    Mode 0: Full control. The slicer has full control over both x-carriages and can achieve optimal travel results
  //                           as long as it supports dual x-carriages. (M605 S0)
  //    Mode 1: Auto-park mode. The firmware will automatically park and unpark the x-carriages on tool changes so
  //                           that additional slicer support is not required. (M605 S1)
  //    Mode 2: Duplication mode. The firmware will transparently make the second x-carriage and extruder copy all
  //                           actions of the first x-carriage. This allows the printer to print 2 arbitrary items at
  //                           once. (2nd extruder x offset and temp offset are set using: M605 S2 [Xnnn] [Rmmm])

  // This is the default power-up mode which can be later using M605.
  #define DEFAULT_DUAL_X_CARRIAGE_MODE 1

I'll try changing the max feedrate for the X axis but i still dont know how the parking speed (under firmware control) is defined.

caenan0, 2015-10-25:
Your parking speed will be the same as whatever your DEFAULT_MAX_FEEDRATE is.

thinkyhead, 2016-02-08:
We'll see about making this more clear in the configuration files and documentation. Perhaps this speed should be independently-adjustable in the future.

closed 2016-04-23 by jbrazio