MarlinFirmware/Marlin #2725: Pullrequest made for updates to make a smoother printing flow with no stutterings at high speed

justmyopinion, 2015-10-22:
I still have problems with Github and PR's but i hope someone could give it a view.
Corrections has been tested and is insstalled in a running printer,
1. remove an old hack from Stepper.cpp requesting input, that is necessary if you have baudrate higher than 115200 because limits to USARTS is making overrun because stepper ISR is too slow and blocking serial input.. set default 115200 which is adequate. tested
2. A bug that reduces cmd buffer to BUFSIZE -1 corrected to BUFSIZE. tested
3 an inadequate design in marlin_main  cmdprocessing has been changed to a more steady flow, and gives a very fast smooth printing.tested
It also helps on flow that delays in printer updates are kept to a minimum (display) by redusing amount of facility updates and use updatet display drivers set to max speed like TWI/WIRE speeds.which are typical 100k but should be at least 400k.

paulusjacobus, 2015-10-22:
Seems that the USB link is very critical. I had various prints stopping in the middle of printing when I switched on the light or fan in the room. An Ethernet port extension for the Arduino would be providing a better way for the comms with the host. Smoothie board has used this strategy as well.

For anyone interested in improving the code in Marlin. A good book is written by Simon Monk http://www.amazon.com/Programming-Arduino-Next-Steps-Sketches/dp/0071830251
Especially insight full are the interrupt code ISR examples: i.e. wave generation of max 84kHz can be moved up to 2 Mhz with some smart coding. Also explains the use of 8 bit int variables and the improvement in ram allocations. It really has improved my understanding of Arduino immensely.

justmyopinion, 2015-10-22:
@paulusjacobus 
Usb is always sensitive, so always use short cables with double screen where possible. 
However it was not part of my investigation into marlin. 
Ethernet could be another way of avoiding USB but maybe new problems could show up, latency etc..? 
Looks like an interesting book you have found.

paulusjacobus, 2015-10-23:
@justmyopinion indeed an interesting book. I had a look at the stepper ISR (stepper.cpp) to see how it works and actually I might have found another issue related to microstepping. There is a section that defines the micro steps (switch case statement 1 til 16). 
void microstep_mode(uint8_t driver, uint8_t stepping_mode) {
  switch(stepping_mode) {
    case 1: microstep_ms(driver,MICROSTEP1); break;
    case 2: microstep_ms(driver,MICROSTEP2); break;
    case 4: microstep_ms(driver,MICROSTEP4); break;
    case 8: microstep_ms(driver,MICROSTEP8); break;
    case 16: microstep_ms(driver,MICROSTEP16); break;
  }
}
That fragment does not extend to Micro step 32 so wonder if that affects my micro stepping (upgraded my step sticks to DRV8255).

Wurstnase, 2015-10-23:
That part is only available for some boards where you can configure the microstepping via software. Most boards, RAMPS etc., have jumpers.

closed 2015-10-31 by justmyopinion