MarlinFirmware/Marlin #2726: MAX31855 Temp 0 error

Grogyan, 2015-10-23:
Both a request and a bug
Marlin works fine with a MAX6675 thermocouple
Doesn't work with a MAX31855 thermocouple

Using as a base RAMPS1.4 and an Adafruit MAX31855
https://www.adafruit.com/products/269 
Marlin 1.1 RC2

Marlin reports ID 0 error.

This is because the MAX31855 uses a 32 bit address vs the MAX6675 16 bit address
With having a 32 bit address, it opens up the possibility of the firmware to recognise on the fly, problems with an attached.
See datasheet MAX31855
http://www.adafruit.com/datasheets/MAX31855.pdf page 10
vs datasheet MAX6675
http://www.adafruit.com/datasheets/MAX6675.pdf page 6

Request adding another Thermocouple option in configuration "-3" MAX31855

As a side note, I will eventually be testing out the MAX31850, which uses a 1-wire interface, an enables multiple thermocouples to be added to the same 1-wire bus

justmyopinion, 2015-10-24:
I do not have any thermocouples, but looking into datasheet i would think that 31855 should be compatible with 6675 because the first 16bits are clocked out in both cases.
Marlin only use the first 12 (13) bits anyway. The fact that 31855 has 32 bits avaiable should not affect the reading of the first 16 bits.
But I might read it wrong of course :-)

Grogyan, 2015-10-24:
The way that the MAX31855 presents the data is different to the MAX6675 with the first  bits reserved for flagging to the firmware of potential problems with the thermocouple, Open Circuit, Short to GND or VCC
It also presents the internal compensator temperature. Plus a couple of sign bits.
It isn't just a quick replacement for the MAX6675 unfortunately

justmyopinion, 2015-10-24:
I noticed it is different of course, but if you look at the first 16 bits (MSW) and ignore signbit only the least bits could worry and that may be what is failing with Marlin as an (ID) bit is tested (and maybe not even correct 4/2). if this test is ignored/changed i think it should work out-of-the-box. but with 12 bits resolution as 6675. I may be wrong :-)

Grogyan, 2015-10-24:
It would require a bit shift of 13 to the left to another variable to have
the LSB in the right place to read data correctly.
The code for the MAX6675 has a bit shift of 3 to the right of the same
variable which places the remaining bits in the right place for easy
reading.
Though IMO it is better to work with the 32bit word as a whole and utilize
the 4 fault bits.
I'm looking at the code, attempting in my limited coding capacity to find a
workable method.

justmyopinion, 2015-10-24:
As i am reading it:
6675 has a 16bit register and Marlin is reading this as a 2 byte word and shifting 3 bits to the right, correct.
31855 has a 32 bit register that is read with MSB first so if you let Marlin read the first 16 bit (MSW) just like 6675 it would not know the difference 8with positive temps). The remaing 16 bits which actually is a compensating temperature is left unreaded.
Marlin is probably failing because it is testing wrong bit and with 31855 it is a LSB.

Why not just try my suggestion above, i have no hw so it would not work here.

Grogyan, 2015-10-26:
I tried your suggestions, and Marlin just reads 0 degrees

justmyopinion, 2015-10-26:
strange, but I have no clue on this.
of course it was a hack to get you going as the correct way should be to read the compensation as well but at 25 deg error is nill without compensation reading in the last 16 bit.
i still would say that you should get a reading but as I have no sample I cannot check it.
I wonder why MAX made this chip as replacement, as 6675 did the compensation internally.

AnHardt, 2015-10-26:
Could
https://github.com/reidrankin/MarlinDev/commit/1cacab9b7a5ec1135a59644dbf8776ade9036622
be helpful for you?

Grogyan, 2015-10-26:
@AnHardt It looks very promising, as I was trying to do a similar hack of  the code with my limited coding skills.
I'll give his code a try out, hopefully this evening

Grogyan, 2015-10-27:
I only had a short time to try out the code last night.
Unfortunately there is still no temperature readback

Grogyan, 2015-10-27:
/Doh!
I should have known the SCK pin is also a 3V3 sensitive pin, right now I am ordering more of the Adafruit MAX31855 boards.
Currently am working on the assumption that I have blown up my current MAX31855 boards.

Grogyan, 2015-10-29:
OK, new MAX31855 board attached
No reading unless I enable a display with SD card enabled

AnHardt, 2015-10-29:
If its depending on SDSUPPORT,recheck if your wiring is the same as in your pin file.
https://github.com/reidrankin/MarlinDev/blob/1cacab9b7a5ec1135a59644dbf8776ade9036622/Marlin/pins_RAMPS_13.h#L245

Grogyan, 2015-10-29:
I will try to do some more testing this evening to see if I can narrow down
the cause.
I literally only had 10mins free yesterday.

And will post on his pull request.
Hopefully we can get this to the released version soon.

Cheers

thinkyhead, 2016-02-10:
Any update on this? Would love to see a PR to support the new hardware.

Grogyan, 2016-02-10:
Appears to still be a random communication drop out

thinkyhead, 2016-02-19:
@Grogyan Random is tricky. Is the thermocouple board getting a constant 3.3V, etc. etc.? In discussing thermocouples in the past (and maybe other devices) I vaguely remember issues concerning the select pin. Check out this thread: #753

Grogyan, 2016-02-19:
I'm planning on working on printer with the MAX31855 over the weekend.
Will let you know more after the weekend.

Grogyan, 2016-03-04:
Sorry for the delay.
Regarding my last post, I got confused with MAX6675 support.
Trying again this weekend with using the RCBugfix

Grogyan, 2016-03-05:
Still applying the changes from  #https://github.com/reidrankin/MarlinDev/commit/1cacab9b7a5ec1135a59644dbf8776ade9036622 still RCBugfix, still results in no temp measurement

Grogyan, 2016-03-06:
More digging required, my thermocouple does work as expected when using the Adafruit example.
Neither Repetier or Marlin work

Grogyan, 2016-03-08:
Have got this working, with the caveat that SDSUPPORT needs to be enabled in the configuration file

thinkyhead, 2016-03-08:
Sounds like an SPI conflict or issue of some kind. I wish I knew more about this sort of thing. I'm just looking at the [initial commit for MAX31855 support](https://github.com/reidrankin/Marlin_old/commit/2850a950b48b94b4151ee9a71cf0ab18d8607434).

thinkyhead, 2016-03-08:
Try out #3112 and see if it has any effect. You should use `#define TEMP_SENSOR_0 -3` for the MAX31855.

Grogyan, 2016-03-12:
Closing this as this has been superseded by PR https://github.com/MarlinFirmware/Marlin/pull/3112

closed 2016-03-12 by Grogyan