MarlinFirmware/Marlin #2727: Compiling error pin.h for board sanguinololu_12 line 1352

jataadroid, 2015-10-24:
In arduino ide 1.0.6 using marlin firmware and supporting files for sanguinololu 1.3a functionality, attempting to compile a configured firmware return error. Line 1352 of pin.h is pin layout for ulitmaker when its supposed to be sanguinololu_12. Returns saying wrong board.

No matter what you select if it is for MOTHERBOARD_SANGUINOLOLU_12  it will always return a compiling error for pin.h line 1352.

Running win 7 with all necessary drivers. Attempting to get a prusa i3 from geeetech working.

jataadroid, 2015-10-24:
Yes the right board is selected in Arduino IDE

jataadroid, 2015-10-25:
sanguinololu_12 is only referenced in pins.h for the section just before ultimaker that is dedicated to sanguinololu_11.

thinkyhead, 2016-02-08:
This should be all fixed in the latest RC (currently 1.1.0-RC3). We can re-open this issue if the problem has not been fixed in that version.

closed 2016-02-08 by thinkyhead