MarlinFirmware/Marlin #2729: SERVO0_PIN incorrectly defined for Ramps1.3 in pins_RAMPS_13.h

arthur500, 2015-10-25:
For Marlin 1.1 RC2 in file pins_RAMPS_13.h the define for SERVO0_PIN is set as 11.
Ramps1.3 boards use pin 7 for SERVO0_PIN while Ramps1.4 boards use pin 11. The file needs a test to distinguish between 1.3 and 1.4 version boards.  This pin is usually used for auto bed levelling control.

AnHardt, 2015-10-26:
http://reprap.org/wiki/File:Arduinomega1-3connectors.png
http://reprap.org/wiki/File:Arduinomega1-4connectors.png

AnHardt, 2015-10-31:
There are more differences between the graphics
```
1.3 1.4
  7  11
  2   1
  1   0
 48  40
```
But when tracing http://reprap.org/wiki/File:Ramps_v1.3_shield_bothsides.png and http://reprap.org/wiki/File:Ramps_v1.4_shield_bothsides.png you find the last 3 differences are typos in http://reprap.org/wiki/File:Arduinomega1-3connectors.png.

AnHardt, 2015-10-31:
Anyone a better solution then https://github.com/MarlinFirmware/Marlin/pull/2741 ?

AnHardt, 2015-11-24:
#2741 is merged.

closed 2015-11-24 by AnHardt