MarlinFirmware/Marlin #2734: RC-2 Multiple Extruder Auto Bed Leveling does not work!

Roxy-3DPrintBoard, 2015-10-27:
I apologize...  I thought I saw a thread where ThinkyHead had made a fix for this.  But I might be imagining that because I can't find it anywhere.

But right now, it looks like the Z-Probe offset is being lost when the printer switches the active extruder from one to the other.

Roxy-3DPrintBoard, 2015-10-27:
I found it:  https://github.com/MarlinFirmware/MarlinDev/pull/124

It looks to me like this needs to be applied to RC-2 also.   But I would appreciate somebody else jumping in and confirming that!   :)

thinkyhead, 2016-02-08:
Yes, if this patch hasn't also been submitted to `RCBugFix`, I should do that asap. This bug was discovered a while ago and the fix is pretty simple.

thinkyhead, 2016-02-08:
Ok, to make this fix properly, according to @jminardi just one change needs to be done. The line which reads:
```cpp
offset_vec.apply_rotation(plan_bed_level_matrix);
```
…should be changed to…
```cpp
offset_vec.apply_rotation(plan_bed_level_matrix.transpose(plan_bed_level_matrix));
```
I will make a new set of PRs for `dev` and `RCBugFix` that make this change as specified.

closed 2016-04-23 by jbrazio