MarlinFirmware/Marlin #2736: pruisa i3 - g28 fail home after g29

NesThaesis, 2015-10-28:
Hello Guys.

Im fairly new to 3d printing, i started to learn it about a week ago, because one of my friend has a pruisa i3 and i asked him to print me a model. It turns out, he cant print anymore since his bed is unbalanced, and we couldn't level it manually, the modells kept failing at the first 2-3 layer.

So i dug into how to do the auto leveling process.

I bought a LJ12A3-4-Z/BX inductive proximity sensor. ( it should be blue, but its orange. dunno why ), but anyway the motherboard is an MKS Base v1.2 and i connected the sensor to the Z min data, along with the 5V and ground connections.

The sensor works, it has enough power to run on 5 volt, the led comes on, it triggers when reaching the bottom of the bed. So i shouldn't complain about it.

It is installed around 72mm to the right of the extruder, and so i set 
#define X_PROBE_OFFSET_FROM_EXTRUDER 72

Now auto bed leveling is set to Grid mode and it is set to probe the following points.

    #define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 160
    #define BACK_PROBE_BED_POSITION 220
    #define FRONT_PROBE_BED_POSITION 20

This is where my problem occurs. Upon giving the command G28XYZ the extruder goes back to XY and Z 0 point where its home should be. Then after G29 the probe starts the auto bed leveling, it does check the 4 point given above, then it sets itself around halfway of the Y axis, and around 10mm into the X axis. Now both auto home and g28 comes back into this point,

Starting a print makes the printer start the printing on the other side of the board ( about halfway of the X axis ) and it makes the extruder run off the printing bed.

Any idea where did i go wrong ?

Thank you for any support i really appriciate it.

AnHardt, 2015-10-29:
Sending a G27 after a G29 makes no sense at all, because it makes the results (matrix) of G29 invalid.

The hosts (Repetier Host, Pronteface) use relative moves to move the head when you hit the buttons. After a G29 they do not know exactly where the head is. Try to sync with a manual G0/G1 code like `G0 X0 Y0 Z20`

NesThaesis, 2015-11-03:
As it turned out, my marlin wasnt up to date, so we reconfigured the newest one and now it seems to work.

closed 2015-11-03 by NesThaesis