MarlinFirmware/Marlin #2737: Website is down

dcohenb, 2015-10-29:
When trying to visit http://www.marlinfirmware.org/
I get an error "Sorry! This site is experiencing technical difficulties."
Not sure if this is the place to report it but I can't think of anywhere else

AnHardt, 2015-10-29:
(Cannot contact the database server)

@thinkyhead  @scotty1024

a4jp-com, 2015-10-29:
Same error message here

Wackerbarth, 2015-10-29:
I thought that, because we could not get access for commit rights, we were abandoning that site in favor of one which we could actually maintain.

dcohenb, 2015-10-29:
If it's a simple static website we can host it on Github
If it's more complicated I can donate an instance to host the website

Wackerbarth, 2015-10-29:
It was my understanding that we had moved it to the wiki associated with this github account

AnHardt, 2015-10-29:
@Wackerbarth 
The wiki here is far from ideal. A Mediawiki is much more comfortable. But every wiki is better than one you can't write to.
At least we have a backup now.

@dcohenb 
No, it was a Mediawiki.

Wackerbarth, 2015-10-29:
@AnHardt -- "But every wiki is better than one you can't write to." 
Precisely! Isn't that why we changed?
But "CHANGED" is the operative word.

So, from my perspective, www.marlinfirmware.org does not exist and we should stear all Marlin users away from it.

If the owners of that website wish to rejoin, then they must address the access issues. As a community, we cannot have a critical part of the infrastructure which is out of the control of the core community.

AnHardt, 2015-10-30:
For me it was more an 'emergency operation' then a wanted change. Without any positive feedback a 'we' was questionable for me.

However. If someone could __provide__ and __maintain__ a Mediawiki-site i'd highly appreciate it.

dcohenb, 2015-10-30:
@AnHardt do you know who's managing the domain?

AnHardt, 2015-10-30:
The domain was, as far as i know, provided by @scotty1024. But up to now he has not reacted to any recent requests here. Sadly we have no mail-address.
As far as i know @thinkyhead and @boelle have had shell accounts. As long as the web-server is running they could set a redirection - hopefully.

mkubik, 2015-11-10:
Still no progress? 
In the meantime, can anyone make the docs available somewhere else?

KiteLab, 2015-11-10:
https://github.com/MarlinFirmware/Marlin/wiki
is what we have.

thinkyhead, 2016-02-10:
Gone forever, as I cannot connect by SFTP even to implement a redirect. The server does not respond to ping. Just use the wiki here on Github to work on the documentation. Make it awesome, maybe it can become the basis of the official Marlin for Dummies book....

closed 2016-02-10 by thinkyhead