MarlinFirmware/Marlin #2738: When extruders >1 with Azteeg X3 Board DIO error generated

prcdslnc13, 2015-10-29:
On the release branch if I set my extruders to anything greater than 1 I get an error in the fastio.h section. Error message is pasted below. If I set the board to RAMPS 1.3 EEB it compiles fine. Ideas?

Arduino: 1.6.5 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

{DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); }\
                                                                        ^
fastio.h:45:80: note: in expansion of macro '_WRITE_C'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                ^
fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:1270:52: note: in expansion of macro 'WRITE'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                    ^
In file included from Marlin.h:21:0,
                 from temperature.cpp:32:
pins.h:687: error: invalid suffix "_PIN" on integer constant
     #define HEATER_1_PIN       -1
                                 ^
fastio.h:17:26: note: in definition of macro 'MASK'
 #define MASK(PIN)  (1 << PIN)
                          ^
fastio.h:45:80: note: in expansion of macro '_WRITE_C'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                ^
fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:1270:52: note: in expansion of macro 'WRITE'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                    ^
temperature.cpp:1270:58: note: in expansion of macro 'HEATER_1_PIN'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                          ^
In file included from Marlin.h:21:0,
                 from temperature.cpp:32:
fastio.h:39: error: pasting "DIO" and "-" does not give a valid preprocessing token
                                          {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }\
                                           ^
fastio.h:45:80: note: in expansion of macro '_WRITE_C'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                ^
fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:1270:52: note: in expansion of macro 'WRITE'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                    ^
pins.h:687: error: invalid suffix "_WPORT" on integer constant
     #define HEATER_1_PIN       -1
                                 ^
fastio.h:39:51: note: in definition of macro '_WRITE_C'
                                          {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }\
                                                   ^
fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:1270:52: note: in expansion of macro 'WRITE'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                    ^
temperature.cpp:1270:58: note: in expansion of macro 'HEATER_1_PIN'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                          ^
fastio.h:39: error: pasting "DIO" and "-" does not give a valid preprocessing token
                                          {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }\
                                                                         ^
fastio.h:45:80: note: in expansion of macro '_WRITE_C'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                ^
[... 407 lines cut ...]
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                        ^
fastio.h:45: error: 'DIO' was not declared in this scope
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                    ^
fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:1270:52: note: in expansion of macro 'WRITE'
     if(soft_pwm_1 > 0) WRITE(HEATER_1_PIN,1); else WRITE(HEATER_1_PIN,0);
                                                    ^
fastio.h:45: error: 'DIO' was not declared in this scope
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                    ^
fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:1292:30: note: in expansion of macro 'WRITE'
   if(soft_pwm_1 < pwm_count) WRITE(HEATER_1_PIN,0);
                              ^
pasting "DIO" and "-" does not give a valid preprocessing token

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

justmyopinion, 2015-10-29:
i am using Arduino 1.65 but Marlin RC1 or RC2 in W7/64
setting this one :   #define MOTHERBOARD BOARD_AZTEEG_X3 and 2 extruders and I do not have any compiler errors. Have you changed anything that could make this happen?

Wackerbarth, 2015-10-29:
"On the release branch” — That is your problem. The Release branch is Marlin 1.0.x and that branch does not support your configuration. (And the issue will not be fixed on that branch)

You need to be using Marlin1.1-RC2, or later.
Further, you need to use the Marlin platform (board definitions) rather than the Arduino platform definitions.

prcdslnc13, 2015-10-29:
Ok in marlin1.1-RC-2 I get the following error. The only two changes I made were in configuration.h setting motherboard to #define MOTHERBOARD BOARD_AZTEEG_X3 and extruders to 2. Setting motherboard to 67 produces same error. 

Arduino: 1.6.5 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from Configuration_adv.h:601:0,
                 from Configuration.h:861,
                 from Marlin.h:26,
                 from blinkm.cpp:5:
SanityCheck.h:313: error: #error HEATER_1_PIN not defined for this board.
       #error HEATER_1_PIN not defined for this board.
        ^
SanityCheck.h:319: error: #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
       #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
        ^
Exception in thread "AWT-EventQueue-0" java.lang.Error: Error: could not match input

	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.zzScanError(CPlusPlusTokenMaker.java:2077)

	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.yylex(CPlusPlusTokenMaker.java:2311)

	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.getTokenList(CPlusPlusTokenMaker.java:1901)

	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)

	at org.fife.ui.rsyntaxtextarea.SyntaxView.paint(SyntaxView.java:719)

	at javax.swing.plaf.basic.BasicTextUI$RootView.paint(BasicTextUI.java:1434)

	at javax.swing.plaf.basic.BasicTextUI.paintSafely(BasicTextUI.java:737)

	at javax.swing.plaf.basic.BasicTextUI.paint(BasicTextUI.java:881)

	at javax.swing.plaf.basic.BasicTextUI.update(BasicTextUI.java:860)

	at org.fife.ui.rtextarea.RTextAreaBase.paintComponent(RTextAreaBase.java:719)

	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.paintComponent(RSyntaxTextArea.java:1993)

	at javax.swing.JComponent.paint(JComponent.java:1056)

	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5219)

	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1572)

	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1495)

	at javax.swing.RepaintManager.paint(RepaintManager.java:1265)

	at javax.swing.JComponent._paintImmediately(JComponent.java:5167)

	at javax.swing.JComponent.paintImmediately(JComponent.java:4978)

	at javax.swing.RepaintManager$4.run(RepaintManager.java:824)

	at javax.swing.RepaintManager$4.run(RepaintManager.java:807)

	at java.security.AccessController.doPrivileged(Native Method)

	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)

	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:807)

	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:782)

	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:731)

	at javax.swing.RepaintManager.access$1300(RepaintManager.java:64)

	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1720)

	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)

	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)

	at java.awt.EventQueue.access$500(EventQueue.java:97)

	at java.awt.EventQueue$3.run(EventQueue.java:709)

	at java.awt.EventQueue$3.run(EventQueue.java:703)

	at java.security.AccessController.doPrivileged(Native Method)

	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)

	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)

	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)

	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)

	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)

	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)

	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

#error HEATER_1_PIN not defined for this board.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

justmyopinion, 2015-10-29:
Well I think the intention was to have an include file for each of these implementations, but it looks like only one is implemented ( RAMPS_13_EFB) yours is probably  RAMPS_13_EEB but the list is not complete (I have 2 extruders, BED and a couple of fans :-D ) so some improvisation may be needed. Bu that is probably solved in next release or next or....

 *  RAMPS_13_EFB (Extruder, Fan, Bed)
 *  RAMPS_13_EEB (Extruder, Extruder, Bed)
 *  RAMPS_13_EFF (Extruder, Fan, Fan)
 *  RAMPS_13_EEF (Extruder, Extruder, Fan)
 *  RAMPS_13_SF  (Spindle, Controller Fan)

The fastest solution is to change the switch in include file pins_RAMS_13_EFB.h from
#define IS_RAMPS_EFB
to this:
#define IS_RAMPS_EEB

Then it may work :-)

prcdslnc13, 2015-11-03:
Thank you. That does work. Panucatts recommendation was to just use RAMPS_13_EEB and not use the azteeg x3 at all. 

Thanks for the responses

closed 2015-11-24 by AnHardt