MarlinFirmware/Marlin #2739: Scaeld Z axis with MANUAL_BED_LEVELING and MESH_BED_LEVELING

oscargilfc, 2015-10-29:
Additional to homing problem #1682
When MESH_BED_LEVELING is used Z axis is scaled and it moves about 1,1 times more when going up, going down there appears there is no problem. I made tests and I think is software error, if I comment the next lines z axis works properly. 

    #define MANUAL_BED_LEVELING  // Add display menu option for bed leveling.
    #define MESH_BED_LEVELING 

I have not tested with AUTO_BED_LEVELING because I don't have the required hardware. I tried with mesh sizes of 3x3 4x4 and 5x5

In short distances is not noticeable, but if I move it 100 mm it goes more than 110. It sounds like it have very short pauses. In 2 occasions my board reset, I don't know if it was another thing but that only had happened to me with those options. 

 I first thought it was my stepping configuration, but when I disable those It works good.

epatel, 2015-10-29:
Hi, great to hear you trying the `MESH_BED_LEVELING`, and sorry you having problems with it.
You testing just moving it straight up? like `G1 Z100`? Do you do this just after homing, `G28`?
You might have a version where `G28` does not go to `Z=0` but rather leave the nozzle slightly above the bed. I added some code to fix this in this pull request https://github.com/MarlinFirmware/MarlinDev/pull/199 but it doesn't seem to have been merged.
So, I recommend doing a `G1 Z0` just after the `G28`.

oscargilfc, 2015-10-29:
Hello, I think is not the offset, I measure the displacement with a caliper. I fixed the Z=0 after homing and still have that issue. I have tested moving just after G28, I will perform up and down tests and see if the same happens after going down, I've actually don't remember if tried that, may be y my z=0 fix is wrong, I will try with your code fix.

epatel, 2015-10-30:
What kind of printer do you have? I have only tested on "normal" none delta

Roxy-3DPrintBoard, 2015-10-30:
> What kind of printer do you have? I have only tested on "normal" none delta

@epatel Have you looked at how Delta's do Auto Bed Leveling?  It seems it is really just an extension or automatic way of doing what you came up with.   When you have some spare time and feel like doing it, can you compare your stuff with what Delta's do?  I think there might be an opportunity to combine the two.

thinkyhead, 2016-02-10:
How is this looking? If there's still an issue, try using the `DEBUG_LEVELING_FEATURE` with `M115 S32` and checking the output for any weird results.

thinkyhead, 2016-04-22:
Is this still an issue with the current RCBugFix code or did we manage to patch it?

oscargilfc, 2016-04-22:
I will test it and post the results.

closed 2016-05-04 by jbrazio