MarlinFirmware/Marlin #2743: Number of axes hardcoded instead of already existing variable in for loop

ayildirim, 2015-10-31:
Hi,

I was looking into code to check if it supports tuning E-Step separately for each extruder.

Below there is configuration for number of axes; 
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Configuration.h#L477

If I increase this to 5, and add 1 more value to each array there in  below. It appears code would ignore those extra values after 4th. 

Please check linked code line below to see the issue. Number of axis is hardcoded as "4". Instead of using NUM_AXIS variable.
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/ConfigurationStore.cpp#L375

Can you confirm that I can not add 1 more extruder e-step configs separately due to this issue?

Thanks you for the work guys.
Cheers,
Ahmet

AnHardt, 2015-10-31:
I can confirm there is no easy way to have more then 4 axis in Marlin. Especially not in the release branch.
I can not confirm the limit you detected is the only one. Almost no part of marlin is able to handle more then 4 axis.

Marlin is able to handle more the one extruder, but not at the same time This is not designed to handle extruders with different esteps. As a workaround you can try to set esteps with M92 when changing the extruder.

ayildirim, 2015-11-02:
Hi @AnHardt,

This is good to know. I saw lots of people on forums trying to increase value to 5 to adjust second extruder.

I sent a merge request adding warning in comment above that value declaration.

thinkyhead, 2016-02-10:
The reason for the `NUM_AXIS` define is so that loops don't just say `while (i < 4) { ... }` making you guess what is meant by the number "4." Thus any loop in the code that says `for (i=0; i<NUM_AXIS; i++) { ... }` pertains to the 4 standard axes, XYZE, currently supported in Marlin.

The way that multiple extruders are handled depends on how they interact. Currently when Marlin switches tools, the new tool "becomes" the E axis. Most of the time this is what we want, and it works fine. Even if we never need to change that behavior, we frequently hear from users who would like to be able to have extruders with different profiles - E steps, retraction, temperature, etc., so we're trying to come up with a good way to accommodate that.

Mixing extruders with up to 3 motors are also interesting. I'm talking to someone who has a pretty good patch for that type of extruder, which still uses a single E axis, but moves the three steppers i proportion to the current "mix" values (which always add up to 1). So you can imagine, mixing R=0.2 Y=0.7 B=0.1. The GCode for that might be something like `G44 A0.2 B0.7 C0.1` (and `ABC` terms can be included in `G1` commands too).

closed 2016-05-04 by jbrazio