MarlinFirmware/Marlin #2745: Help to use Makefile

ipa64, 2015-11-01:
Hello, 
is someone can help a newbie please?
I would like to use Makefile (I use Sublime Text editor so I don't want to open arduino gui to compile)
I'm on OSX El capitan, I have arduino 1.6.5 installed
$ Make
I got the error:
make: *** No rule to make target `/Users/iPa64/Documents/Arduino/wiring.o', needed by `/Users/iPa64/Documents/Arduino/Marlin.elf'.  Stop.

Here is my Makefile header:
HARDWARE_MOTHERBOARD ?= 33
ARDUINO_INSTALL_DIR  ?= /Applications/Arduino.app/Contents/Java
ARDUINO_VERSION      ?= 165
AVR_TOOLS_PATH ?= /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin
UPLOAD_RATE        ?= 115200
AVRDUDE_PROGRAMMER ?= arduino
UPLOAD_PORT        ?= /dev/arduino
BUILD_DIR          ?= /Users/iPa64/Documents/Arduino
LIQUID_TWI2        ?= 0
WIRE               ?= 0

Roxy-3DPrintBoard, 2015-11-01:
You can check a box in Arduino and tell it you are using an external editor.     You can still use Sublime without going to using a Makefile.   I suggest you try that first.

ipa64, 2015-11-01:
Thank you, it's a good workaround

closed 2015-11-24 by AnHardt