MarlinFirmware/Marlin #2746: Compile issue with dual extruder

mkubik, 2015-11-01:
I just picked the latest RC and changed EXTRUDERS to 2.
The compile fails with:

temperature.cpp: In function ‘void manage_heater()’:
temperature.cpp:665:1: error: unable to find a register to spill in class ‘POINTER_REGS’
 }
 ^
temperature.cpp:665:1: error: this is the insn:
(insn 95 94 97 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 141 [ ivtmp.167 ])) [4 MEM[base: _101, offset: 0B]+0 S4 A8])) temperature.cpp:449 99 {*movsf}
     (expr_list:REG_INC (reg:HI 141 [ ivtmp.167 ])
        (nil)))
temperature.cpp:665: confused by earlier errors, bailing out

This happens with different board types. (ulitmaker, ramps, rumba).
Temp sensors are defined, anything else I missed?

KiteLab, 2015-11-01:
The error you describe is very well known for the `release` branch. Are you sure you tried to compile RC?

mkubik, 2015-11-01:
You're right. Sorry... I picked the wrong source zip.

closed 2015-11-01 by mkubik