MarlinFirmware/Marlin #2749: Slight license ambiguity

vogtmann, 2015-11-02:
While the COPYING.md file, the entry on the reprap wiki, and most importantly the source says that it's licensed under GPLV3, the LICENSE file is a copy of the GPLV2

Wackerbarth, 2015-11-02:
Although the content of the reprap wiki has no legal bearing on the license, the commit history does indicate that this is has been under GPLv3 since at least August 2011.

Therefore, rather than performing all of the merges requested, I will cherry-pick the change to each of the branches affected.

closed 2015-11-02 by Wackerbarth

vogtmann, 2015-11-02:
I wasn't sure if I should submit a pull request for each branch, or just the Release branch. So I defaulted to the one that I thought would be the most helpful.

Wackerbarth, 2015-11-02:
I appreciate the effort. However, the way github handles pull requests, I end up expending some effort to "clean up the result". It is actually much easier to just cherry-pick something like this that needs to be replicated in multiple places in the tree. And for changes that affect a file that has moved, I often prefer to simply copy the file from one copy of the tree to the other.