MarlinFirmware/Marlin #2754: Selectet move axis 10mm <> axis move 40mm (RC2)

chaoscrack, 2015-11-02:
If I select "move axis 10mm", the selected axis traveled 40mm. Also the display shows that the axis has moved 40mm.

My settings are:
- GT2 2mm belt
- 16T pulley
- 1/32 microsteps
- NEMA17 1.8°
- 200 steps/mm (at firmware)

Did I forget to set a parameter?

KiteLab, 2015-11-02:
16T/U * 2mm/T = 32mm/U.
360°/U / 1.8°/step =   200 steps/U.
200 steps/U * 32 µsteps/step = 6400 µsteps/U
6400 µsteps/U / 32mm/U = 200 µsteps/mm.

The calculation looks correct.

When your x and y axis is belt driven, it should look like
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {200,200,... ,... }
```

As always, knowing your branch and your Configuration.h could help to analyse your issue.

chaoscrack, 2015-11-02:
Yes, the axis is belt driven and I put the value 200 to the firmware.

The branch is the RC2 form 1.1.0

Unfortunately I can't upload the file (only png, gif, jpg). Which part of the file do you want see?
The stepper travel correct. It's just when I move the stepper by the panel. Than is it 40mm instead of 10mm.

brainscan, 2015-11-02:
Are you sure you aren't moving the encoder 4 steps? My friend bought a cheap i3 and it's a nightmare trying move the head manually as every click seems to register two or three steps.

AnHardt, 2015-11-02:
@brainscan 
The solution for the multiple steps per raster position of the encoder is tweaking ENCODER_PULSES_PER_STEP.

closed 2015-11-24 by AnHardt