MarlinFirmware/Marlin #2759: Some {} got messed up with FAST_PWM_FAN

Lukelectro, 2015-11-05:
In branch RCBugfix:

Marlin_main.cpp: In function ‘void setPwmFrequency(uint8_t, int)’:
Marlin_main.cpp:7167: error: a function-definition is not allowed here before ‘{’ token
Marlin_main.cpp:7208: error: expected `}' at end of input

This is because I enabled FAST_PWM_FAN, and it causes  void setPwmFrequency(uint8_t pin, int val) { to be included which is missing a }.

} // bugfix
#endif // FAST_PWM_FAN

To reproduce this bug enable FAST_PWM_FAN in configuration.h

I'll see if I can make a good pull request this time, but it is probably faster if someone else just includes that } on line 7164 in marlin_main.cpp

AnHardt, 2015-11-05:
PR #2732 is already there. Needs only one who merges that. (I really think i should not merge my own PRs)
Duplicate of https://github.com/MarlinFirmware/MarlinDev/issues/248
Already fixed in MarlinDev/master by https://github.com/MarlinFirmware/MarlinDev/pull/251

Thanks for the hint.

Wackerbarth, 2015-11-05:
PR#2732 merged

closed 2015-11-05 by Wackerbarth

Lukelectro, 2015-11-05:
Ok, that was quick :)

Now, on to find some other bug... Hmm, Lets try M206 again...