MarlinFirmware/Marlin #2761: Z_MIN_POS gets ignored

Lukelectro, 2015-11-05:
My Mendel90 homes to top (like all mendel90's) and I use M206 to set a home offset. Zmax is set to 190, Zmin to 0.

I Home to 190. Then I lower Z to 20, then carefully down onto the bed. M114 report 17.10 when on the bed. So I enter M206 Z-17.10. Home again, now M114 reports home is 172.90, as expected. Z offset is applied correctly.

However, when moving down again it is possible to get to Z < 0 (and into the bed).

Software endstops are enabled.

I searched for Z_MIN_POS in open issues to see if there was a dupplicate, found none but might have overlooked it. Not up to speed yet.

Actually this might be useful behaviour when tuning Z offset... But it is not what I expected with Z_MIN_POS 0, that should afaik limit manual moves.

thinkyhead, 2016-02-10:
That is the correct behavior for `home_offset`. It doesn't limit movement, but only affects the numerical position value. To set movement limits you use the min and max positions.

You should set your Z home position so that when you lower the nozzle to the bed the position is Z=0. This should be the result you get when home offsets are zero (as with `M206 X0 Y0 Z0`). Thus you should set `#define Z_MAX_POS 172.90`.

Does that make sense?

closed 2016-02-10 by thinkyhead

Lukelectro, 2016-02-10:
Yes, makes sense. Documentation issue indeed.

thinkyhead, 2016-02-11:
> Documentation issue indeed.
Documentation is always an issue, isn't it? I hope to have the time and resources to make great Marlin documentation after this release, including some videos, but I would love to get more help on that. I'll post a request for "help!" pretty soon.