MarlinFirmware/Marlin #2762: Problems in checkHitEndstops()

AnHardt, 2015-11-06:
a. In the `checkHitEndstops()` (https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.cpp#L252) we don't throw any messages for max-end-stops.
b. For deltas the Z_MIN and Z_MIN_PROBE messages only show the state of the z-tower - what is basically meaningless.

c. If a hardware end-stop is hit we do basically nothing but throwing the message, except  `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED` && `SDSUPPORT` and `abort_on_endstop_hit` is true.
I think we should stop the printer, or at least the move, if we hit a hardware end-stop before the software end-stops are established. That means before the machine is homed. That needs backing of from the hardware end-stops if they are triggered in the init phase.

AnHardt, 2015-11-06:
To solve a) we could simply replace `BIT(X_MIN)` with `(BIT(X_MIN) | BIT(X_MAX))` for all 3 axes.

b) needs an extra DELTA case. Here we could print endstops_trigsteps[] for all 3 axis. For x and y they have to be calculated in `update_endstops()` in a extra DELTA case close to where `              endstops_trigsteps[Z_AXIS]` is stored.

Roxy-3DPrintBoard, 2015-11-06:
I think I agree with everything you said here.   In the case of Delta's, if we see an end stop get hit unexpectedly, don't we want to freeze the machine?

I don't really care where we discuss this, and mostly I'm asking because I need help understanding...  But shouldn't this thread be over in MarlinDev?

AnHardt, 2015-11-07:
Sorry for the partially false alarm.
Solution a. is crap. `update_endstops()` sets the MIN-flag regardless if it has detected a min- or max- endstop-hit.
solution b. is crap. Output of the "X:", "Y:", "Z:" for deltas would be the tower coordinates and mess up the host program with wrong information, because this will interpret the coordinates as cartesian coordinates.

This makes an other error obvious. On a delta we currently sent the tower coordinate of z when we hit the z-min-endstop.
If " Z_MIN_PROBE:" is also interpeted by Repetier () here it is the same.

I guess we should not do this outputs in `checkHitEndstops()` for deltas at all.

Roxy-3DPrintBoard, 2015-11-07:
OK.    I'm wondering if there is an unintended consequence going on.    I've seen the location of the origin shift during the Z-Probes during the M48.    I don't have enough information yet to understand what is happening.   But I'm wondering the Z-Tower is getting its location set, and that is pushing the location over????

AnHardt, 2015-11-07:
@Roxy-3DPrintBoard 
I have a rough idea where this behavior could come from. But it's not completely analysed jet, nor i have a fix. I already wrote something down at: https://github.com/MarlinFirmware/Marlin/issues/2722#issuecomment-154585216

thinkyhead, 2016-02-10:
Any patch or proposed patch for this yet? Seems like it might be an important thing to fix.

closed 2016-04-23 by jbrazio