MarlinFirmware/Marlin #2763: Heated bed temperature out of control

ProgrammedSoul, 2015-11-06:
[image]

I was using Marlin 1.0.2 and i got this issue, then I updated to Marlin 1.1.0-RC see if it fix it, the answer is no.
Basically what happened is, when the heated bed reached the target temperature, it stays on that temperature normally. However, the moment the motors starts to move the heated bed begins to heat as much as it can reach. I tried bang-bang and PID config both, nothing changed.
Also it seems that heated bed heats up slower in 1.1.0=RC than it would do in 1.0.2, you can see that from the heating curve of the bed (before and after it goes out of control....).
The hardware is Printrboard Rev.D, controlled by a Raspberry Pi B2 with Octoprint. The temperature sensor on heated bed is a 10k sensor that I borrowed from my lab. The heated bed is a home made heated bed, heated by five 32 gauge nichrome wire in parallel, with a total resistance of 1.7ohm.

ProgrammedSoul, 2015-11-06:
By the way the temperature reading is not caused by sensor error, I confirmed that by using a infrared thermometer. The bed can reach up to 100 C.

Roxy-3DPrintBoard, 2015-11-07:
When I print ABS...  105 C works good for me.   But what ever you set it at...  That is where is should stay.

ProgrammedSoul, 2015-11-07:
Well I print PLA, and when the heated bed is too hot, the part will warp as the print finished, as the part cools to fast.
Anyway this is not something that supposed to happen, right?

AnHardt, 2015-11-07:
My interpretation of the curve is different.
-21min: The bed is set to 60 and begins to heat. The aproach to 60 is slow because of the used PID parameters.
-8min: The bead temperature reaches 60. The hotend target temperature is set to 200. The BED starts to heat again with full power. We can't see that, for what reason ever. But you can not explain the jump at -5:30 otherwise.
-6:30min: The hotend temperature begins to rise. Or at least we can see it from here on.
-6min: After overshooting a bit the hotend temperature reaches 200. We get the bed temperature reading back. The bed is still heating.

Is it your bedleveling what blocks the temperature readings from -8 to -6?

My guess is, that your bed is couppled with extruder heater. When the bed-heater-pin is on only the bed heats. If your extruder-heater-pin is on, the extruder and the bed is heating.

Try to heat the extruder only and look what the bed-temperature is doing.

whosawhatsis, 2015-11-07:
Marlin doesn't report changes in bed temperature while an M109 is in effect. If the bed is heating while you're waiting for the extruder to reach temperature, it's not unusual to see the temperature appear to jump as soon as M109 returns and normal temperature reporting resumes.

AnHardt, 2015-11-07:
Yep. I guess we see a 
M190 S60
M104 S200
(G28)?
G29
M107 S200

AnHardt, 2015-11-07:
@ProgrammedSoul 
Do you have a external logic/driver to switch your bed power?
I yes, i'd look there at first.
If not, we need to see your Configuration.h, pins_PRINTRBOARD.h(if changed). pins.h(if changed) and maybe Configuration_adv.h(if changed).

ProgrammedSoul, 2015-11-07:
I did not use a external driver, right now the bed is powered by Printrboard on board MOSFET.
The end stop is not implemented yet so leveling and homing is done by hand and there is no G28 command presented.
I tried only heat up the extruder only and it seems fine....
[image]

ProgrammedSoul, 2015-11-08:
I do not think I changed the pin file, I'll add it as a reference anyway. I do changed a lot config options in Configuration.h, but most of them are not related to heated bed.
http://pastebin.com/bXAKqHSx
http://pastebin.com/dKJ3c8xz

KiteLab, 2015-11-08:
```
// Select PID or bang-bang with PIDTEMPBED. If bang-bang, BED_LIMIT_SWITCHING will enable hysteresis
#define PIDTEMPBED
 #define BED_LIMIT_SWITCHING
```
Please switch of one of these. Theses are different methods to control the heater. They should exclude each other.

KiteLab, 2015-11-08:
When using a custom made bed you really should tune your beds PID parameters.

ProgrammedSoul, 2015-11-09:
I tuned the bed (and yes, the tuned measurements are miles from the default), it reeally helps to heat the bed up fast. I also commented the BED_LIMIT_SWITCHING so only PIDTEMPBED is enabled now, sadly it did not fix that...
[image]

The beginning of the gcode file looks like this (I think it's short enough to just paste it here):
G90
M82
M106 S0
M140 S60
M190 S60
M104 S200 T0
M109 S200 T0
G92 E0
G1 E-1.0000 F1800
G1 Z0.171 F1000
; layer 1, Z = 0.17145
T0
; tool H0.190 W0.300
; skirt
G1 X2.287 Y39.055 F4800
G1 E0.0000 F540
G92 E0
G1 X173.779 Y39.055 E4.0747 F900
...............

KiteLab, 2015-11-09:
So the M303 for the bed worked?
The new graph shows us a never switched off bed heater. (Extrapolating the curves)

The next test could be:
M104 S200 (wait until you see the extruder graph rising)
M109 S60  (wait until you see the bed graph rising)
M109 S0  (wait until you see the bed graph falling)
(If the bed graph is not falling after about 2 minutes) M104 S0
What happens now?

The terminal/tab tells a few more details then the graph if you turn off `Suppress M105 requests/responses`

An other test (for the parser this time) is to remove the `T0` parameters from the M104 and M140 lines in the g-code-file.

The M140 line is not needed before the M190 line.
The M104 line is not needed before the M109 line.

ProgrammedSoul, 2015-11-09:
Before I post anything, is M109 supposed to change the extruder temperature instead? When I executed M109 nothing happened to the bed, and the target temperature of extruder is changed from 200C to 60C.

KiteLab, 2015-11-09:
The definitions are:
M104: Set extruder temperature
M140: Set extruder temperature and wait
M109: Set bed temperature
M190 Set bed temperature and wait

I can not reproduce this error. For my system M109 changes the bed target temperature.

You could try to take a fresh copy from RC, copy in your configuration, load up and test again.

ProgrammedSoul, 2015-11-09:
First of all thanks for all of your attention and patience, I fell really pleasant with the fact that someone is trying to help me out. And please bare with me, I think we are getting somewhere....

In reprap wiki it says M109 means "Set Extruder Temperature and Wait", and M104 is "Set Extruder Temperature". M190 is "Wait for bed temperature to reach target temp" and M140 is "Set Bed Temperature (Fast)". It works exactly the same in my firmware, and you are telling me thant M109 is set bed temperature instead of M140 **(our definition of M140 and M109 is shifted)**. It's really making me confusing.

If M109 is set bed temperature then it would explain everything, because in my gcode, in the line "M109 S200 T0", T0 will just got ignored and the bed target temperature is set to 200C, which it will never reach, so it will never switch off. But as you can see in the curves the bed temperature is set to 60C instead of 200C, and if I run M109 S200(no matter if there is a T0), it heats up the exturder. The target temperature shows correctly and the extruder heats up correctly. I swore I did not change anything that could change gcode behaviors. I did not changed pin config and just changed a few comments and updated some essential values in configuration.h.

KiteLab, 2015-11-09:
Sorry, I messed it up completely.
From MarlinMain.cpp - the definite source:
```
 * M104 - Set extruder target temp
 * M105 - Read current temp
 * M106 - Fan on
 * M107 - Fan off
 * M109 - Sxxx Wait for extruder current temp to reach target temp. Waits only when heating
 *        Rxxx Wait for extruder current temp to reach target temp. Waits when heating and cooling
 *        IF AUTOTEMP is enabled, S<mintemp> B<maxtemp> F<factor>. Exit autotemp by any M109 without F
 * M110 - Set the current line number
 * M111 - Set debug flags with S<mask>. See flag bits defined in Marlin.h.
 * M112 - Emergency stop
 * M114 - Output current position to serial port
 * M115 - Capabilities string
 * M117 - Display a message on the controller screen
 * M119 - Output Endstop status to serial port
 * M120 - Enable endstop detection
 * M121 - Disable endstop detection
 * M126 - Solenoid Air Valve Open (BariCUDA support by jmil)
 * M127 - Solenoid Air Valve Closed (BariCUDA vent to atmospheric pressure by jmil)
 * M128 - EtoP Open (BariCUDA EtoP = electricity to air pressure transducer by jmil)
 * M129 - EtoP Closed (BariCUDA EtoP = electricity to air pressure transducer by jmil)
 * M140 - Set bed target temp
 * M145 - Set the heatup state H<hotend> B<bed> F<fan speed> for S<material> (0=PLA, 1=ABS)
 * M150 - Set BlinkM Color Output R: Red<0-255> U(!): Green<0-255> B: Blue<0-255> over i2c, G for green does not work.
 * M190 - Sxxx Wait for bed current temp to reach target temp. Waits only when heating
 *        Rxxx Wait for bed current temp to reach target temp. Waits when heating and cooling
```

ProgrammedSoul, 2015-11-09:
That's totally fine, I'm just going back on track.
So I'm assuming what you want me to do is:
**"**The next test could be:
M104 S200 (wait until you see the extruder graph rising)
**M140** S60 (wait until you see the bed graph rising)
**M140** S0 (wait until you see the bed graph falling)
(If the bed graph is not falling after about 2 minutes) M104 S0
What happens now?**"**
The result is (the bed do fall, but I do M104 S0 anyway):
[image]
Terminal output:
[...]
Changing monitoring state from 'Operational' to 'Closed'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x2ae7f90, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
[...]
Send: M21
Recv: echo:SD init fail
Recv: ok
[...]
Send: M104 S200
Recv: ok
[...]
Send: M140 S60
Recv: ok
[...]
Send: M140 S0
Recv: ok
[...]
Send: M104 S0
Recv: ok
[...]

I'm not sure what did means did M303 works, I thinks it works for me. It returned "Kp=404.89, Ki=68.65, Kd=597.01", and when I put those values in config the bed heats up much faster (if you see closely the temperature of bed do keeps a while before the extruder heats up, I pre-heated it to target temperature for testing purpose).
T0 seems got ignored in this case and weather having or not having T0 does not changed the behavior.

AnHardt, 2015-11-09:
M303 worked in the sense of the bed heater switched of and was not heating to the infinite.
The test right above also worked in this sense.

~~Because the test worked the terminal output is not very interesting. For the next test, please don't edit out all the M105 answers, please let one above and one below the commands. That's where the extra info, like target temperatures and heater-power for all heaters, is in.~~

I'll prepare a patch for you.

ProgrammedSoul, 2015-11-10:
Thanks a lot for that. Here is some of my test result:

I borrowed a oscilloscope and hock it up to the gate of the heated bed MOSFET. Still, it behave normally, it stays on 0V while have nothing to do, it stays at 5V while heating the bed up, and it gives out PWM when keeping the temperature. However when i load the printing file again, the moment the extruder starts heating, the gate of heated bed goes to full 5V (sorry I cannot make sure if that's the exact moment because I'll running back and forth between 2 bench, 3m from each other and like 20 cables sitting in between). So what I did next is turn off the heated bed and extruder, paste in the gcode line-by-line. Then I saw right after M190 S60, the recv information **is the temperature of the extruder instead of the bed, and bed is not heating up at all, and the target temperature of the extruder did not changed**. Then it keeps what it be and not accepting any command (probably because the server is waiting for the ok message which is never gonna happen), until I cut the power.
I'm pretty sure something went wrong here, and you probably know it batter than I'm.

AnHardt, 2015-11-10:
Please download https://github.com/AnHardt/Marlin/archive/M105andCo.zip, apply your config and test your original g-code-file.
More info about the patch https://github.com/MarlinFirmware/Marlin/pull/2768

ProgrammedSoul, 2015-11-10:
[image]
Here is the test result:
Update the firmware, load the gcode and print (~-26min), bed never turns off. Cancel the print(~-20min), manually set the bed temperature to 60(~-19min), bed no longer heats up, printer does not accept amy commands other than M105.
Reset the power, manually set bed temperature to 60C(~-12min), the bed temperature stays at 60C. Load and print the gcode, the bed temperature run away at the moment the extruder heats up.
Remove "T0" from the gcode(that's the only variable I could find). Reset the power, manually set bed temperature to 60C, bed temperature stays(~-2min). Then manually set the extruder temperature, extruder temperature and bed temperature stays. Then load the gcode, bed temperature run away before the motor starts moving(~-1min).
There must be something going wrong here. The firmware is showing different behaviors between manually entered gcode and loaded gcode file.
I'm setting temperatures manually using M104 and M140. The beginning of the gcode file is shown as:
G90
M82
M106 S0
M140 S60
M190 S60
M104 S200
M109 S200
G92 E0
G1 E-1.0000 F1800
G1 Z0.171 F1000
; layer 1, Z = 0.17145
T0
; tool H0.190 W0.300
; skirt
G1 X2.287 Y39.055 F4800
..................

Wurstnase, 2015-11-10:
Can you shut off your fan the hole time and test again? Maybe you have a short between your heatbed and the fan.

ProgrammedSoul, 2015-11-10:
The fan never turns on over the whole test. My extruder design is well vented and using the fan on tirst layer can lead to serious warp, so I only turn the fan on at line layer 5.

ProgrammedSoul, 2015-11-12:
I removed everything. Without any M106, M140, M190 and M104, preheat the temperature maunally, and the temperature still run away at the beginning of printing.

ProgrammedSoul, 2015-11-12:
Ok i got it. This is my fault from a long time ago, but it do expose a problem. A long time ago, before I build my printing bed, I was using heated bed MOSFET for some extra cooling fan, so I add this scripts to octoprint to automate it:
[image]
And after a long while I actually moved this function to my slicer, and removed it from octoprint. But recently a fatal power loss broke the raspberry SD image, so I have to roll back the image, which restored the octoprint start script without my notice.
And that script is causing problems. It overrides the IO state of the heated bed MOSFET gate pin and forced it always stays on.
It sounds like all my fault.
BUT, this pointed out a bug. When I was learning about M42 command, the website, and the marlin source code shows that it will not change any pin that already assigned in pin.h to prevent accidental operation (source code shows that if the pin number is a functional pin, Marlin will change it to LED pin instead). But this safety function is not working. M42 uses arduino diditalWrite pin number, but a majority of functional pin number in pin.h uses your coston fastio.h, this might be the gap that causing that problem.

AnHardt, 2015-11-19:
@ProgrammedSoul 
```
/**
 * M42: Change pin status via GCode
 */
inline void gcode_M42() {
  if (code_seen('S')) {
    int pin_status = code_value_short(),
        pin_number = LED_PIN;

    if (code_seen('P') && pin_status >= 0 && pin_status <= 255)
      pin_number = code_value_short();

    for (uint8_t i = 0; i < COUNT(sensitive_pins); i++) {
      if (sensitive_pins[i] == pin_number) {
        pin_number = -1;
        break;
      }
    }

    #if HAS_FAN
      if (pin_number == FAN_PIN) fanSpeed = pin_status;
    #endif

    if (pin_number > -1) {
      pinMode(pin_number, OUTPUT);
      digitalWrite(pin_number, pin_status);
      analogWrite(pin_number, pin_status);
    }
  } // code_seen('S')
}
```
With a RAMPS I could not reproduce your error. 
The mapping was ok for the pins of the Mega i tried.
When i removed the `for ()` loop above, i was able to set the `HEATER_BED_PIN`, but that had no bad consequences, because all of the 3 heater algorithms corrected the heater output immediately.
Finally i got the idea to set the` FAN_PIN` to the same pin number as the `HEATER_BED_PIN`. In that case i could set the `FAN_PIN` to 255, the `#if HAS_FAN` came into effect, fanSpeed was set and the fan algorithm overrule the heater algorithm, and finally I got a unlimited heating bed.

Tried to shift fan-writing before bed-heater-writing in the heater-ISR. That helped, but only for fans using `FAN_SOFT_PWM`. If the fan uses a pwm-pin the update is effectively handled in the idle loop. For that i don't now any way to coordinate the writing with the heaters. So sorry, no good solution from this side.
The real problem is your pin mapping between Arduinio-IO and FastIO
Providing a correct `pins_arduino.h` or compiling in the correct one is beyond my scope. Sorry.
Maybe the new environment in MarlinFirmware/MarlinDev can help here.

Is there the smallest possibility you defined the `HEATER_BED_PIN` to the same pin as `FAN_PIN`? If so, that would help me to sleep better.

CONSULitAS, 2015-11-21:
@Anhardt

What about

SanityCheck.h:
---
// HEATER_BED_PIN may never be same pin as FAN_PIN (see https://github.com/MarlinFirmware/Marlin/issues/2763)
#if HAS_FAN
  #if HEATER_BED_PIN == FAN_PIN
    #error You cannot set HEATER_BED_PIN equal to FAN_PIN. This would cause uncontrolled Heating when turning the fan on and resulting risk of fire.
  #endif
#endif
---

@ProgrammedSoul: Could you please insert this snippet into SanityCheck.h and post the results?

BTW: Did not find a HAS_HEATER_BED (just a short check), if there is one it should be #if HAS_FAN && HAS_HEATER_BED.

AnHardt, 2015-11-22:
@CONSULitAS 
That really should have been caught by the for loop. (wrong pin mapping)
If not caught there, we need to check also for the extruder heaters.

AnHardt, 2015-11-22:
Edited the my last message.

ProgrammedSoul, 2015-11-22:
Sorry for no response. I actually changed the pin mapping again as I proceed on my project. I'll change the firmware back and test it out as soon as I got a chance.

thinkyhead, 2016-02-21:
Doing a little housecleaning. What is the status of this issue?

AnHardt, 2016-02-21:
If FastIO and Arduino pin mappings are different we need a mapping-table.
Normally we use FastIO pin mapping in our pin files. `analogWrite()` `digitalWrite` `digitalRead()` use Arduino pin mapping.  `digitalWrite` `digitalRead()` can be replaced (if the pin is known at compile time (not a variable)) by `READ()` `WRITE()` from FastIO. There is no replacement for `analogWrite()`.
So if we do not have and use adapted `pins_arduino.h` files in the core we have to use an extra mapping.

Still a bit confusing to me. So no real idea how to fix.

Roxy-3DPrintBoard, 2016-02-21:
It seems like there should be an easily available mapping table between the
two different pin modes.

AnHardt, 2016-02-21:
I guess it's done magically if the right `pins_arduino.h` files are used.
I scare, if we do that mapping by our selves, we have to do in in every pin-file, or at least one per variant.

ProgrammedSoul, 2016-02-21:
No I tried I can still use M42 to control the heated bed even it is claimed as a functional pin in pins.h.
This problem can be easily avoided through...

thinkyhead, 2016-02-28:
There's nothing in Marlin to prevent messing around with pins that you shouldn't using `M42`. Maybe there should be!

Blue-Marlin, 2016-02-28:
@thinkyhead 
You may have not seen
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/pins.h#L251
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L3518

Sadly this does not work in __every__ case.

thinkyhead, 2016-03-02:
> You may have not seen…

Veni. Vidi. Vici.

closed 2016-05-04 by jbrazio