MarlinFirmware/Marlin #2764: Question RE E steps/

ruggb, 2015-11-07:
DEFAULT_AXIS_STEPS_PER_UNIT in Configuration_h defines E axis steps/unit (mm)
There is also a STEPS_MM_E defined in Configuration_h_adv
Instead of redefining this and having 2 variables for the same parameter, would it not be better to use the definition in _H for the ADVANCE function?

If so, the same would apply for the filament diameter
DEFAULT_NOMINAL_FILAMENT_DIA in _H
and D_FILAMENT in _H_adv

Since these are inside if stmts and used in multiple places they should be located at the top.
Just saying... defining the same thing in 2 places is a major source for future issues.

AnHardt, 2015-11-24:
`ADVANCE` is considered as broken. Reimplementation is needed. When that will be done we'll look after this. `DEFAULT_AXIS_STEPS_PER_UNIT` and `STEPS_MM_E` have different units. They may be calculated from each other but are not the same.

ruggb, 2015-11-24:
Well, I guess then I am totally confused.
now I have no idea why I enabled this and set the parameters as I did or what it is suppose to affect.
do i need to disable this?

AnHardt, 2015-11-25:
If you don't know what `ADVANCE` is - you do not need it

ruggb, 2015-11-25:
well I thought I knew - can u give me a clue as to some info on it?

Roxy-3DPrintBoard, 2015-11-25:
AnHardt obviously knows more about this than I do.   But I remember a discussion where NopHead was experimenting with stuff to smooth out the amount of material the extruder actually put on the print.  There are some weird issues that happen with the retraction and re-engagement of the filament.  I don't know (and lets wait for somebody that really knows to offer an opinion) but I think this was an effort to fix some issues with retraction.

lrpirlet, 2015-11-26:
There is a discussion of the advance here https://github.com/MarlinFirmware/Marlin/issues/916... The pdf is not accessible anymore :(

            The very first page of the wiki,  <http://reprap.org/wiki/Marlin> http://reprap.org/wiki/Marlin has a reference to advance "preliminary support for Matthew Roberts advance algorithm For more info see: http://reprap.org/pipermail/reprap-dev/2011-May/003323.html"

 

            I found quite a few talk about the subject just google "marlin advance"

 

            I was really interested by the concept of regulating the pressure in the head function of the speed to fight inertia (this is how I understood it! )

 

            Enjoy.

ruggb, 2015-11-27:
can someone tell me what STEPS_MM_E is?
And/or how is it determined?
I assumed it was STEPS/MM for E but maybe not.
thx

Nandox7, 2015-11-27:
Funny that the discussion about this feature keeps coming back again and again. :)

Most of the slicers now have features that in a way mimic this BUT the slicer is never aware of the speed and acceleration settings in the FW so I think there is still need for it.
I had a initial test, where I started looking at what sailfish did and porting it to marlin but never ended it.

@ruggb yes that is correct. steps/mm for the E. That is duplicated and can be obtained from the main settings.

ruggb, 2015-12-08:
anhardt said they r not the same - different units - but no definition of such.
I guess it is really broken and as such should be totally removed if no one can define it.

KiteLab, 2015-12-08:
STEPS_MM_E is defined in 15 config. files but is never used.
What is used in the advanced algorithm, is STEPS_PER_CUBIC_MM_E.
That is defined in `Conditionals.h` as `#define STEPS_PER_CUBIC_MM_E (axis_steps_per_unit[E_AXIS] / EXTRUSION_AREA)`. `EXTRUSION_AREA` is defined one line above as `#define EXTRUSION_AREA (0.25 * D_FILAMENT * D_FILAMENT * M_PI)`
Its counterpart `DEFAULT_NOMINAL_FILAMENT_DIA` was introduced with the `FILAMENT_SENSOR` and is used only in that context.
Believe it or not; your slicer needs the same value an other time.

Anyone else here with the source code of these mysterious Marlin

Wackerbarth, 2015-12-08:
Should we remove "STEPS_MM_E" to reduce the confusion?

KiteLab, 2015-12-08:
I think so.

Roxy-3DPrintBoard, 2015-12-08:
If we don't remove it, we should put a very descriptive comment on it so people know exactly what it is.

Wackerbarth, 2015-12-08:
@Roxy-3DPrintBoard -- As pointed out earlier in this thread, although it is defined in Configuration_adv.h, it is NOT USED in any of the code.

Rather than "documenting" a non-feature, I think it would be better to eliminate it and, when/if needed, introduce a different, more descriptive name.

KiteLab, 2015-12-08:
Advance should use the filament diameter of the active extruder anyway. But that's only one of the many subsystems what can't handle __different__ extruders

thinkyhead, 2016-02-23:
I've submitted PRs to remove the `STEPS_MM_E` setting.

It does look like the `D_FILAMENT` setting could go away, with `EXTRUSION_AREA` calculated from `DEFAULT_NOMINAL_FILAMENT_DIA`. Although `DEFAULT_NOMINAL_FILAMENT_DIA` was originally used in the context of `FILAMENT_SENSOR`, that is no longer the case. `DEFAULT_NOMINAL_FILAMENT_DIA` has been in use by the EEPROM code as a default when writing, even if `FILAMENT_SENSOR` is not enabled.

I discovered these facts when I submitted MarlinFirmware/MarlinDev#359 grouping the `FILAMENT_SENSOR` options together. `DEFAULT_NOMINAL_FILAMENT_DIA` should be considered a standard global option at this point.

So,

AnHardt, 2016-02-23:
@thinkyhead 
That's a good plan.

PR #3024 (2016-02-24): Remove obsolete STEPS_MM_E setting
This resolves #2764
closed 2016-03-10 by thinkyhead