MarlinFirmware/Marlin #2765: Clarification on how to enable endstops when the sensor probe and autoleveling are enabled

albertolg, 2015-11-08:
Forgive me buit I can't find how to have the endstops and the bed leveling working togeter. Right now to use the autolevel bed feature I need to uncomment the ENDSTOPS_ONLY_FOR_HOMING in the confiiguration_adv and I will lose all the endstops after the homing.

AnHardt, 2015-11-10:
Despite the name ENDSTOPS_ONLY_FOR_HOMING this only switches the evaluation of the __hardware__ endstops on and off. The __software__ endstops are not switched off. There are some problems with the software endstops, especially with delta printers, but else they are supposed to work.

albertolg, 2015-11-10:
yes, my fault, bad explanation, the software endstops keeps working, I was refering to the hardware switches, there is nothing to change to keep the hardware endstops enabled? I'm using the 1.1.0RC2 thanks

AnHardt, 2015-11-24:
Sorry bad explanation.
Despite the name `ENDSTOPS_ONLY_FOR_HOMING ` hardware endstops, as well as the probe,  work for probing, when `ENDSTOPS_ONLY_FOR_HOMING ` is defined.

thinkyhead, 2016-02-22:
> nothing to change to keep the hardware endstops enabled?

Did you get this figured out? If you still need help please re-open this issue.

closed 2016-02-22 by thinkyhead