MarlinFirmware/Marlin #2766: v 1.1.0 RC2 - Z homing and bed levelling not working with CoreXZ

GinaDavis, 2015-11-09:
I have just modified my RepRap style 3D printer to CoreXZ and though the normal movements are fine, the Z homing and bed levelling doesn't work properly - the X carriage only moves upwards. Bed levelling was working fine before converting to CoreXZ.

Wurstnase, 2015-11-10:
The problem of that is ```zPosition = st_get_position_mm(Z_AXIS);``` in Marlin_main.cpp. This is the right way for cartesian printers and axis which are direct driven, like Z-axis on a CoreXY or cartesian printers. But on a CoreXZ this is only the position of the motor. Not of the axis/head.

You can see the different when using ```M114```. The former is the position of your head, the latter the position of the motors.

GinaDavis, 2015-11-10:
Thank you for the explanation.

mkubik, 2015-11-15:
Actually, I have the same issue but neither CoreXZ nor CoreXY defined. I use this one here:http://www.thingiverse.com/thing:301715
But when I run G29,  I only see the extruder move up, not down.
Not sure if this is really the same, but the behavior is similar.
I may be missing something, I admit, so any hint is welcome.
Here's my config:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/34906/Configuration.h.txt)

KiteLab, 2015-11-15:
@mkubik 
Your probe is triggered when the servo is in its upper position?
That's a common problem. The hardware endstops are activated before the probe is deployed.

mkubik, 2015-11-15:
@KiteLab 
Yes, you're right.  Any workaround you can point me to?

KiteLab, 2015-11-15:
#2769 is not ready to solve all the hardware endstop problems. But your problem could already be solved.

Roxy-3DPrintBoard, 2015-11-15:
Alternatively...   I ran into this problem with the Delta printer I'm trying to bring up.   I tried every configuration option I could think of and ended up just changing the code.   (I think I just did not understand the ALLEN_KEY options well enough at the time.)

If push comes to shove and you can't make it go using the configuration options, I can send you the code that only looks at the Allen Key if the Z-Probe is actually deployed.

Wurstnase, 2015-11-15:
Sorry guys. But this problem is a different one as from the OP. Please make a new issue for that.

thinkyhead, 2016-02-11:
> The problem of that is zPosition = st_get_position_mm(Z_AXIS); in Marlin_main.cpp

@Wurstnase I'm working on a patch for this now. It involves an awful lot of note-taking!

Wurstnase, 2016-02-11:
Great, the st_get_position() is a good way to go. 

The st_get_pos_mm works only on cartesian printers and I think is not a good naming for it. Steppers don't know any positions in mm.

thinkyhead, 2016-02-11:
@Wurstnase Core support is more broken than I thought, but I think it will be relatively easy to patch. There are several places where I'll need to make corrections:

- When an endstop is triggered in `update_endstops` we do `endstops_trigsteps[axis] = count_position[axis]` — Then `endstops_trigsteps` is used when displaying the axis positions in `checkHitEndstops` (`stepper.cpp`). It is directly divided by `axis_steps_per_unit[axis]` to get mm.
- The `M114` command prints the current position according to the explicit coordinates, followed by the `st_get_position_mm` for each axis.
- The `run_z_probe` function calls `st_get_position_mm` directly to get the Z position.
- So does the `M48` handler.
- So does the `G29` handler.
- The `plan_get_position` function uses `st_get_position_mm` to get the raw XYZ coordinates.

I did a search for `st_get_position_mm ` and found that there's no place in the code where it is specifically used to get a raw stepper (not axis) position. Everyplace where it's called expects a standard Cartesian result. I also searched for `st_get_position` and found no change needed around that function. DELTA code uses it to get the raw Z steps.

So… I believe I can derive the Cartesian position from the stepper position pretty easily, but correct my maths if I'm totally off. Converting `st_get_position_mm` should give us something like this:
```cpp
float st_get_position_mm(AxisEnum axis) {
  float axis_pos = st_get_position(axis);
  #if ENABLED(COREXY)
    if (axis == X_AXIS)
      axis_pos = (axis_pos + st_get_position(B_AXIS)) / 2; // ((x+y)+(x-y))/2 -> (x+y+x-y)/2 -> (x+x)/2
    else if (axis == Y_AXIS)
      axis_pos = (st_get_position(A_AXIS) - axis_pos) / 2; // ((x+y)-(x-y))/2 -> (x+y-x+y)/2 -> (y+y)/2
  #elif ENABLED(COREXZ)
    if (axis == X_AXIS)
      axis_pos = (axis_pos + st_get_position(C_AXIS)) / 2; // ((x+z)+(x-z))/2 -> (x+z+x-z)/2 -> (x+x)/2
    else if (axis == Z_AXIS)
      axis_pos = (st_get_position(A_AXIS) - axis_pos) / 2; // ((x+z)-(x-z))/2 -> (x+z-x+z)/2 -> (z+z)/2
  #endif
  return axis_pos / axis_steps_per_unit[axis];
}
```

Roxy-3DPrintBoard, 2016-02-11:
> The run_z_probe function calls st_get_position_mm directly to get the Z position.
> So does the M48 handler.

Scott, does M48 have to change to be 'correct'.   Or do your patches fix everything so the higher level code does not need to change?

I think what I'm seeing is G29 and M48 never worked correctly for Core-XY machines.  Is that true?

Wurstnase, 2016-02-11:
> I think what I'm seeing is G29 and M48 never worked correctly for Core-XY machines. Is that true?

Yes, I think the same. I have never big issues in the past when using ABL, but my bed was relativ good leveled manually before.

AnHardt, 2016-02-11:
> I think what I'm seeing is G29 and M48 never worked correctly for Core-XY machines. Is that true?

Rubbish.  Core-XY has a conventional Carthesian z-axis. All you need for G29, M42 is z.
If you tried to talk about Core-XZ you are right.

AnHardt, 2016-02-11:
It works for deltas because in a __pure__ z-move all towers are doing the same steps. If not - chaos!

thinkyhead, 2016-02-11:
This is the PR I rolled for the dev branch: MarlinFirmware/MarlinDev#358

Wurstnase, 2016-02-12:
> Rubbish. Core-XY has a conventional Cartesian z-axis. All you need for G29, M42 is z.
> If you tried to talk about Core-XZ you are right.

I don't mean Z. I'm thinking about this section: https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L3589-L3590

thinkyhead, 2016-02-16:
@GinaDavis Hopefully this patch will solve your issue, if you are able to give it a try. The patch is linked above, and the full code is at this branch… https://github.com/thinkyhead/Marlin/tree/rcbugfix_corexy_corexz

closed 2016-04-23 by jbrazio