MarlinFirmware/Marlin #2767: Error: M109 S??? is waiting for cooling

KiteLab, 2015-11-09:
M109 S??? is waiting for cooling.
Normally only `M109 R???` should do so.

M190 works normally

AnHardt, 2015-11-09:
I can see this only when TEMP_RESIDENCY_TIME is defined.

I really don't know if `TEMP_RESIDENCY_TIME` should overrule the `S` = "non't wait when cooling" parameter, or not.
```
// Actual temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.
```

What is your interpretation?

thinkyhead, 2016-02-22:
So the current behavior for `M109 S180` is to wait for a temperature of 180 or higher to be reached. And it doesn't care if the temperature is 220°C — it will simply return. Meanwhile `M109 R180` should _always_ wait for the temperature to be reached, even if the temperature is higher.

Currently the `TEMP_RESIDENCY_TIME` setting always waits for the temperature to be within `TEMP_HYSTERESIS` of the target, regardless of whether you used `M109 S180` or `M109 R180`. It sounds like what you want is for `TEMP_RESIDENCY_TIME` to consider any temperature above the target to be within range when using the `S` (`no_wait_for_cooling`) option.

thinkyhead, 2016-02-22:
The updated logic would look like this…
```cpp
    #ifdef TEMP_RESIDENCY_TIME
      // start/restart the TEMP_RESIDENCY_TIME timer whenever we reach target temp for the first time
      // or when current temp falls outside the hysteresis after target temp was reached
      float deg_curr = degHotend(target_extruder),
            deg_target = degTargetHotend(target_extruder);
      if (residency_start_ms >= 0) {
        if (no_wait_for_cooling) {
          if (deg_curr <= deg_target - TEMP_HYSTERESIS)
            residency_start_ms = millis();
        }
        else if (labs(deg_curr - deg_target) > TEMP_HYSTERESIS)
          residency_start_ms = millis();
      }
      else if (target_direction || no_wait_for_cooling) {
        if (deg_curr >= deg_target - TEMP_WINDOW)
          residency_start_ms = millis();
      }
      else if (deg_curr <= deg_target + TEMP_WINDOW)
        residency_start_ms = millis();

    #endif //TEMP_RESIDENCY_TIME
```

AnHardt, 2016-02-22:
Or if we prefer a very compact code and easy to follow logic:

```
cancel_heatup = false;
+if (no_wait_for_cooling && !target_direction) return;
#ifdef TEMP_RESIDENCY_TIME
```

thinkyhead, 2016-02-23:
Makes sense! I will see about compacting it further.

thinkyhead, 2016-02-23:
Ok, so I've changed the code quite a bit. But I think it's the right idea. If you start `M190 S` or `M109 S` and the temperature is already above the target, these commands will simply exit without reporting any heater states. Thus, if the code continues after that test, it is either heating (in which case it should wait for heating) or it is cooling (and it should wait for cooling).

If `TEMP_RESIDENCY_TIME` is set then the temperature has to remain stable for a period of time. If it is not set, then the temperature must reach an exact (integer) match to exit the wait loop.
```cpp
// Exit if the temperature is above target and not waiting for cooling
if (no_wait_for_cooling && !isHeatingHotend(target_extruder)) return;

#ifdef TEMP_RESIDENCY_TIME
  long residency_start_ms = -1;
  // Loop until the temperature has stabilized
  #define TEMP_CONDITIONS (residency_start_ms < 0 || now < residency_start_ms + TEMP_RESIDENCY_TIME * 1000UL)
#else
  // Loop until the temperature is exactly on target
  #define TEMP_CONDITIONS (degHotend(target_extruder) != degTargetHotend(target_extruder))
#endif //TEMP_RESIDENCY_TIME

cancel_heatup = false;
millis_t now = millis(), next_temp_ms = now + 1000UL;
while (!cancel_heatup && TEMP_CONDITIONS) {
  now = millis();
  if (now > next_temp_ms) { //Print temp & remaining time every 1s while waiting
    next_temp_ms = now + 1000UL;
    #if HAS_TEMP_0 || HAS_TEMP_BED || ENABLED(HEATER_0_USES_MAX6675)
      print_heaterstates();
    #endif
    #ifdef TEMP_RESIDENCY_TIME
      SERIAL_PROTOCOLPGM(" W:");
      if (residency_start_ms >= 0) {
        long rem = ((TEMP_RESIDENCY_TIME * 1000UL) - (now - residency_start_ms)) / 1000UL;
        SERIAL_PROTOCOLLN(rem);
      }
      else {
        SERIAL_PROTOCOLLNPGM("?");
      }
    #else
      SERIAL_EOL;
    #endif
  }

  idle();

  #ifdef TEMP_RESIDENCY_TIME
    // Start the TEMP_RESIDENCY_TIME timer when we reach target temp for the first time.
    // Restart the timer whenever the temperature falls outside the hysteresis.
    if (labs(degHotend(target_extruder) - degTargetHotend(target_extruder)) > ((residency_start_ms < 0) ? TEMP_WINDOW : TEMP_HYSTERESIS))
      residency_start_ms = millis();
  #endif //TEMP_RESIDENCY_TIME

} // while(!cancel_heatup && TEMP_CONDITIONS)
```

AnHardt, 2016-02-23:
Much better. I think this time i understand whats going on. :-)

Some remarks
```
  // Loop until the temperature is exactly on target
  #define TEMP_CONDITIONS (degHotend(target_extruder) != degTargetHotend(target_extruder))
#endif //TEMP_RESIDENCY_TIME
```
This is hard to fail `degHotend()` and `degTargetHotend()` are float. Ether we have to cast to integers or we need a defined tolerance (TEMP_WINDOW ?). If the temperature is moving fast, there is a good chance to miss, even a tolerance, completely

This time i like the macros. I think the difference is, i don't have to look them up __somewhere__. 
If not defined clothe by, a first time reader has to look up a macro at one of about half a dozen places (at the top of xxx.cpp, Marlin.h, macros.h, xxx.h, Conditionals.h, somewhere in the standard librarys, ... ?). It's like learning a new language with undocumented librarys. Only when you got used to the (new) macros and continue to use them (to not forget) they are helpful. On the other side, if i have analysed a bit-shift pattern once or twice i can also recognize that.

thinkyhead, 2016-02-28:
> This is hard to fail degHotend() and degTargetHotend() are float

I vote for the simplest solution. The sample time is frequent enough that it should catch the temperature being "close enough" to the target value. Something like this would be okay, I should think…
```cpp
#define TEMP_CONDITIONS (abs(degHotend(target_extruder) - degTargetHotend(target_extruder)) <= 0.5f)
```

> if i have analysed a bit-shift pattern once or twice i can also recognize that

Some days I wish I could just yell at the computer in my native tongue and it would know what I mean.

AnHardt, 2016-02-28:
Jep. 1° is ok. If the first overshoot is to fast we get it on the way down.

I'm still not convinced by these macros. This afternoon i tried a variation of #2991 (axis_homed ...) and indeed could save 4 bytes of RAM, but spent ~100 bytes of program memory for them. Not a very good deal.

thinkyhead, 2016-02-29:
> could save 4 bytes of RAM, but spent ~100 bytes of program memory

Scientific comparison is good. I will have to do some profiling on the difference between bit-fields and `bool` arrays. It's probably true that `if (myBool)` is smaller and faster than `if (myBitMask & B0100 != 0)`. But is `if (myBool1 && myBool2)` faster and smaller than `if (myBitMask & B0110 == B0110)`, or is `if (myBool1 || myBool2 || myBool3)` faster and smaller than `if (myBitMask & B0111 != 0)`?

While I would like to get the "`axis_homed` vs `axis_known_position`" code committed eventually, it isn't absolutely vital for 1.1.0. So we can continue to experiment with that a bit.

closed 2016-03-06 by thinkyhead

ProtomakerSprint, 2017-11-23:
THANKING YOU AnHardt 2000%

https://github.com/AnHardt

My GitHub https://github.com/ProtomakerSprint

Cooling Head Bed before carrying out next instruction.

M190 R32; Seems to go at 54C and not 32C

Ideal to allow Heat Bed to cool to a required Temperature before removing a model part from the Heat Bed

I found that just using the End Code M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature), as requested.

This issue seems to happened because my Heat Bed can take some time to cool down, about one degree a minute, half an hour for about 30 C to cool down

After some investigation, thanking AnHardt 2000% https://github.com/AnHardt I discovered that changing Marlin_main.cpp code seem to work for me.

Now the Heat Bed Temperature waits until it is 32C as requested before continuing with the next G-Code command.

NOTE: When the M190 R32; command is reached in the End G-Code the 3D Pint Time on the LCD 12864 is lost because the Display will say:-

Bed Heating.  (Even though it is cooling) and the Time on LCD 12864 the will restart.

Once the Heat Bed is cooled down to the required temperature it will say:-
 
Bed Done.

Then the End G-Gode will continue...

End G-Code
M400 is most likely not required.

; Wait for Heat Bed to cool down to 32 C
; Part removal Temperature
M400; Wait for other commands to finish
M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature
; Below fails to set cool down at all
; M190 S32 ;set heatbed temperature to 32
; for cooldown M116 H1 ;wait for heatbed to reach 32
; M116 H1 ;wait for heatbed to reach 32
M140 S0 ; make sure the bed is turned off.

Marlin (currently 1.1.6 – October 2017)
 
In Marlin_main.cpp code, about line 7666 Aprox.

#if HAS_TEMP_BED

  #ifndef MIN_COOLING_SLOPE_DEG_BED
    #define MIN_COOLING_SLOPE_DEG_BED 1.50
  #endif
  #ifndef MIN_COOLING_SLOPE_TIME_BED

/*
Protomaker Black Sprint Original  

Increase twenty times
I am asuming this is seconds.
; Wait for Heat Bed to cool down to 32 C
; Part removal Temperature
M400; Wait for other commands to finish
M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature)
; Below fails to set cool down at all
; M190 S32 ;set heatbed temperature to 32
; for cooldown M116 H1 ;wait for heatbed to reach 32
; M116 H1 ;wait for heatbed to reach 32
M140 S0 ; make sure the bed is turned off.

WAS     #define MIN_COOLING_SLOPE_TIME_BED 60
*/
   #define MIN_COOLING_SLOPE_TIME_BED 1200
  #endif

The above works but adjusting this may also work , not tried by me yet:
// Extruder temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.