MarlinFirmware/Marlin #2771: pruisa i3 - changes doesnt update

NesThaesis, 2015-11-13:
Hey.

I have a pruisa i3 with marlin 1.1.0rc2 and ramps1.4.

It seems like g29 doesnt update the matrix, nor the zprobe offset, since the first layer of the print is not equally thick. I tried changeing the z probe offset many times. from -1.5 to -1. the extruder is too close to the bed at the left side of the print, and only the right side seems to be ok.

Tried disabling the eeprom, or making it go default with m502. none seems to work.

any ideas?

brainscan, 2015-11-13:
What are you using as a build plate? Glass, aluminium? Are your X and y rails straight and parallel? Most people that have a problem with one side over another have an issue with their mechanics which G29 cannot compensate for.

KiteLab, 2015-11-13:
@NesThaesis 
Pleas send us the complete output of a:
```
G29 V4
```

NesThaesis, 2015-11-13:
I switched to mesh leveling and the problem is the same.  Manually leveled the aluminium plate the problem is the same. Im starting to suspect that my z axis servo has voltage problems and it cant make the mikrosteps untik it reaches a certain poibt or move enough. Because when i started to print a model. The base of the model (a filled circle ) started badly. Then the right side was ok. Then the head gone around the base circle to the left and it was ok. On the same layer. And from that point it was ok. If it wouldbe a leveling problem then the left side should all be bad.

NesThaesis, 2015-11-13:
@kitelab

With auzo bed lvl or mesh leveling?

NesThaesis, 2015-11-14:
@KiteLab 

Num X,Y: 3,3
Z search height: 4
Measured points:
  3.10000  3.67500  4.22500
  2.65000  3.35000  3.90000
  2.12500  2.80000  3.40000

NesThaesis, 2015-11-14:
G29 Auto Bed Leveling
Bed X: 170.000 Y: 10.000 Z: 1.999
Bed X: 70.000 Y: 10.000 Z: 1.090
Bed X: 70.000 Y: 225.000 Z: 0.075
Bed X: 170.000 Y: 225.000 Z: 1.158
Eqn coefficients: a: 0.00995731 b: -0.00431594 d: 0.39273262
Mean of sampled points: 1.08048725
planeNormal x: -0.009957 y: 0.004316 z: 1.000000

Bed Height Topography:
+-----------+
|...Back....|
|Left..Right|
|...Front...|
+-----------+
 -1.00515 +0.07722
 +0.00942 +0.91851

Corrected Bed Height vs. Bed Topology:
 +0.00000 +0.08669
 +0.08674 +0.00015

Bed Level Correction Matrix:
+0.999950 +0.000000 +0.009957
+0.000043 +0.999991 -0.004315
-0.009957 +0.004316 +0.999941

NesThaesis, 2015-11-14:
i have the impression that the matrix is flipped,
its like he corrects the front position on the back and the back position on the front.

KiteLab, 2015-11-15:
@NesThaesis 
Thanks for your afford with the G29 output.
With so few tested points it's hard to say whether your bed is warped or not. The noise of manual probing, or a unreliable z-probe is overwhelming. From the data we have i'd say:
Grid levelling:
The difference of the min and the max is about 2 mm - that's a lot.
After correction the diagonal left/back to right/front is cloth to level.
From the centre of the bed to right/back as well to left/front the bed is rising about 1/10mm. That's warp. 1/10 mm is enough to cause problems.
Manual levelling:
The output of the mesh-levelling has a mirrored y-axis compared to the grid levelling. (small y at the top, big at the bottom) . 
Again we see about 2mm difference between back/left and front/right. 
Calculating the differences between the neighbour points (in ()) we can see if the slope for right and left of a midpoint is the same. If it is not, the slope is not constant - the bed warped - or the measurement was not as exact as we hoped. 
Again the most eye catching difference to me is at the diagonal front/left to back/right. 1.2mm difference of the differences is hopefully more then the measurements error.

```
  2,125 -  (2,725) - 2,8   - (-0,6)  -  3,4
    |   \             |              /   |
(-0,525)  (-1,225) (-0,55)   (-0,05)  (-0,5)
    |              \  |    /             |
  2,65  - (-0,7)   - 3,35  - (-0,55) -  3,9
    |              /  |    \             |
(-0,45)   (-1,25)  (-0,325)  (-0,875) (-0,325)
    |   /             |              \   |
  3,1   - (-1,575) - 3,675 - (-0,55) -  4,225
```

What to do:
Try a M48 (Z probe repeatability). If the repeatability is bad - try to improve it. If you can not -
Try to test much more points with grid levelling. More points even out measurement errors a bit.
If that does not help, your bed appears warped. If it appears shaped like a saddle, likely one of your axes is a bit twisted. Anyway grid levelling can not handle this.
Try to use mesh bed levelling with more points. MBL can handle warped beds, but 9 points are usually not enough. With unreliable probing even that will not work.
2mm difference between the highest and the lowest point is a lot. Within 5 minutes you can probably be better than 0.5mm. Physical bed levelling to a certain degree is worth the time!

KiteLab, 2015-11-15:
MBL can be tested by setting on point much (2-3mm) to high. Driving manually to that point should show a obvious gap between bed an nozzle then.

NesThaesis, 2015-11-17:
@KiteLab 

Hey. I made a 7x7 mesh bed leveling.

Printed  5, 50x50 mm 0.1 layer height squares on my bed, to test every part of it.
The first print was flawless.

Then i tried to print a modell. and the printing started about 2-3 mm above the bed.

Someone told me it may be the stepping motors are skipping steps because it accelerates too fast, and it results in incorrect height.

any idea ?

NesThaesis, 2015-11-17:
Well. I releveled my bed manually with the on screen menu.
After that, i went to. control. store memory . then it seems to work fine now.

Roxy-3DPrintBoard, 2015-11-17:
> Then i tried to print a modell. and the printing started about 2-3 mm above the bed.

I have seen an error where you get perfect bed leveling and everything prints great.   And then the next print is off horribly.   I don't know what is causing this problem, but if you reset the printer, the first print is much more likely to work correctly.

paulusjacobus, 2015-11-18:
I have encountered this behaviour as well but blamed my printer's hardware for it (i.e. heated bed is warping slowly over time).

@Roxy-3DPrintBoard thanks for the tip. It indicated that the corrections are added rather than overwritten. So more debugging to be done. I thought we got G29 finally out of the woods but apparently not....

shampine1, 2015-11-23:
I have just been AMAZED at what mesh bed leveling does for me!  I machined my Aluminum bed flat... which led to realizing just how much my Y axis sagged!  I am using mesh to teach the printer to make a planar motion despite the sag.  With this correction, I can print large scale parts (15" cube machine).
That said, I am unable to get the machine to either boot up with mesh leveling running or find a way to kick it into that mode using gcode. 
Here's my mesh:
->G29S0

<-Num X,Y: 3,3
<-Z search height: 10
<-Measured points:
<-  8.00000  8.47500  8.85000
<-  7.70000  7.80000  7.80000  
<-  8.12500  7.67500  7.70000
<-ok
I have tried M500 to save the points and M501 to load them, but that doesn't kick it into mesh leveling mode (I haven't checked to see if the parameters were loaded or not).
I tried G29 S3 X1 Y1 Z8.00000   G29 X2 Y1 Z8.47500  etc. in my start.gcode, but that doesn't kick in mesh leveling either.
I have downloaded the latest RC and I will re-try things tonight.
If anyone knows what I'm missing, I would love to hear!
BTW, it would be really nice to have a few more menu items for leveling: save parameters, load parameters, and kick in mesh leveling mode (unless load parameters does it automatically).

Roxy-3DPrintBoard, 2015-11-23:
You do know you have to #define MESH_BED_LEVELING  when you build the firmware, right?

shampine1, 2015-11-23:
Yes, I have it set up.  Here's the pertinent part of the configuration.h.  It works just fine when I go through the leveling procedure after EACH power-up cycle.  The problem is that it doesn't retain those measurements after shut down.

#define MANUAL_BED_LEVELING  // cobbble Add display menu option for bed leveling.
#define MESH_BED_LEVELING    // cobble Enable mesh bed leveling.

#if ENABLED(MANUAL_BED_LEVELING)
  #define MBL_Z_STEP 0.025  // Step size while manually probing Z axis.
#endif  // MANUAL_BED_LEVELING

#if ENABLED(MESH_BED_LEVELING)
  #define MESH_MIN_X 40  //cobble
  #define MESH_MAX_X (X_MAX_POS - MESH_MIN_X)
  #define MESH_MIN_Y 40  //cobble
  #define MESH_MAX_Y (Y_MAX_POS - MESH_MIN_Y)
  #define MESH_NUM_X_POINTS 3  // Don't use more than 7 points per axis, implementation limited.
  #define MESH_NUM_Y_POINTS 3
  #define MESH_HOME_SEARCH_Z 10  // cobble Z after Home, bed somewhere below but above 0.0.
#endif  // MESH_BED_LEVELING

Roxy-3DPrintBoard, 2015-11-23:
Hmmmm....    This block from Configuration_store.cpp does the read of the mesh leveling data.  
~~~~

    uint8_t dummy_uint8 = 0, mesh_num_x = 0, mesh_num_y = 0;
    EEPROM_READ_VAR(i, dummy_uint8);
    EEPROM_READ_VAR(i, mesh_num_x);
    EEPROM_READ_VAR(i, mesh_num_y);
    #if ENABLED(MESH_BED_LEVELING)
      mbl.active = dummy_uint8;
      if (mesh_num_x == MESH_NUM_X_POINTS && mesh_num_y == MESH_NUM_Y_POINTS) {
        EEPROM_READ_VAR(i, mbl.z_values);
      } else {
        mbl.reset();
        for (int q = 0; q < mesh_num_x * mesh_num_y; q++) EEPROM_READ_VAR(i, dummy);
      }
    #else
~~~~
If you notice, mbl.active is the first piece of information from the set that gets read.  Take a look at the store side of things:
~~~~
 uint8_t mesh_num_x = 3;
  uint8_t mesh_num_y = 3;
  #if ENABLED(MESH_BED_LEVELING)
    // Compile time test that sizeof(mbl.z_values) is as expected
    typedef char c_assert[(sizeof(mbl.z_values) == MESH_NUM_X_POINTS*MESH_NUM_Y_POINTS*sizeof(dummy)) ? 1 : -1];
    mesh_num_x = MESH_NUM_X_POINTS;
    mesh_num_y = MESH_NUM_Y_POINTS;
    EEPROM_WRITE_VAR(i, mbl.active);
    EEPROM_WRITE_VAR(i, mesh_num_x);
    EEPROM_WRITE_VAR(i, mesh_num_y);
    EEPROM_WRITE_VAR(i, mbl.z_values);
  #else
~~~~

Did you see your printer do a G28 at the end of setting all the mesh leveling points?  Because that is the point in time where mbl.active gets set to say everything is good to go.  If you have not seen the last point cause a G28, you have not activated mesh bed leveling and it doesn't matter if you do a M500, it will save the state of the machine with mesh leveling not active.
~~~~
        // Is there another point to sample? Move there.
        if (probe_point < MESH_NUM_X_POINTS * MESH_NUM_Y_POINTS) {
...
        else {
          // After recording the last point, activate the mbl and home
          SERIAL_PROTOCOLLNPGM("Mesh probing done.");
          probe_point = -1;
          mbl.active = 1;
          enqueuecommands_P(PSTR("G28"));
        }
        break;
~~~~

shampine1, 2015-11-24:
Argh!  I feel stupid!  #define EEPROM_SETTINGS  is commented on the standard build...  I'll see if that helps (along with the latest RC).

closed 2016-02-22 by thinkyhead