MarlinFirmware/Marlin #2772: chk sum & line # errors

ruggb, 2015-11-13:
I am using 1.1.0.RC2 M105 patch, with a Arduino Mega 2560 bd, Ramps 1.4 and compiled with Arduino 1.6.6 on a CoreXY platform. - BTW, that patch seems to be working logically now, at least for E0. I need to chk the other hotend to be complete.
I am seeing a lot of chk sum & line # errors and wonder what might be causing it.
I have not seen these in the past BUT I am running a lot faster than I did in the past.
I am printing now and though Slic3r was set for 100mm/sec perimeter speed it is running much slower on this print - maybe because the perimeter is smaller than the last one. When it gets into the infill it seems to be worse as that appears to be running faster. I think on this print that is the only time I get errors. Obviously, it must be due to the print speed. The models print out OK.
The other model had longer perimeters and looked like it might be getting to 100 then and I think it generated errors then also.
I don't think it is a comm issue in the h/w interface otherwise it would do it regardless of the print speed. I would guess that it is the f/w not responding fast enough.
Is there something that can be tweaked to allow it to respond at higher speeds?
Or do u have some tips re my config - which hasn't changed for a long time.
Otherwise, maybe u need to specify a speed limit....

thinkyhead, 2016-02-22:
We have some communication patches pending that you might want to try, such as #3012 which prevents errors when mixing serial commands with commands from the LCD menu or SD printing. There's also #2994 which takes another approach to the same problem. @AnHardt also has a patch to fix the way that the `N` line number is handled in the `ADVANCED_OK` response. Perhaps we can put together a branch with all these latest patches for you to test out.

ruggb, 2016-02-22:
thx Steve - that isn't exactly the issue I am having, though it may affect it.
It is just if I set the print speed around 50 or above I start getting errors. Errors increase with speed.
I am assuming it is b/c the printer can't accept the data and respond fast enough or loses a response.
I don't know if corexy makes a difference in the response time.
If u get something together I will try it.

ruggb, 2016-03-01:
I just tested  bugfix 9a3408e where I was getting comm errors and it appears they r no longer there.
It was a quick test but the issue was rather consistent and it didn't happen this time so it seems fixed. I will continue to try to break it but right now it looks good for this issue. A lot seems to have gotten thrown together in that fix so I have no idea what may have stopped the comm errors, but thanks anyway.

thinkyhead, 2016-03-02:
Great! If you see this again, please re-open.

closed 2016-03-02 by thinkyhead