MarlinFirmware/Marlin #2774: Problem with heating security

Eyolon, 2015-11-14:
Hello.

For this moment i've disabled the heating security because my J Head extruder is slow between 230 and 240° (because i use a fan on the radiator) and before the end of the heating process, the firmware stop the printing. 
For the next release, it could be interesting to create a manual time out setting to calibrate this security system

Wackerbarth, 2015-11-14:
Please go to the MarlinFirmware/MarlinDev repository and create a FEATURE_REQUEST issue there.
It would be helpful if you could describe how you think the feature should work.

Roxy-3DPrintBoard, 2015-11-14:
@Wackerbarth    I had trouble with the predefined values in RC2 also.  And I remember somebody else getting timeouts because their heaters could not change temperature as aggressively as the parameters insisted.   There was a very polarized discussion.  The people that wanted the tight and aggressive parameters were arguing from a safety perspective.

The RC2 code currently is:
~~~~
  #define WATCH_TEMP_PERIOD 16                // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
~~~~

I had to back my RC2 settings off to:
~~~~
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
~~~~

What I'm wondering (and want to hear opinions about) is this:  Does it make sense to leave the parameters very tight, but add a comment saying that it is very possible the user has to relax them and discourage people from turning the feature off.   Right now, by having the values so aggressive people are running into trouble and turning the feature off.  This person did just that.  And I can tell you I did the exact same thing.   I'm sure this wording can be improved, but this is for the purpose of discussion:
~~~~
  /**
   * NOTICE:  Thermal protection of the hotends is important.  These values are deliberately set very
   *                aggressively.   A lot of Printer Control Boards will have trouble not timing out with these
   *                values.  It is better to relax the parameters than to turn the THERMAL_PROTECTION_HOTENDS off!!!
   *
   * Whenever an M104 or M109 increases the target temperature the firmware will wait for the
   * WATCH_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M104/M109,
   * but only if the current temperature is far enough below the target for a reliable test.
   */
  #define WATCH_TEMP_PERIOD 16                // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
~~~~

Wackerbarth, 2015-11-14:
As I see it, there are, effectively, two issues. One is the concern that a heater might lock "full on" and thereby create a hazard. The other is that a thermistor might have become disconnected, or otherwise inoperable, and fail to report an appropriate temperature.

Again, I argue that this is not a "bug" in the implementation, but rather a poorly designed feature.

I am not against providing additional documentation about the way that the system functions and suggestions about how the user might best configure his system.

However, I do have severe issues with the way we are presently expected to provide that documentation in EVERY copy of the Configuration files. Attempting to do so is an unsustainable methodology.

IMHO, Thermal Runaway deserves its own page in the setup documentation giving as much insight as possible about the choice of parameters.

And, I do believe that we can design a better test than the one presently implemented.
Such design is the topic of the issue that I requested the OP to post.

Roxy-3DPrintBoard, 2015-11-14:
OK!  I took away the 'Verified Bug!' label!   :)    But I do think we should have some extra information in the Configuration_adv.h file until we come up with a better solution.

ruggb, 2015-11-14:
it isn't a controller issue, it is a bed issue and the fact that 12v just can't heat the 8x8 bed fast enough.
Mine is turned off also till I figure out what I can do with it. HOWEVER, since I can hardly get it to 110°C I don't think there is much change of it running away. If it were full on for 24hrs, I would be surprised if it got to 150°C.
I am trying to figure out how to put another 12v supply in series but as yet have not.
IMHO, It isn't an issue with the design of the feature either, the target variation is just too big - not like hotend variations....
The only thing left is for ppl to understand the issue and set the parameters accordingly.

Roxy-3DPrintBoard, 2015-11-14:
> it isn't a controller issue, it is a bed issue and the fact that 12v just can't heat the 8x8 bed fast enough.

Yes, I understand in your case this is true.   I was speaking to the bigger picture.  We have an issue with the hotends also.   (And in my case...  I think my controller is whimpy.  Part of the problem in my case is related to the controller electronics.)

ruggb, 2015-11-14:
I had an issue with my hotends - turned out it was all my fault.
I had fan wires too close to the hotend thermistor wires and there was a lot of electrical noise on the thermistor. Drove the controller crazy. I separated the wires and it is perfect now.

KiteLab, 2015-11-15:
@Eyolon 
Thanks for your feature request. But before we try to invent something new i'd like to encourage you to use the already existing features.
If you are on the `Release` branch try to tune this (https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Configuration.h#L251) parameters. The description for the parameters is right above.
If you are on the `RC` or `RCBugFix` branch things have changed a bit. In `Configuration.h` you now can turn on and off the protection features (https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L299). A bit of explaining text is still above. You can tune the values in `Configuration_adv.h` (at the top of the file). If you get a "Thermal Runaway" error please tune the first block (~ line 13). If you get a "Heating failed" error please tune the second block (~ line 30).

@Wackerbarth 
The feature is already there - maybe it can be documented better.
@Roxy-3DPrintBoard 
Without having more information it's not the best idea to mention only one of the possibly involved mechanisms.
@ruggb 
The OP is talking about "extruder is slow between 230 and 240° ". Where did you got the idea this could have to do with the bed? If you are unsure, you could point on the fact, that the error-message (at least on the console) includes the heater number.

Roxy-3DPrintBoard, 2015-11-15:
@KiteLab 
> Without having more information it's not the best idea to mention only one of the possibly involved mechanisms.

I don't understand your comment.   Can you explain the issue?   Are you talking about the proposed comment text?   Probably some text would be appropriate for the bed too but I was just trying to get a discussion started.   If that is the issue, would we want a more complete notice or a separate notice in front of the bed parameters?

ruggb, 2015-11-15:
@KiteLab 
Sorry, I Ass-u-me-d "heater" referred to bed. Hotend didn't register. But the issue would still apply to both - except with the hotend I would think something else is going on.

closed 2015-11-24 by AnHardt

ghuser2, 2015-11-27:
I too turned it off until I found out how to change the settings:

#define THERMAL_RUNAWAY_PROTECTION_PERIOD 40
#define THERMAL_RUNAWAY_PROTECTION_HYSTERESIS 30
#define THERMAL_RUNAWAY_PROTECTION_BED_PERIOD 20
#define THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS 10