MarlinFirmware/Marlin #2775: Using something else instead of encoder?

CoderSquirrel, 2015-11-14:
I'm studying marlin a little bit and i'm wondering if anyone uses it without encoder and using arrows or joysticks.
I'm interest on making these modifications but i saw that there's a looooooooooooooot to modify since it's build up for encoder.
So i was thinking if anyone else has made this change or if it's not a useful thing.

Roxy-3DPrintBoard, 2015-11-14:
I guess it depends on what you mean by 'encoder'.  Marlin does use endstops to know where the limits are.  But other positioning is done by stepping the motors the correct amount.  There is not an encoder to read the position of the various axis.

ntoff, 2015-11-14:
I assume he/she means the rotary encoder you navigate the menu with.

CoderSquirrel, 2015-11-15:
Hi, yeah i meant rotary encoder for menu navigation
SOrry  xD

KiteLab, 2015-11-15:
There already is code in Marlin, able to handle some buttons. (look for REPRAPWORLD_KEYPAD)

The changes would not be too much, if you stay with the current menu system. All you need is a "Up", a "Down" and a "Select" button. "Up" and "Down" could directly work on the `encoderPosition` variable.

jasane, 2015-11-16:
Thanks KiteLab i didn't know that.
Actually i made a way to my joystick work and it's good, but i'm having another trouble now.
AS I'M using 16x2 lcd my menu isn't scrolling down, i tried the default setting with rotary encoder and my joystick setup, none of then scrolled.
Have you guys worked with 16x2 lcd?

CoderSquirrel, 2015-11-16:
Thanks KiteLab i didn't know that.
Actually i made a way to my joystick work and it's good, but i'm having another trouble now.
AS I'M using 16x2 lcd my menu isn't scrolling down, i tried the default setting with rotary encoder and my joystick setup, none of then scrolled.
Have you guys worked with 16x2 lcd?

KiteLab, 2015-11-16:
You can find the actual scroll code in the `START_MENU()` and  `END_MENU()` macros in ultralcd.cpp. As far i can see they use `LCD_HEIGHT` but i have not tested if that works for 2.

CoderSquirrel, 2015-11-21:
HI all, just for information, i made the code for working with joystick =D

closed 2015-11-21 by CoderSquirrel