MarlinFirmware/Marlin #2776: M42 issues

hackermagnet, 2015-11-14:


Roxy-3DPrintBoard, 2015-11-14:
Is Pin-4 where the LED is?   Does it actually turn on the LED when you do a M42 P4 S255 ?

Because, I don't see anything in the code that would keep the M42 from completing and going on to the next command.

You can try replacing your M42 command with the follow code to get more information about what is happening:
~~~~
inline void gcode_M42() {
SERIAL_PROTOCOL("Starting M42\n");
  if (code_seen('S')) {
    int pin_status = code_value_short(),
        pin_number = LED_PIN;

SERIAL_PROTOCOL("Starting M42\n");

    if (code_seen('P') && pin_status >= 0 && pin_status <= 255)
      pin_number = code_value_short();

SERIAL_PROTOCOL("pin_number=");
SERIAL_PROTOCOL(pin_number);
SERIAL_PROTOCOL("\n");

    for (uint8_t i = 0; i < COUNT(sensitive_pins); i++) {
      if (sensitive_pins[i] == pin_number) {
        pin_number = -1;
SERIAL_PROTOCOL("Yikes!  You selected a sensitive pin.\n");
        break;
      }
    }
SERIAL_PROTOCOL("Done checking for a sensitive pin.\n");

    #if HAS_FAN
      if (pin_number == FAN_PIN) fanSpeed = pin_status;
    #endif

    if (pin_number > -1) {
      pinMode(pin_number, OUTPUT);
      digitalWrite(pin_number, pin_status);
      analogWrite(pin_number, pin_status);
SERIAL_PROTOCOL("Done changing pin value\n");
    }
  } // code_seen('S')
SERIAL_PROTOCOL("Leaving M42\n");
}
~~~~

hackermagnet, 2015-11-14:
I tried some other words on M117 command and it worked.
I was using the words "ON" and "Nozzle" on my M117 commands. Is there a list with forbidden words?
The G1 command wasn't executing because I was using absolute coordinates and the nozzle was already at that spot.

Roxy-3DPrintBoard, 2015-11-14:
Well, I'm pretty sure the amount of text allowed is very small.   Maybe less than 20 characters.  I don't know.   But having a space in the text is for sure going to cause problems.   I think you will find a lot of people that don't even think the M117 command should exist.  It breaks all the GCode rules.

hackermagnet, 2015-11-14:
It worked when I changed "ON" to "On" without removing spaces.
I will probably stop using M117.

closed 2015-11-14 by hackermagnet

Roxy-3DPrintBoard, 2015-11-14:
@hackermagnet 

Oh!   It isn't a forbidden word!!!  Capital N's are used by the lower level serial protocol.   You can't ever send an N.   I ran into that adding functionality to the G29 command.   I wanted an N parameter to say how many points to probe and everything goes sideways if you do that.

hackermagnet, 2015-11-14:
Now it makes sense. Thanks for pointing that out.

KiteLab, 2015-11-16:
I can confirm the capital 'N' problem in M117 for `Release` and `1.0.x` but not for `RC`, `RCBugFix` and `master`.
@hackermagnet 
Are you sure you compiled the right branch?

hackermagnet, 2015-11-16:
@KiteLab Yes, I use Marlin 1.1.0-RC2 - 29 September 2015

KiteLab, 2015-11-16:
Fixed with https://github.com/MarlinFirmware/Marlin/commit/c0e791dbe9a25b830e46511117d2c5ed6a5c4be4 13. Oct 2015