MarlinFirmware/Marlin #2777: Autolevel position/offset clarification

therippa, 2015-11-15:
To ease the transition to the new firmware, you might want to move the PROBE_OFFSET section above the PROBE_BED_POSITION section, and use them as variables along with the MAX_POS/MIN_POS settings (like LEFT_PROBE_BED_POSITION = X_MIN_POS + X_PROBE_OFFSET + MIN_PROBE_EDGE) to make the offset settings more clear and extensible.  I'm very familiar with Marlin 1.8 and its configuration, and tonight when upgrading to 1.9-RC2, the functional changes weren't super clear to me, and this was the only hiccup I ran into.

KiteLab, 2015-11-15:
Pardon?
Where did you get these versions?

therippa, 2015-11-15:
Sorry, got the versioning mixed up with an unrelated project :)  I meant 1.1-RC2.

thinkyhead, 2016-02-22:
Even though the `*_PROBE_OFFSET_FROM_EXTRUDER` settings come after the `*_PROBE_BED_POSITION` settings, it's a feature of the C++ compiler preprocessor that you can actually refer to a later setting in an earlier setting. So this is okay to use:
```cpp
#define RIGHT_PROBE_BED_POSITION (X_MAX_POS - X_PROBE_OFFSET_FROM_EXTRUDER)
```

closed 2016-05-04 by jbrazio