MarlinFirmware/Marlin #2779: Can't go below z probe trigger height

MarkJB, 2015-11-15:
I have been experimenting with an IR height probe (https://miscsolutions.wordpress.com/mini-height-sensor-board/ - with a fixed position mount http://www.thingiverse.com/thing:950478) and auto bed leveling on an i3 with RC2 and am stuck trying to get the head near enough to the bed to print anything. 

I've determined the Z height offset as per the first link above and entered that in the firmware and uploaded (Z_PROBE_OFFSET_FROM_EXTRUDER -0.79).

When I run G28 the X & Y home on their respective endstops as before, and Z now moves to the middle of the bed and homes Z using the Z probe. So thats good.

When I run G29 (3 x 3 grid) all 9 points are probed and the LCD tells me that I am at z=0.79. Trying G1 Z0 will show the Z probe endstop is hit and Z won't go any lower.

I've set Z_MIN_POS = -2.

I've tried this with the probe on Z_MIN_PIN and on its own Z_MIN_PROBE_PIN  (I set Z_MAX_PIN = -1 and used that pin for the Z_MIN_PROBE_PIN). I've tried it with and without Z_MIN_PROBE_REPEATABILITY_TEST but I can't get it to go below the Z probe trigger point. Any suggestions as to what I am missing?

Roxy-3DPrintBoard, 2015-11-15:
Do you have 

~~~~
#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.
~~~~

MarkJB, 2015-11-15:
Yes, they are defined. I set the Z_MIN_POS = -2 but that didn't help.

#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS -2
#define X_MAX_POS 225
#define Y_MAX_POS 270
#define Z_MAX_POS 200

Wackerbarth, 2015-11-15:
> 
> Yes, they are defined. I set the Z_MIN_POS = -2 but that didn't help.
The issue for you is that by defining a min endstop, you are telling the printer that you are at the limit of travel when the endstop triggers.  But, because your trigger point is higher than the bed, you cannot travel any lower to actually reach the bed,

What you want is a probe, but no min_endstop.

KiteLab, 2015-11-15:
Please check in `Configuration_adv.h` if `ENDSTOPS_ONLY_FOR_HOMING` is at it's original state.
```
#define ENDSTOPS_ONLY_FOR_HOMING
```

MarkJB, 2015-11-15:
Yup, it was //#define ENDSTOPS_ONLY_FOR_HOMING. 

Enabled and I can go below the probe trigger height now.

I can't remember when I disabled that. Must have been some time yesterday...

Thanks.

closed 2015-11-15 by MarkJB