MarlinFirmware/Marlin #2783: DualExtrude feature

nikolindima, 2015-11-16:
Can Marlin print with 2 extruders? I mean when you print 2 parts in same time with 2 extruders simultaneously 
[image]

dintid, 2015-11-17:
Maybe that questions is more appropriate on http://forums.reprap.org/ where you can find marlin and/or your specific printer :)

Roxy-3DPrintBoard, 2015-11-17:
I do not believe there is anything in the current code base to do this.   But it would not be a very difficult change to make.   You would just turn on both hot ends and tell both extruder stepper motors to move at the same time.

AnHardt, 2015-11-24:
A dual extruder setup seems to be possible, but then with alternating printing the right and left object.
The second possibility could be DUAL_X_CARRIAGE in Mode 2 (https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Configuration_adv.h#L191).

caenan0, 2015-11-27:
I've got a dual x carriage setup and it's definitely possible to print simultaneously with Marlin.

thinkyhead, 2016-02-21:
This is something to make sure to add to the documentation on dual X carriages. For now it is found in the configurations.

closed 2016-02-21 by thinkyhead