MarlinFirmware/Marlin #2784: MakerBot - Mighty Board Rev G - Port Marlin

metropt, 2015-11-16:
Hello,

I have a mighty board rev g lying around from makerbot 2. I would like to run Marlin code to test it. I have the information about X/Y/Z STEP, DIR and ENA pin and also ENDSTOP pins.

What are the minimums needed for a one extruder printer and what other pins should I look for?

It is what I have for now.

define X_STEP_PIN          49
#define X_DIR_PIN           50
#define X_ENABLE_PIN       47
#define X_STOP_PIN         60

#define Y_STEP_PIN          40
#define Y_DIR_PIN           42
#define Y_ENABLE_PIN       39
#define Y_STOP_PIN         59

#define Z_STEP_PIN          36
#define Z_DIR_PIN           37
#define Z_ENABLE_PIN       35
#define Z_STOP_PIN         58

I didn't found any board that have the pins similar to this on pins.h, will I be able to use it anyway? Can you confirm me if I'll be able to use?

Thank you!

Roxy-3DPrintBoard, 2015-11-17:
There is a very stable version of an older Marlin for MakerBot at:

https://github.com/beckdac/Marlin

If you start with the beckdac fork and start updating the various pin definitions you should be able to make progress pretty quickly.

metropt, 2015-11-22:
It looks like it is for a very old markerbot.

I'll compile it and flash it into the atmega and try to move a stepper. I'm just afraid of mis configure the IO and burn something...

thinkyhead, 2016-02-22:
Any luck? There have been more than a few requests for MightyBoard support.

closed 2016-04-23 by jbrazio