MarlinFirmware/Marlin #2785: Link temperature sensor input to servo control output

alex-gibson, 2015-11-17:
Could you point me to the right place in Marlin to make some mods please?
 
I have a servo attached to Servo pin 0, and I want to make it move according to temperature monitored by Marlin from an NTC100K thermistor.
The application is for food extrusion.  The servo is to push the cap of a spray can of freezer spray, whenever the temperature exceeds say 10 degrees C.  
 
I have enabled Marlin to respond to M280 P0 S***, and can control the position of the servo using this Mcode.

How can I link the state of this output to a temperature threshold monitored by a NTC100K thermistor input?
 
On/off would be fine (push servo to top value when thermistor reads <Xdegrees, bottom when >Xdegrees).

Variable output, moving servo between upper and lower thresholds according to measured temperature would be even better.  

Many thanks!!  Alex

KiteLab, 2015-11-17:
For an example how to use a servo search for M280 in MarlinMain.cpp. Basically it is 
```
servo[servo_index].move(servo_position);
``` 

You can integrate your code to control the heater/cooler in `manage_heater()` in temperature.cpp. There you have a lot of examples how to read temperatures and how to react react on their changes.

alex-gibson, 2015-11-17:
Thanks for responding, trying this now!

alex-gibson, 2015-11-17:
Hi there, is there any chance you could give me more clues please?  I found the right areas but got a bit lost in all the nested variables.

I am not too stupid today, as I have managed to write a brand new Arduino sketch which uses the Servo and temperature sensing libraries, and works standalone on a separate Arduino Uno, keeping the temperature at a steady 10 degrees Celsius.

However I would really like to do this through Marlin/RAMPS to be able to monitor and set temperatures from Pronterface...!

KiteLab, 2015-11-17:
```
#include "servo.h"  //??
...
for (int e = 0; e < EXTRUDERS; e++) {
  if (current_temperature[e] >  target_temperature[e]) {
    servo[e]..write(servo_ON_position);
  } else {
    servo[e]..write(servo_OFF_position);
  }
...
}
```

Roxy-3DPrintBoard, 2015-11-17:
Is this more correct?
~~~~
  if (current_temperature[e] <  target_temperature[e]) {
~~~~

KiteLab, 2015-11-17:
No.
>The application is for food extrusion. The servo is to push the cap of a spray can of freezer spray, whenever the temperature exceeds say 10 degrees C.

Roxy-3DPrintBoard, 2015-11-17:
Right!  I typed the "Oh!" after I realized my thinking was backwards...

alex-gibson, 2015-11-18:
Thanks!  Implementing this now - hope to get further!!

closed 2015-11-24 by AnHardt