MarlinFirmware/Marlin #2787: M300 Beeps much higher pitch than they should be.

KriLL3, 2015-11-18:
Upgraded from 1.0.2 to 1.1.0-RC2 and using the same g-code as before all of my M300 entries give pretty much the same really high pitched chirp instead of different tones denoted by the S variable, it's not a big issue but I liked having my printer make 2-4 note sounds based on it's state, (homing axis, heating bed, heating extruder, starting to print, done printing) and using a snippet of a song like a ringtone is pretty popular to put into the end g-code, but all those sounds are now indistinguishable high pitched squeaks.

I'm using RAMPS 1.4 with the discount full graphic smart controller

KriLL3, 2015-11-18:
Ok I've tried it I get no error if I put it on the line after

 #if ENABLED(SPEAKER) // a speaker needs a AC ore a pulsed DC (#12)

I get an error if I put it after 

 #else // buzzer has its own resonator - needs a DC  (#22)

And no error if I put it after

#else (#27)

It's defiantly going down the DC route, which is weird, doesn't that only allow one tone? In 1.0.2 it could do a whole range just fine.

Roxy-3DPrintBoard, 2015-11-18:
Yes, it is possible the configuration isn't correct...   But if you want to try to get back the old functionality, you can try a couple of things.  You can replace buzz() with each possible combination and see if one of them does what you want
~~~~
  void buzz(long duration, uint16_t freq) {
    if (freq > 0) 
        lcd_buzz(duration, freq);
  }
~~~~
If that doesn't work...  Try:
~~~~
  void buzz(long duration, uint16_t freq) {
    if (freq > 0) {
        SET_OUTPUT(BEEPER_PIN);
          unsigned int delay = 1000000 / freq / 2;
          int i = duration * freq / 1000;
          while (i--) {
            WRITE(BEEPER_PIN, HIGH);
            delayMicroseconds(delay);
            WRITE(BEEPER_PIN, LOW);
            delayMicroseconds(delay);
           }
    }
}
~~~~
If that doesn't do it...  We can put the original code back in place.

KiteLab, 2015-11-19:
@KriLL3 
Did you try to define SPEAKER in configuration.h?

KriLL3, 2015-11-19:
@KiteLab 
In 1.0.2 all I had to do to use the "RepRapDiscount Full Graphic Smart Controller" was uncomment that one line "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" and it worked speaker and all, I assumed 1.1.0-RC2 wouldn't have changed that? Looking around in the old configuration files it seems uncommenting that line made 1.0.2 set up a bunch of other stuff like right pins, using the graphical LCD, SD card and speaker etc.

@Roxy-3DPrintBoard 
I'll try that tomorrow, family is asleep and I doubt they'd appreciate a series of fairly loud beeps.

KiteLab, 2015-11-19:
@KriLL3 
That has changed. To make it work as it did before you now have to uncomment `SDSUPPORT` and `SPEAKER`. Some call it progress. 
For me the much more heatable sound is a progress over the possibility to vary the frequency - i have a loud environment. Technically its more correct now, as a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER has a buzzer, not as speaker. Else you could not hear anything if `SPEAKER` is not defined.
`SDSUPPORT` is an other thing. It used to be defined in `ULTIPANEL` i think. `ULTIPANEL` is used by almost all other panels/displays. For one of the new displays it was needed to #undefine `SDSUPPORT`. There was/is someone around who means that should never occur, So the setting was removed from `ULTIPANEL`. If someone would be interested enough, he could search for the panels/displays using `ULTIPANEL` and readd the `SDSUPPORT` setting in the special sections of that displays in `Conditionals.h`.

KriLL3, 2015-11-19:
Not entirely sure I'd call it progress to have to dig into the configuration to enable the features of a known LCD controller design, before the SD card and multi-frequency speaker were enabled by just specifying that you have that LCD controller... I use octopi so I don't use the SD card but still would like that feature to be accessible...

Roxy-3DPrintBoard, 2015-11-19:
> Not entirely sure I'd call it progress to have to dig into the configuration

Well, to be fair, it sounds like somebody got 'clever' and started using buzzers in an undocumented mode as a speaker.   I can understand why that should not be encouraged.

>  For one of the new displays it was needed to #undefine SDSUPPORT. ... So the setting was removed from ULTIPANEL. If someone would be interested enough, he could search for the panels/displays using ULTIPANEL 

Can you tell me which files I should look at?

Wackerbarth, 2015-11-19:
"For one of the new displays it was needed to #undefine SDSUPPORT. ... So the setting was removed from ULTIPANEL." -- Well, the problem is caused by the misuse of ULTIPANEL. When it is overloaded to represent additional displays, the conflict arises. Instead, the ULTIPANEL should be replaced with a more generic display definition. Then, in Conditionals.h, ULTIPANEL can define both that more generic display and the SD support.

Roxy-3DPrintBoard, 2015-11-19:
If something is not a true ULTIPANEL clone, shouldn't it have its own configuration.h file?    If something is very close to being a true ULTIPANEL clone, could it's configuration.h file #include the standard ULTIPANEL file and then #undef what needs to change?

Wackerbarth, 2015-11-19:
No, you don't undefine things.  Instead, you create a definition for the common property and have both include that common element.

A Chevrolet is like a Ford EXCEPT ...
I don't start by defining a Chevrolet as a Ford.
Instead, I take those common properties and use them to define a Car.

Then, a Chevrolet is a Car plus .... <some properties>
And a Ford is a Car plus ... <some other properties>

Then someone shows up with a Honda

Well, it, also, would be a Car plus ... except it is not built by an American manufacturer.
So we have to change something.

We start by removing the "American Manufacturer" property from "Car" since not all cars have this property.  But the Ford and Chevrolet still need the property.

We can add it back by adding it explicitly to each of those definitions.
OR, we can create a new class "AmericanCar" which is a "Car" and also has the "American Manufacturer" property.

Either way, we never undefine a property.

Roxy-3DPrintBoard, 2015-11-19:
Well...   It seems like UltiPanel should be its own thing.  It is important enough to have its own files.  And to have all of its features included for use.

And rather than build up a new thing that is almost an UltiPanel, it doesn't seem that bad (to me) to start with an UltiPanel and un-define the one thing that is different.  From a perspective of trying to make the code understandable, that is easier for me to get my head around than looking at two big files, line by line trying to find the one difference.

Wackerbarth, 2015-11-19:
You're thinking "flat" rather than "hierarchal". Rather than starting at the specific case (the UltiPanel), think about the more generic feature (the user control panel). The UltiPanel is just a special case of this more general class.

Just as when the biologists describe some dog, they don't start by listing every characteristic individually.
Instead, the simply say that a dog is a mammal and we automatically know that it has certain characteristics (because those characteristics are common to ALL mammals).

Similarly, we can say that each of the panels has all of the characteristics common to a panel and need specify only the distinctive components.

Now, this "inclusion" can be done either by defining a symbol for the generic class and including that symbol in the specific descriptions or, equivalently, we can use the pre-processor "#include" to pull in a file specifying those properties. In either case, we don't have "two big files" to compare. Instead, we have two small files which contain the distinctions and one larger file that contains the commonality.

Roxy-3DPrintBoard, 2015-11-19:
> "For one of the new displays it was needed to #undefine SDSUPPORT. ... So the setting was removed from ULTIPANEL." 

Yes, but some how...   The UltiPanel lost its ability to do SD Memory cards.  It would seem UltiPanel's should not suffer just because some new display shows up that is close, but not really an UltiPanel.   

It would seem to me the new display should fend for itself and come up with its own definitions

In a perfect world, UltiPanel's should probably get defined by pulling in this and that defination file for the various features.   And the new panel could pull in most of the stuff, but just omit the SD Memory card stuff.   

Right?

KiteLab, 2015-11-19:
@Roxy-3DPrintBoard 
>Can you tell me which files I should look at? 
I don't want to scare you, but about everything what has to do with displays, menus, pins, configuration, conditionals, sd-cards, ... . A grep for `ULTIPANEL` has about 150 hits in 44 files.
A somewhat challenging task, to sort that out. Worse - every change affects all kinds of displays/panels. I don't know anyone who can test more than 3 types of displays.

Roxy-3DPrintBoard, 2015-11-19:
What is the display name and type that is very close to a UltiPanel but is missing the SD Memory card feature?   It doesn't seem like it should be allowed to call itself an UltiPanel if it doesn't do everything an UltiPanel does.

KiteLab, 2015-11-20:
I remembered wrong. The actual reason for the change was different.
The change: https://github.com/MarlinFirmware/MarlinDev/pull/139
The discussion before: https://github.com/MarlinFirmware/MarlinDev/issues/133
What i remembered and prepared Thinky was dealing with `SDCARDDETECTINVERTED`: https://github.com/MarlinFirmware/Marlin/pull/2144

Anyway, all panels define ether `ULTIPANEL` or `ULTIMAKERCONTROLLER` what defines `ULTIPANEL`. It used to be a synonym for 'has hardware capable to use menus and sd'. Now it only means 'has hardware capable to use menus'  - and that is the way to go i think. Splitting off the single components from `ULTIPANEL`, part for part, property for property, until we get a `ULTIPANEL` only meaning itself. For that we'll need more abstract constructs like
```
#if (ENABLED(HAS_ENCODER) || ENABLED(HAS_MENU_KEYS)) && ENABLED(USE_MUNUINPUT)
  #define MENUINPUT
#endif
#if ENABLED(DISPLAY) && ENABLED(MENUINPUT) && ENABLED(USE_MENU)
  #define MENU
#endif

#if ENABLED(MENU)
  void menu() {
  ...
  }
#endif

#if ENABLED(DISPLAY) && ENABLED(USE_DISPLAY)
 void status_screen() {
  ...
#endif
```
It's a good thing to have `#define USE_DISPLAY` __ON__ by default but only if it is sourounded by a
```
#if ENABLED(DO_NOT_USE_DISPLAY)
  #define USE_DISPLAY
#endif
```
Or how about `DISPLAY_OFF`?
Or in this special case `SD_CARD_OFF`?
Maybe that is the way to handle the capability of `ULTIPANEL` to handle `SDSUPPORT`.

For me this is a complex we should return to a bit later, but not handle it right now.

Wackerbarth, 2015-11-20:
@KiteLab --
We seem to have two problems here.
The first is that SD card was removed from the UltiPanel, but it appears that some functionality might have been dropped.

The other is that "ULTIPANEL" appears to have been used in many places where something like "USE_MENU" would be more appropriate.

-- your comment about "USE_DISPLAY" being on by default has the right motivation. However,your code snippet is entirely wrong.

I don't see any reason why we cannot start engineering this migration in the dev branch.
We would start by doing a global replacement of ULTIPANEL with USE_MENU in all of the code (not configuration) files and have ULTIPANEL define USE_MENU by default.

From there, someone will need to look at each code instance to determine if that code applies to all of the MENU systems, or only to the specifically ULTIPANEL ones.  But, the transition should not disrupt working configurations.

ntoff, 2015-11-20:
@Roxy-3DPrintBoard 
>Well, to be fair, it sounds like somebody got 'clever' and started using buzzers in an undocumented mode as a speaker. I can understand why that should not be encouraged.

https://www.arduino.cc/en/Tutorial/PlayMelody

Playing tunes with a piezo is part of Arduino's tutorials. Hardly "undocumented", unless a "buzzer" is something other than a piezo? (Which is what I thought these panels were using to make sound). I know mine uses some sort of piezo thing which is more than capable of playing tunes on it. Are there others out there that use something else?

AnHardt, 2015-11-20:
@ntoff 
Please don't confuse things.
A __piezo__ is a sound device using a __crystal__ instead of an magnet.
A __buzzer__ is a sound device having its own __frequency generator__.
That's completely independent from each other.

Roxy-3DPrintBoard, 2015-11-20:
@KiteLab   Thank You for all the work it took to put together that synopsis of what happened with UltiPanel and SD Memory cards!!!   That was very helpful in understanding things.

closed 2015-11-24 by AnHardt