MarlinFirmware/Marlin #2792: Head crash during G29

ex-nerd, 2015-11-23:
Kossel Pro (delta).  I'm playing with some precision settings and set myself up with 13 auto-probe points.  Starting with the second or third row, the head crashes and bumps against the glass when moving between rows.  That is, it probes a row fine but the head crashes when it moves from the end of one row to the start of the next.

This seems to be because the movement of the head is in a U shape rather than a straight line across.  The previous firmware was a year or so old, so something appears to have changed in the movement routines since then.

Wurstnase, 2015-11-23:
Thats a known issue.
A solution could be https://github.com/clefranc/Marlin4Delta/commit/c3bfd3c5589b06474bc187600ad1b3c6e7b07d68

KiteLab, 2015-11-23:
An other solution is to increase Z_RAISE_BETWEEN_PROBINGS.

Wurstnase, 2015-11-23:
> An other solution is to increase Z_RAISE_BETWEEN_PROBINGS.

Right, forgot to write this also :+1:

Roxy-3DPrintBoard, 2015-11-23:
I had the same problem.  It turns out recursion is an easy way to fix this issue.   If you replace the existing code with this, the movement will not be U-shaped.

~~~~
  /**
   *  Plan a move to (X, Y, Z) and set the current_position
   *  The final current_position may not be the one that was requested
   */

#define MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	25

static void do_blocking_move_to(float x, float y, float z) {
    float oldFeedRate = feedrate;

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (marlin_debug_flags & DEBUG_LEVELING) {
        print_xyz("do_blocking_move_to", x, y, z);
      }
    #endif

    #if ENABLED(DELTA)

      feedrate = XY_TRAVEL_SPEED;

 //
 // Should we break up the move because it is too big ???
 //

      if (abs(x-current_position[X_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(y-current_position[Y_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(z-current_position[Z_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE ) {

  		do_blocking_move_to( current_position[X_AXIS] + (x-current_position[X_AXIS])/2.0,
  				     current_position[Y_AXIS] + (y-current_position[Y_AXIS])/2.0,
  				     current_position[Z_AXIS] + (z-current_position[Z_AXIS])/2.0);
//
// At this point, we are halfway to where we want to be.  Do the other half of the move.
//

		do_blocking_move_to(x, y, z);
		feedrate = oldFeedRate;
recursion_cnt--; // Roxy debug stuff
		return;
      } else {		// No...  It is small enough to do in one piece.

      	destination[X_AXIS] = x;
      	destination[Y_AXIS] = y;
      	destination[Z_AXIS] = z;
      	prepare_move_raw(); // this will also set_current_to_destination

      	st_synchronize();
      }
    #else

      feedrate = homing_feedrate[Z_AXIS];

      current_position[Z_AXIS] = z;
      line_to_current_position();
      st_synchronize();

      feedrate = xy_travel_speed;

      current_position[X_AXIS] = x;
      current_position[Y_AXIS] = y;
      line_to_current_position();
      st_synchronize();
    #endif

    feedrate = oldFeedRate;
  }

~~~~

Roxy-3DPrintBoard, 2015-11-23:
@Wurstnase 
> Thats a known issue.
> A solution could be clefranc/Marlin4Delta@c3bfd3c

Have you had a chance to check out this solution?   Would you feel comfortable recommending this approach over what I did?     (Obviously, we need a good solution...  And we have two to pick from right now.   We only need one.)

Wurstnase, 2015-11-23:
I don't have a Delta to test this. But @clefranc maybe can tell us more.

Wackerbarth, 2015-11-23:
@Roxy-3DPrintBoard -- That is a rather poor solution in that it severely defeats the look-ahead speed control because it fills the planning buffer with many small movements. This impairs the ability to reach maximum travel speed because the planner must anticipate that it must be able to stop in a short distance.

A much better solution is to allow G0 and G1 to work as originally designed in RS-274 and provide sufficient clearance to allow for the "sagging" motion when you are in G0 mode.

Roxy-3DPrintBoard, 2015-11-23:
No...   In fact, I don't think the planner gets called at all.  That is the whole problem with do_blocking_move_to().    This function is just called by the various MCodes and GCodes that need to move where the nozzle is.  That is why the U-Shaped moves show up.   And in fact, it doesn't fill the planning space at all.   It winds up the stack a little bit as it breaks a move in half again and again.  But it doesn't use any of Marlin's buffer space.

Wackerbarth, 2015-11-23:
I suggest that you look at the implementation of prepare_move_raw() which is the actual routine that you use to implement each of your fractional moves. 

```
    void prepare_move_raw() {
      #if ENABLED(DEBUG_LEVELING_FEATURE)
        if (marlin_debug_flags & DEBUG_LEVELING) {
          print_xyz("prepare_move_raw > destination", destination);
        }
      #endif
      refresh_cmd_timeout();
      calculate_delta(destination);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], destination[E_AXIS], (feedrate / 60) * (feedrate_multiplier / 100.0), active_extruder);
      set_current_to_destination();
    }
```

Roxy-3DPrintBoard, 2015-11-23:
OK, I understand what you are saying.   It doesn't really matter when moving around to position the Z-Probe, but you can hear each of the moves end and the next move start.

closed 2015-11-24 by AnHardt

ex-nerd, 2015-11-24:
Just curious why this was closed with no comment when there seem to be a number of viable options being discussed.

Wackerbarth, 2015-11-24:
"solution is to increase Z_RAISE_BETWEEN_PROBINGS"

ex-nerd, 2015-11-24:
Thanks, though leaving off the "another" at the beginning of that quote really changes the meaning from its original.

The larger the print area, the higher to raise between probings, and the slower the probe process grows.  Looking at the depth of the U movement on my 24cm bed, I could see needing to go 2-3cm up from the 0.5cm that the examples all seem to suggest is a good value.  Moving up and down that much on every probe point (esp. between all of the mid-row probings where it doesn't matter) would make for a very slow probe process when the Z speed is slowed down for better accuracy on a physical switch (vs FSR).

KiteLab, 2015-11-24:
Increasing Z_RAISE_BETWEEN_PROBINGS is the simple solution you can do within the Configuration.h.
For the other possible solutions you have to know much more what you are doing. They may be better, but are not tested that good, or total crap.

Wackerbarth, 2015-11-24:
I wasn't responsible for the closure, but I do agree with it. For the record, in my opinion, the entire G29 regime is "total crap" and needs to be reworked. Therefore, I have little interest in expending effort to refine the present implementation.
Eventually, I would like to see the entire probing sequence replaced with some string-based description that could then be customized for particular configurations.
Potentially, Marlin can be used on such a wide range of designs that I think it to be a mistake to be attempting to solve all of them with the same procedure.
This is just another case where "one size fits all" works, but poorly for everyone. Custom tailored solutions will always be able to handle things, particularly outlying cases, better.

AnHardt, 2015-11-24:
I closed this issue because we had a solution for the particular problem (Head Crash).
We already know there are curved moves in G29 with deltas. We don't need a additional reminder.

ex-nerd, 2015-11-24:
@AnHardt Thanks for the specifics.