MarlinFirmware/Marlin #2793: Independent PID parameters for each extruder Fail

TGMods, 2015-11-23:
As no two heaters are exactly the same, I thought it would be a good idea to enable independent PID parameters for each of my E3D Chimera extruders as per #1344. Unfortunately this failed with the following error: 

C:\Users\Sally\AppData\Local\Temp\build511072971039923312.tmp\temperature.cpp -o C:\Users\Sally\AppData\Local\Temp\build511072971039923312.tmp\temperature.cpp.o 
C:\Users\Sally\AppData\Local\Temp\build511072971039923312.tmp\temperature.cpp: In function 'float get_pid_output(int)':
C:\Users\Sally\AppData\Local\Temp\build511072971039923312.tmp\temperature.cpp:521:71: error: invalid operands of types 'float' and 'float [2]' to binary 'operator*'
             cTerm[e] = (lpq[lpq_ptr] / axis_steps_per_unit[E_AXIS]) * Kc;
                                                                       ^
Error compiling.
Any help would be appreciated.

Wurstnase, 2015-11-23:
You could try to comment ```#define PID_ADD_EXTRUSION_RATE``` in the config_adv.

Why is this on by default?

Wackerbarth, 2015-11-23:
"On, by default, or otherwise", I just pushed a correction and will issue a PR for RCBugFix. Please test it.

TGMods, 2015-11-23:
@Wackerbarth 
I have just updated the copy of  temperature.cpp I am using with your code and it all compiled correctly.
Thank you for your prompt responce.

closed 2015-11-24 by AnHardt