MarlinFirmware/Marlin #2795: FlashAir SD card not working with Marlin

ByReaL, 2015-11-24:
a flash air SD card is a sd card that has also a wifi connection.
now there is something in the marlin initialization code of the sd/wifi card that when it accesses the sd card the wifi is not working anymore.
if i disable any of the CS/MOSI/MISO/SCK pins (cutting the communication with the atmel cpu the wifi works fine)

Interface	Compliant standard SD Memory Card Standard Ver.4.00

http://www.toshiba-memory.com/cms/en/products/wireless-sd-cards/FlashAir/
https://flashair-developers.com/en/support/forum/#/discussion/1642/how-to-get-sd-card-functionality-in-same-time-with-wifi-arduino
http://www.printrbottalk.com/forum/viewtopic.php?f=20&t=9326&p=67762#p67762

KiteLab, 2015-11-24:
Most (all?) SD-card-sockets used with RepRap hardware are connected with 7 from 11 wires only, to save pins on the processor. That works very well, but needs the SD-card to work in MMC-card-compatibility-mode. Marlin supports only that mode.

KiteLab, 2015-11-24:
If there would be a proper hardware around, for sure someone would try to make the natural sd-card-mode work.

thawkins, 2016-02-19:
The solution for this has been determined by an enterprising marlin user, it's a timing issue on SD card initialisation.  

http://www.rc-cam.com/forum/index.php?/topic/4058-marlin-reprap-with-toshiba-flashair-wifi-firmware-patch/

Blue-Marlin, 2016-02-20:
@thawkins 
Did you test is the additional
```
spiSend(0XFF);
```
disturbs the communication with 'normal' sd-cards? If not, the additional configuration parameter cold be omitted - the implementation would be trivial.

jbrazio, 2016-03-26:
Fixed in #3207

closed 2016-03-26 by thinkyhead