MarlinFirmware/Marlin #2796: Support for graphic LCD with KS0108-like driver

siirus, 2015-11-24:
Does Marlin FW support 128x64 graphic lcds with KS0108 (or it's clone NT7108) driver? From reading configuration headers I got the impression that only ST7565R driver and only in SPI mode is supported. Am I correct?

AFAIK Marlin uses u8glib library for interfacing with graphic lcds and it supports KS0108 driver. KS0108 have only 1 operating mode - 8 bit parallel, and from what I know, 8 bit parallel interface is pretty much same for ST7565 and KS0108 (except the fact there is actually 2 KS0108 drivers on 128x64 lcd with shared data bus and separate CS, otherwise they look very similar; correct me if I am wrong, please)

So my question is this: is there possibility of using KS0108 lcd with Marlin FW running on pretty standard arduino+RAMPS 1.4 combo? Does it require only software rework or is there some hardware problems to solve as well?

AnHardt, 2015-11-24:
If  KS0108 is supported in u8glib and you can find enough unused pins for your display and a click-encoder (or an other way to provide UP, DOWN and SELECT) it is possible to make that work in Marlin.
The only different part should be the initialization of the u8glib-display-device (near https://github.com/MarlinFirmware/MarlinDev/blob/master/dogm_lcd_implementation.h#L115):

siirus, 2015-11-24:
OK, thanks for answer and pointing to code! 
KS0108 is definitely supported in u8glib and it seems that there are just enough free pins (need to sacrifice SW reset feature of LCD, yet I am not sure it's so necessary, may hook it to reset pin of atmega or to separate button) on RAMPS to connect LCD and use it!
Going to try this right now, may be if I succeed, KS0108 support can be added to official Marlin FW :)
Should I close the issue now?

thinkyhead, 2016-02-24:
If you manage to get the KS108 working, please post about it on the MarlinDev side of the wall. It won't be making it into the next RC, but it should definitely be proposed for the next 1.1.x and 1.2.x versions.

closed 2016-02-24 by thinkyhead

siirus, 2016-02-27:
I managed to get it to work, actually it's only a couple of lines need to be changed, one thing left to be done is buzzer, hopefully will handle it soon. However I did all this patching on old and heavily modded version of Marlin, so I need to backport it to current Marlin version and clean it up a bit. After it will definitely post patch to MarlinDev

thinkyhead, 2016-02-28:
Looking forward to it! :-1: